linuxwacom/input-wacom #193: Both Wacom One and Intuos no longer working

lschatzkin, 2020-06-16:
I'm running ubuntu 18.04.3. Up until a month ago I could use both my One and Intuos, but they no longer work.  I would occasionally have a problem in the past where I would have to reinstall input-wacom, following the instructions "Installing input wacom from source" but even doing this does not solve the problem. The devices indicator light is lit but the pen does not control the cursor.   

When I try running `sudo rmmod wacom`, I get the error: `rmmod: ERROR: Module wacom is not currently loaded`.

I'm not sure what to try next.

rjbruin, 2020-06-20:
I have the same problem. The output of `sysinfo.sh` is attached.

[sysinfo.eq9rRwYswO.tar.gz](https://github.com/linuxwacom/input-wacom/files/4807883/sysinfo.eq9rRwYswO.tar.gz)

rjbruin, 2020-06-21:
My problem was solved by reinstalling the drive rmanually.

lschatzkin, 2020-06-22:
> My problem was solved by reinstalling the drive rmanually.

Cool, which drivers?

rjbruin, 2020-06-23:
The `input-wacom` driver, using [these instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source). Specifically version 0.46.0.

jigpu, 2020-06-23:
@lschatzkin could you please run [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to gather information about your system and its drivers?

The input-wacom driver stops working after any kernel update (which is why you had to occasionally re-install it in the past), but its odd that your most recent attempt at installing failed. The sysinfo script should let us know if the driver was installed successfully or not, and help us understand what might being going wrong if it is.

lschatzkin, 2020-06-24:
Thanks @jigpu , I'm attaching the file here.
[sysinfo.schatzkin.tar.gz](https://github.com/linuxwacom/input-wacom/files/4828261/sysinfo.schatzkin.tar.gz)

jigpu, 2020-06-25:
Looks like the driver wasn't re-installed properly. The logs show copies of input-wacom built for several previous kernels as you describe, but your current kernel only has the distribution's stock driver. Please try rebuilding the driver again, and copy/paste the terminal output so we can see what is happening.

lschatzkin, 2020-06-25:
Huh, for some reason it worked this time! Thanks for helping me out!

jigpu, 2020-06-25:
Sounds good :smile: Feel free to re-open if re-installing doesn't work in the future.

closed 2020-06-25 by jigpu