linuxwacom/input-wacom #194: Intuos CTL-480 was working fine until a few days ago (Arch Linux, Wacom driver 0.43.0)

ArtAbu, 2020-06-18:
Installed the driver about a week ago, was working fine until a few days ago it suddenly stopped working. It works fine for a few minutes, then it start disconnecting until it goes dead. Tried rebooting, reinstalling, everything and it still doesn't work.

jigpu, 2020-06-18:
Please reboot with the tablet plugged in, wait for the tablet to stop working, and then run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh). Attach the generated tarball here afterwards. Also, please answer the following questions if you can:

* Did you update your system since installing the driver a week ago?
* Is there a reason you are using version 0.43.0 of the driver (which is out-of-date)?
* Is there a reason you are using the input-wacom driver instead of the driver that is installed alongside the Arch Linux kernel?
* Can you go into more detail about "It works fine for a few minutes, then it start disconnecting until it goes dead."
    * Does the tablet temporarily work after a reboot?
    * Does the tablet temporarily work after unplugging and reconnecting? Does it work for longer/shorter than if you reboot?
    * Any other detail about the symptoms that you can think of...
* What desktop environment are you using, and is it a Wayland or Xorg session? (run `echo $XDG_SESSION_TYPE` if you are unsure)

rjbruin, 2020-06-20:
Could this be the same issue as #193? I've added my output of `sysinfo.sh` there.

ArtAbu, 2020-06-20:
-  Did you update your system since installing the driver a week ago?
I did not update at all.
  - Is there a reason you are using version 0.43.0 of the driver (which is out-of-date)?
Sorry about that, I'm actually using 0.46.0, I ended up misreading when I used grep "" /sys/module/wacom*/version.
 - Is there a reason you are using the input-wacom driver instead of the driver that is installed alongside the Arch Linux kernel?
Sorry, I do not understand the question, I just followed the instructions I saw here on this repository, which I found through the arch wiki.
  - Can you go into more detail about "It works fine for a few minutes, then it start disconnecting until it goes dead."
     -   Does the tablet temporarily work after a reboot?
Yes, after rebooting it works fine for about ten minutes, then the pen stops working, even though the white light on the tablet is still lit up, after a while that same light starts blinking (like it's connecting and disconnecting) until it simply stops.
      -  Does the tablet temporarily work after unplugging and reconnecting? Does it work for longer/shorter than if you reboot?
Yes, but in a unusual way. When I unplug it and plug it again, it doesn't light up immediately, taking a while before it can work. But the problem repeats.
       - Any other detail about the symptoms that you can think of...
I noticed that sometimes, while using the pen during the short time in which it is working, the mouse would simply not move, even after taking the pen away from the tablet, it stayed still like I was making contact with the tablet, but that only happened on rare occasions.
  - What desktop environment are you using, and is it a Wayland or Xorg session? (run echo $XDG_SESSION_TYPE if you are unsure)
I'm using Xorg.

ArtAbu, 2020-06-20:
> Could this be the same issue as #193? I've added my output of `sysinfo.sh` there.

I'm not sure about that, it looks like a similar problem but I did not get that error when running `sudo rmmod wacom`

rjbruin, 2020-06-21:
> > Could this be the same issue as #193? I've added my output of `sysinfo.sh` there.
> 
> I'm not sure about that, it looks like a similar problem but I did not get that error when running `sudo rmmod wacom`

Actually, the problem was solved for me by reinstalling the driver with manual installation.

jigpu, 2020-06-23:
Thanks for the information @ArtAbu.

The kernel log seems to match up with your description: it stops working about 10 minutes after you plug it in, and then it just repeatedly connects and disconnects (which would explain the light blinking). I don't see any obvious software issues in the logs provided, however.

Do you have access to a Windows machine with the Wacom drivers that you could try the tablet with? Its possible that this is a hardware problem rather than a software problem.

Also, if you don't mind, could you please perform the following steps to upload a complete system log for a single boot so I can check for other potential sources of the problem:

1. Reboot and log in (so we can isolate the logs to just the current boot and make issues easier to see).
2. Connect the tablet.
3. Try to not use the computer. Just wait for the issue as signaled by the blinking light (again to make issues easier to see by not having unrelated log entries).
4. When the issue occurs, open a terminal and run `sudo journalctl -b0 | gzip > /tmp/out.log.gz` to capture the full system log for the current boot.
5. Attach the /tmp/out.log.gz file here for analysis.

jigpu, 2020-06-25:
Another test request: would you mind trying to install version 0.45.0 of the driver rather than the current 0.46.0? The latest release has some updates that affected the CTL-480 and its possible that they are the source of the issue. After installing, please double-check with `grep "" /sys/module/wacom*/version` that version 0.45.0 is loaded.

jigpu, 2020-06-26:
And one more request since the submitter of #195 found that rolling back to 0.45.0 fixed their problem. If rolling back also works for you, could you try using the steps outlined at https://github.com/linuxwacom/input-wacom/issues/195#issuecomment-650211742 to try and isolate the buggy commit?

Pinglinux, 2022-08-03:
It's been more than 2 years without any follow-up reports. One of the steps should have resolved the issue.

closed 2022-08-03 by Pinglinux