linuxwacom/input-wacom #195: Wacom Intuos  CTH-680 not detected by Fedora 32

pgerlier, 2020-06-25:
After updating my desktop to Fedora 31 then 32, my Intuos CTH-680 is no longer recognised by the OS.

After attaching it to an USB port, the white power LED at top left blinks for ever.

`lsusb` shows it is attached to a port as device `056a:0303` but the KDE configuration utility says "No tablet detected".

The CTH-680 works like a charm on my laptop with an outdated Fedora release (also under KDE desktop).

Installed packages:
- kcm_wacomtablet 3.1.1 (4.fc32)
- libwacom 1.3 (1.fc32)
- libwacom-data 1.3 (1.fc32)
- xorg-x11-drv-wacom 0.39.0 (2.fc32)
- xorg-x11-drv-xacom-serial-support 0.39.0 (2.fc32)

Unless I'm wrong, I seem to remember Fedora has switched to Wayland by default. Is this the cause of the problem? How to remedy it?

jigpu, 2020-06-25:
Given the symptoms, it is unlikely to be related to any switch to Wayland. You should be able to request the use of an Xorg session instead by clicking the gear icon at the login screen and selecting "GNOME on Xorg" rather than "GNOME".

The circumstances also remind me of #194. Same flashing light, same generation of tablet (the smaller CTH-480), similar kernel (running Linux 5.6, which I believe is the same as Fedora 32 at the moment). Their issue appears after a several minutes rather than immediately, however.

Would you mind trying to install version 0.45.0 (*specifically* that version **not** the latest) of the input-wacom driver to see if that changes the behavior? Instructions for building and installing can be found at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source. Be sure to verify the active kernel version after installation by running `grep "" /sys/module/wacom*/version`.

pgerlier, 2020-06-26:
A few precisions for @jigpu 

- After checking, I am still under X11 (when forcing Wayland, the computer freezes)
- GNOME is not installed, my desktop is KDE Plasma and Wayland integration in Plasma is lagging a lot behind GNOME. This may explain the freeze.

I'll proceed with driver compilation and installation.

pgerlier, 2020-06-26:
Using the 0.45.0 driver seems to have solved the problem. Thanks for your help.

Hoping for Fedora to upgrade soon. 0.39.0 seems to date back to March 2018. Fedora claims to be "nearly" bleeding edge. So, I don't understand why it is lagging that much.

jigpu, 2020-06-26:
> Using the 0.45.0 driver seems to have solved the problem. Thanks for your help.

Thanks for checking that out -- looks like we introduced a bug... Would you mind doing a bit more work so we can isolate exactly which change was responsible?

You'll need to install the `git` command and then use `git clone https://github.com/linuxwacom/input-wacom.git` to get a copy of the development driver. Next, enter the input-wacom directory and run `git bisect start input-wacom-0.46.0 input-wacom-0.45.0` to start the process of hunting for the bad change. You should see a message like this in response:

~~~
Bisecting: 4 revisions left to test after this (roughly 2 steps)
[661c0e7d740f88bcb979d81df726a2347d44a7da] backport: HID: wacom: Read HID_DG_CONTACTMAX directly for non-generic devices
~~~

Next, repeat the following steps:

1. Take note of the first seven characters of the second line (e.g. "661c0e7" for the message above).
2. [Build and install the driver](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source#build--install) as you did before.
3. Reboot
4. Run `grep "" /sys/module/wacom*/version` and verify that its output ends with the seven characters you took note of.
5. See if the tablet works or not. If it doesn't, run `git bisect bad` inside the input-wacom directory. If the tablet does work, run `git bisect good` instead.
6. If you get another "Bisecting" message, go back to step 1. If instead you see a bunch of output beginning with the line "<bunch of characters> is the first bad commit", let us know what that bunch of characters is.

> Hoping for Fedora to upgrade soon. 0.39.0 seems to date back to March 2018. Fedora claims to be "nearly" bleeding edge. So, I don't understand why it is lagging that much.

You're confusing the two drivers. This ("input-wacom") driver is an out-of-tree version of what comes with the Linux kernel. The most recent 0.46.0 release includes all of the changes that were made to the most recent Linux 5.7 kernel. The "xorg-x11-drv-wacom" driver that is at version 0.39.0 on your system is our driver for the Xorg server, also known as "xf86-input-wacom". Version 0.39.0 is the most recent release of that driver.

pgerlier, 2020-06-26:
@jigpu: I discovered something which puzzles me and may totally invalidate my statement about the OS not detecting my tablet. I need to check first.

How can I remove the custom-built driver to verify the behaviour with the distro driver?

`cat /sys/module/wacom/version` reports `v2.00-0.45.0`. Before going for a bisection, I want to double check the distro driver.

jigpu, 2020-06-29:
If you still have the input-wacom-0.45.0 directory that you built the driver from, you can go into it and run `sudo make uninstall` to remove the driver and then reboot. We don't overwrite the system driver when installing, so when you uninstall the system driver should take over again after a reboot. If you don't have the directory anymore, you can just install the driver from source again and then immediately uninstall it.

Once you're back on the system driver, running `cat /sys/module/wacom/version` should report `v2.00`

pgerlier, 2020-06-30:
It was quite hard to get rid of the driver. For an unknown reason, I had to reboot in an older version before it disappeared.

I send you very humbly my apologies. It was all my fault. The driver works fine.

The tablet comes with a rather short USB cable. The desktop cabinet is put on the floor aside my desk. Consequently, the tablet cannot be used comfortably with the shipped cable unless I set clumsily on my knee with my armchair offset in an odd position. Of course, I did not connect the tablet to the keyboard because of the limited power which can be drained from the keyboard.

I used a Sony USB extension cable to be able to have the tablet on the desk. This is where the power LED was blinking. To check the tablet state, I connected to my laptop only with the shipped cable, leaving the extension on my desktop. I also tested all USB ports on the frontside as well as on the backside in case the frontside could not deliver the amperage needed by the tablet. Same result.

After installation of the 0.45.0 driver, I monitored the system log to have a look on the mount events. This was only at this moment that I noticed that every time I plugged in the tablet, a USB hub was also detected.

I then exchange the Sony cable for another (shorter) one I had in a drawer. With this cable, there were no more any problems.

I am ashamed for not having fully tested my environment before posting here. But I would never have imagined that an extension cable could contain a hub. This very small hub could not let enough current through for the tablet. Present-day integrated circuits are really so small they can fit inside a plug!

Once again, apologies.

I don't know if you're affiliated in any way with Wacom, but I'd suggest that USB cable for medium- and high-end tablets be long enough to be compatible with workstation office environment where the PC tower is not on the desk surface.

Thanks for your help and patience. You can close the report as NOT-A-BUG (user error).

jigpu, 2020-06-30:
Glad you were able to figure it out! A hidden in-cable hub not letting enough power though certainly wouldn't have been the first diagnosis to spring to my mind... I'll have to remember that blinking may be a sign of power issues in the future. :+1:

closed 2020-06-30 by jigpu