linuxwacom/input-wacom #196: Wacom One 13 (DTC) has mismatched screen

FallenAngel97, 2020-07-09:
Hello!
I just bought Wacom One 13 and successfully booted up in Windows and did a test drive. 
Now I try to check it in my Arch environment, and it seems, that it's working, but screen coordinates are mismatched.
Also, `xsetwacom list devices` doesn't show it in terminal and `dmesg | grep wacom` Shows this:
```
[  263.118395] wacom 0003:056A:03A6.0001: hidraw0: USB HID v1.10 Device [Wacom Co.,Ltd. Wacom One Pen Display 13] on usb-0000:00:14.0-1/input0
[  263.119482] wacom 0003:056A:03A6.0002: Unknown device_type for 'Wacom Co.,Ltd. Wacom One Pen Display 13'. Ignoring.
```

I have tried to install from Arch repositories and from this source (master branch) and still no luck. Can you, please, help me?

FallenAngel97, 2020-07-09:
Output of `libwacom-list-local-devices`:

```
# Device node: /dev/input/event5
[Device]
Name=Wacom One Pen Display 13
ModelName=DTC133
DeviceMatch=usb:056a:03a6;
Class=PenDisplay
Width=11
Height=6
IntegratedIn=Display;
Styli=0xffffe;0xfffff;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=0

---------------------------------------------------------------
```

jigpu, 2020-07-09:
Could you please reboot with the Cintiq attached and then run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to gather system logs and hardware info? Attach the generated tarball here for review.

FallenAngel97, 2020-07-10:
[sysinfo.I1ztcFbWgO.tar.gz](https://github.com/linuxwacom/input-wacom/files/4901091/sysinfo.I1ztcFbWgO.tar.gz)
Yeah, here you go :)

jigpu, 2020-07-10:
It looks like the source of the problems is an `/etc/X11/xorg.conf.d/30-touchpad.conf` file which you've created. It has its `MatchIsTouchpad` rule commented out, so it ends up trying to have the synaptics driver take control of every input device. You don't seem to have the synaptics driver installed at all, however, so everything falls back to using libinput. The Wacom One doesn't work well with libinput, however.

Since you don't have the synaptics driver on your system, I would recommend simply deleting the `/etc/X11/xorg.conf.d/30-touchpad.conf` file. That should result in everything going back to the default behavior, which has most devices powered by libinput and the tablet powered by the wacom driver.

You should also be able to uninstall this driver (open a terminal, navigate into the input-wacom source directory, and run `sudo make uninstall`) since it probably isn't providing any benefit over the default Arch linux driver.

FallenAngel97, 2020-07-10:
Yeah, thanks!
It showed up and I could run following command:
```
xsetwacom set 10 MapToOutput 1920x1080+1920+0
```
so it was properly offsetted (stylus).
But removing that config file disabled my single tap functionality on touchpad... I will try to investigate this issue. 
Thanks for the help!

closed 2020-07-10 by FallenAngel97