linuxwacom/input-wacom #197: Modprobe exec format error

rjbruin, 2020-07-23:
I installed updates on my Ubuntu 18.04 machine, and ran the install procedure for input wacom again. Usually, running `sudo rmmod wacom` followed by `sudo modprobe wacom` is sufficient to get my tablet to work again. Now though I'm getting an error:

```
modprobe: ERROR: could not insert 'wacom': Exec format error
```

The tablet is not responsive anymore.

jigpu, 2020-07-24:
I believe this error can occur if the module was compiled for a different kernel than is currently in use. If you re-ran the install procedure after performing an update but _before_ rebooting, that might explain the problem (module gets built against updated kernel, but can't be loaded since old kernel is still running). I would reboot and check the version number of the loaded module, and retry the build/install process if the version number is not what you expect.

If that doesn't work, I would suggest deleting your current input-wacom source directory and rebuilding from a freshly-extracted copy.

rjbruin, 2020-07-25:
I installed again from a freshly-extracted copy of the library, then rebooted the PC, and that solved it. Thanks very much! :)

closed 2020-07-25 by rjbruin