linuxwacom/input-wacom #198: Pen Not Connecting [Ubuntu 18.04, Intuous M, v2.00-0.46.0]

tetsef, 2020-07-24:
My pen is no longer working when connecting the Wacom tablet through Bluetooth. Previously when this happened I would rerun
`if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"`
and it would work after restarting my computer.

Connecting the tablet through USB works and the pen registers, but the mouse jumps around sporadically and doesn't map 1:1 from my screen to the tablet. This has always been the case, however, so I've always used Bluetooth.

jigpu, 2020-07-24:
If you don't see a "Build Failed" message after running the command, the driver should have been properly re-configured, built, and installed for whatever updated kernel you may be using. Please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) twice for us: once with the tablet connected over Bluetooth, and a second time with it plugged in over USB. Hopefully this will help us figure out what might be going on...

BTW, your mapping issue could be the [infamous Android-misdetect problem](https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect). If you have one of the affected models, try the workaround in that link.

tetsef, 2020-07-27:
That fixed the mapping issue, thank you. The LED is fixed when plugged in, but now it doesn't recognize the pen movements. So the behavior is now the same with Bluetooth and USB.
Here are the sysinfo.sh results:
[sysinfo.bluetooth.tar.gz](https://github.com/linuxwacom/input-wacom/files/4983845/sysinfo.bluetooth.tar.gz)
[sysinfo.usb.tar.gz](https://github.com/linuxwacom/input-wacom/files/4983846/sysinfo.usb.tar.gz)

jigpu, 2020-07-28:
Thanks. It looks like there's some kind of versioning issue going on. Whenever the kernel tries to load our driver, it prints out the following error:
~~~
[  378.393252] wacom: version magic '4.15.0-111-generic SMP mod_unload ' should be '4.15.0-112-generic SMP mod_unload '
~~~
Looking at the list of installed drivers, it shows that version 0.46.0 of the input-wacom driver was successfully installed to `/lib/modules/4.15.0-112-generic/extra/wacom.ko` as I would expect, but its version magic string is exactly as the error message indicates.

I would try to run `sudo make uninstall` from inside the input-wacom source directory to remove the input-wacom driver, reboot, and then run `modinfo wacom | grep -E "filename|vermagic"`. If the uninstall was successful you should see the following output:
~~~
filename: /lib/modules/4.15.0-112-generic/kernel/drivers/hid/wacom.ko
vermagic: 4.15.0-112-generic SMP mod_unload 
~~~
This output confirms that your system is back to using the stock Wacom kernel driver. Next, delete your input-wacom source directory and then re-download/build/install a fresh copy of input-wacom. That will hopefully fix whatever strange configuration state it managed to get itself into.

tetsef, 2020-07-28:
That worked, thank you! I should've tried that earlier.

closed 2020-07-28 by tetsef