linuxwacom/input-wacom #199: X session crashes using  Wacom One Pen Display 13

enriqueartal, 2020-07-28:
I am using a Wacom One Pen Display 13 in a computer with Fedora 32. It works quite well with some small annoyances. The main issue is that the whole X systems crashes randomly when using the tablet. I see no helpful message in the logs, maybe I am not looking at the right place. I can access the system and kill my processes and login again.

jigpu, 2020-07-28:
Can you run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) both before and after a crash and attach each tarball it generates to this issue?

enriqueartal, 2020-07-29:
I ran the script this morning and got another crash (the first pass was around 20 minutes earlier).
[sysinfo_after.tar.gz](https://github.com/linuxwacom/input-wacom/files/4995083/sysinfo_after.tar.gz)
[sysinfo_before.tar.gz](https://github.com/linuxwacom/input-wacom/files/4995085/sysinfo_before.tar.gz)

jigpu, 2020-07-29:
Thanks. Did you gather the "after" from a VT or SSH session (as opposed to from a fresh Xorg session or from inside a Wayland session)? It's completely fine to have done so, but I just want to be sure I properly understand the logs and their context.

I'm also not seeing a lot of useful information, unfortunately. Comparing the two log snapshots doesn't show any kind of error message at all. I did notice that you appear to have a USB DisplayLink device connected. I haven't used DisplayLink in quite a while, but in the past I found their driver to be somewhat unstable. I wonder if it might be involved somehow. Could you try disconnecting the DisplayLink device to see if it has any effect on your system's stability?

Also, can you please run `sudo coredumpctl list > crashes.txt` and attach the crashes.txt file here? If the X server is crashing hard enough to not leave anything in its logs, perhaps the system was able to capture a coredump...

enriqueartal, 2020-07-29:
I gathered it from an SSH session, I couldn't access VT and to access Xorg again I should kill my session. I will try the coredump and also I will buy an DisplayPort-HDMI adapter as maybe  displaylink is to blame.
Thanks.

enriqueartal, 2020-07-30:
I tried this morning with a DP-HDMI adapter and no crash. Maybe evdi module is behind the crashes. I will try again.