linuxwacom/input-wacom #200: Random change of display settings

AmineCybernetics, 2020-07-29:
When the computer is booted with the wacom tablet connected issues appear with the external display. When some software is opened or the system settings are opened or some notifications are received, the system changes randomly from two screens to one or to mirror screens. When the tablet is not connected this behavior does not appear. 
I am running ubuntu 20.04 with input-wacom-0.46.0 and using wacom intuos.

jigpu, 2020-07-29:
Can you please try to trigger the issue, and once it occurs immediately open a terminal and run `sudo journalctl --since="5 minutes ago" > error.log` for us? Attach the error.log file and we'll review it to see if it has any clues about what may be going on.

Also, could you please attach the tarball generated by our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to get some additional information about your hardware, software, and drivers?

AmineCybernetics, 2020-07-29:
Please find the files attatched here.
[error.log](https://github.com/linuxwacom/input-wacom/files/4995637/error.log)
[sysinfo.riYj9y2sYR.tar.gz](https://github.com/linuxwacom/input-wacom/files/4995651/sysinfo.riYj9y2sYR.tar.gz)

Pinglinux, 2022-08-04:
Which Wacom tablet do you use? I can't find a Wacom device from the log files.

Pinglinux, 2022-08-12:
I can't find any Wacom devices in the error.log and the tar file.

closed 2022-08-12 by Pinglinux