linuxwacom/input-wacom #202: Intuos BT S problems with Ubuntu 18.04

efernbach, 2020-08-04:
followed https://linuxwacom.github.io/ to install kernel driver and xinput driver
modinfo wacom | grep version output:
   version:        v2.00-0.46.0
   srcversion:     62537F6171CFA60B57FF9D7

tablet not detected in settings > devices .> Wacom Tablet
xsetwacom --list devices output has pad, stylus, eraser and mouse (after boot),
upon unplugging and replugging xsetwacom --list devices output has only stylus and eraser

the pad works fine after boot but no longer after unplugging and replugging (for symptom details see https://askubuntu.com/questions/1259512/wacom-intuos-configuration-lost-after-hotplug).

These bug reports might be related
https://bugs.launchpad.net/ubuntu/+source/gnome-control-center/+bug/1753525
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1498664

jigpu, 2020-08-05:
Given the tablet model and some of the description given in your AskUbuntu post, its possible that this is a case of the tablet mis-identifying your system as an Android device. See https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect for more details and a workaround. Let me know if the workaround seems to fix the problem after hotplug.

efernbach, 2020-08-19:
The workaround does not fix the fact that the tablet isn't recognized in Ubuntu settings, but it does fix the usability of the tablet. 
After hotplug in some cases the tablet works well, in some cases it doesn't. If it doesn't work, the indicator led has turned from white to grey. Pressing the left an right button together for a prolonged time makes the led turn white again and the tablet works.

jigpu, 2020-08-19:
To fix the tablet not being recognized in Ubuntu settings you will need an updated libwacom database. You can try to install the current libwacom release, but a simpler option would be to download the following file and then extract it to the root directory by running `sudo tar xvzf intuos-s-p3-wl.tar.gz -C /` (this tarball contains just the new/updated files for your tablet).

* [intuos-s-p3-wl.tar.gz](https://github.com/linuxwacom/input-wacom/files/5097301/intuos-s-p3-wl.tar.gz)

rav0rmat, 2021-01-21:
Hello, I own the same tablet and installing the database does not make it visible. Bluetooth connects fine but as the system does not recognise it, it disconnects after a few seconds.

jigpu, 2021-01-21:
@rav0rmat have you also installed this input-wacom driver, or did you only install the database? Both are required for Ubuntu 18.04 to work properly. You can find instructions for installing input-wacom at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

jigpu, 2021-01-21:
@rav0rmat I have opened a new issue for you since it sounds like you're experiencing a different problem.

For future reference, this tablet should work out of the box with Ubuntu 20.04. Installing input-wacom / libwacom are only required for older distros.

closed 2021-01-21 by jigpu

RangerMauve, 2025-02-28:
Sorry to necrobump but I'm currently dealing with this issue. After updating the `.tablet` file from the tarball provided by @jigpu I'm still not getting bluetooth support and now when I try to list local devices I get the following error:

```

(libwacom-list-local-devices:4619): libwacom-WARNING **: 16:07:17.346: Duplicate definition for stylus ID '0x806'
failed to match 'usb:056a:0376' for product/vendor IDs. Skipping.
'usb:056a:0376' is an invalid DeviceMatch in '/usr/share/libwacom/intuos-s-p3-wl.tablet'
failed to match 'bluetooth:056a:0377' for product/vendor IDs. Skipping.
'bluetooth:056a:0377' is an invalid DeviceMatch in '/usr/share/libwacom/intuos-s-p3-wl.tablet'

(libwacom-list-local-devices:4619): libwacom-WARNING **: 16:07:17.351: Ignoring invalid .tablet file intuos-s-p3-wl.tablet
```

When I try to connect over USB it's able to show up with this info:

```
devices:
  - name: 'Wacom Intuos BT S'
    bus: 'usb'
    vid: 0x056a
    pid: 0x0376
    nodes: 
      - /dev/input/event20: 'Wacom Intuos BT S Pen'
      - /dev/input/event21: 'Wacom Intuos BT S Pad'
    styli: []
```

RangerMauve, 2025-02-28:
Unsure if this is relevant but here's the bluetooth info for the device:

```
> bluetoothctl info
Device DC:2C:26:08:D7:61 (public)
	Name: Intuos BT S
	Alias: Intuos BT S
	Class: 0x00002594 (9620)
	Icon: input-tablet
	Paired: yes
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: yes
	WakeAllowed: no
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v056Ap0377d0000
	RSSI: 0xffffffca (-54)
```

Could it be that the device match needs to be adjusted?

dkniffin, 2025-06-10:
@RangerMauve I'm also seeing that issue. Did you find a solution?

RangerMauve, 2025-06-10:
@dkniffin I just gave up on bluetooth and have been using USB 😅

dkniffin, 2025-06-10:
https://github.com/linuxwacom/input-wacom/issues/439 seems to be the same issue as me as well.

@jigpu could we re-open one of these? Or do you want me to make a new issue for my problem?

BarracudaLake, 2026-09-29:
necro, but +1 to (re)open the issue.

I am actively experiencing this issue on Fedora 44.
my Intous M shows as connected, yet doesnt appear in settings.
details listed above seem to match my situation.