linuxwacom/input-wacom #203: Build never failed but driver installation not successful

Shru-Tri, 2020-08-19:
shruti@ThinkPad-Edge-E431:~/wacom-driver/input-wacom-0.46.0$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking if build path '/home/shruti/wacom-driver/input-wacom-0.46.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/5.3.0-59-generic/build
checking kernel version... 5.3.0-59-generic
checking RHEL6 minor release... 0
checking RHEL7 minor release... 0
checking power supply version... v4.1+
checking abs accessor version... v2.6.36+
checking CONFIG_MODULE_SIG... y
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... y
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... off
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha512
checking for module signing hash algorithm... sha512
checking for kernel autogenerated key at /lib/modules/5.3.0-59-generic/build/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/5.3.0-59-generic/build/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... no
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... no
checking for rEFInd MOK key at /etc/refind.d/keys/refind_local.key... no
checking for rEFInd MOK cert at /etc/refind.d/keys/refind_local.cer... no
checking for module signing key... 
checking for module signing certificate... 
configure: WARNING: Kernel module signing is recommended but we are unable to find signing keys. Consider re-running configure with the options '--with-signing-key=<key>' and '--with-signing-cert=<cert>' set.

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/5.3.0-59-generic/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
Making all in 4.5
make[2]: Entering directory '/home/shruti/wacom-driver/input-wacom-0.46.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/5.3.0-59-generic/build M=/home/shruti/wacom-driver/input-wacom-0.46.0/4.5
make[3]: Entering directory '/usr/src/linux-headers-5.3.0-59-generic'
  Building modules, stage 2.
  MODPOST 2 modules
make[3]: Leaving directory '/usr/src/linux-headers-5.3.0-59-generic'
make[2]: Leaving directory '/home/shruti/wacom-driver/input-wacom-0.46.0/4.5'
make[2]: Entering directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
make[2]: Leaving directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
make[1]: Leaving directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
Making install in 4.5
make[1]: Entering directory '/home/shruti/wacom-driver/input-wacom-0.46.0/4.5'
make -C /lib/modules/5.3.0-59-generic/build M=/home/shruti/wacom-driver/input-wacom-0.46.0/4.5 modules_install mod_sign_cmd=''
make[2]: Entering directory '/usr/src/linux-headers-5.3.0-59-generic'
  INSTALL /home/shruti/wacom-driver/input-wacom-0.46.0/4.5/wacom.ko
/bin/sh: 1: /lib/modules/5.3.0-59-generic/extra/wacom.ko: Permission denied
  INSTALL /home/shruti/wacom-driver/input-wacom-0.46.0/4.5/wacom_w8001.ko
/bin/sh: 1: /lib/modules/5.3.0-59-generic/extra/wacom_w8001.ko: Permission denied
  DEPMOD  5.3.0-59-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.3.0-59-generic'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin" depmod -a 5.3.0-59-generic
/usr/sbin/update-initramfs -u -k 5.3.0-59-generic
update-initramfs: Generating /boot/initrd.img-5.3.0-59-generic
I: The initramfs will attempt to resume from /dev/sda6
I: (UUID=92523db5-b063-4062-908a-a863593c2c24)
I: Set the RESUME variable to override this.
make[1]: Leaving directory '/home/shruti/wacom-driver/input-wacom-0.46.0/4.5'
make[1]: Entering directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
make[2]: Entering directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
make[1]: Leaving directory '/home/shruti/wacom-driver/input-wacom-0.46.0'
[sysinfo.w5gRYdKMv5.tar.gz](https://github.com/linuxwacom/input-wacom/files/5097312/sysinfo.w5gRYdKMv5.tar.gz)

jigpu, 2020-08-19:
I don't see any issues in the compile log or sysinfo logs to indicate any kind of failure. The 0.46.0 driver appears to be installed to the correct location, loaded by the kernel, and showing an Intuos S tablet connected.

Can you please describe the symptoms are you experiencing? Does the cursor move in response to the pen? Does the pen not work in certain applications or at certain times?

Shru-Tri, 2020-08-19:
Thanks Jason, yes the cursor moves in response to the pen. However, I am unable to do any settings on the wacom tablet. I want to use it to write equations/ formulae. Attaching the screenshot of the settings on stylus and tablet for you to assess and let me know.
[image]

jigpu, 2020-08-20:
To fix the tablet not being recognized in Ubuntu settings you will need an updated libwacom database. You can try to install the current libwacom release, but a simpler option would be to download the following file and then extract it to the root directory by running sudo tar xvzf intuos-s-p3.tar.gz -C / (this tarball contains just the new/updated files for your tablet). After updating the database you will want to restart.

* [intuos-s-p3.tar.gz](https://github.com/linuxwacom/input-wacom/files/5103764/intuos-s-p3.tar.gz)