linuxwacom/input-wacom #206: Ubuntu 18.04.5 One by Wacom CTL-472 Cannot get it working

daBlesr, 2020-09-10:
I have tried to follow the installation page on the Linux driver, however it still does not seem to work, even though the installation seems to successfully work. There is no response on my machine from the wacom tablet.

Below is the output of the installation: 
[output install wacom.txt](https://github.com/linuxwacom/input-wacom/files/5203468/output.install.wacom.txt)

Could you please help me in getting the Wacom tablet to work?
Thanks in advance.

Jaderpolli, 2020-09-22:
Did you solve it? I'm having this ecact same issue, with the same outputs.

lsaravia, 2020-09-22:
I have a similar result with CTL-472 the output of the compilation is

```
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal 
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf
autoreconf: running: /usr/bin/autoheader
autoreconf: running: automake --add-missing --copy --no-force
autoreconf: Leaving directory `.'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... y
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... off
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha512
checking for module signing hash algorithm... sha512
checking for kernel autogenerated key at /lib/modules/4.15.0-117-generic/build/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/4.15.0-117-generic/build/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... yes
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... yes
checking for module signing key... /var/lib/shim-signed/mok/MOK.priv
checking for module signing certificate... /var/lib/shim-signed/mok/MOK.der

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/4.15.0-117-generic/build
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/leonardo/input-wacom'
Making all in 4.5
make[2]: Entering directory '/home/leonardo/input-wacom/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/4.15.0-117-generic/build M=/home/leonardo/input-wacom/4.5
make[3]: Entering directory '/usr/src/linux-headers-4.15.0-117-generic'
  Building modules, stage 2.
  MODPOST 2 modules
make[3]: Leaving directory '/usr/src/linux-headers-4.15.0-117-generic'
make[2]: Leaving directory '/home/leonardo/input-wacom/4.5'
make[2]: Entering directory '/home/leonardo/input-wacom'
make[2]: Leaving directory '/home/leonardo/input-wacom'
make[1]: Leaving directory '/home/leonardo/input-wacom'
Making install in 4.5
make[1]: Entering directory '/home/leonardo/input-wacom/4.5'
make -C /lib/modules/4.15.0-117-generic/build M=/home/leonardo/input-wacom/4.5 modules_install mod_sign_cmd='"/lib/modules/4.15.0-117-generic/build/scripts/sign-file" "sha512" "/var/lib/shim-signed/mok/MOK.priv" "/var/lib/shim-signed/mok/MOK.der"'
make[2]: Entering directory '/usr/src/linux-headers-4.15.0-117-generic'
  INSTALL /home/leonardo/input-wacom/4.5/wacom.ko
  INSTALL /home/leonardo/input-wacom/4.5/wacom_w8001.ko
  DEPMOD  4.15.0-117-generic
make[2]: Leaving directory '/usr/src/linux-headers-4.15.0-117-generic'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin" depmod -a 4.15.0-117-generic
/usr/sbin/update-initramfs -u -k 4.15.0-117-generic
update-initramfs: Generating /boot/initrd.img-4.15.0-117-generic
make[1]: Leaving directory '/home/leonardo/input-wacom/4.5'
make[1]: Entering directory '/home/leonardo/input-wacom'
make[2]: Entering directory '/home/leonardo/input-wacom'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/leonardo/input-wacom'
make[1]: Leaving directory '/home/leonardo/input-wacom'
```
after reboot 

```
grep "" /sys/module/wacom*/version
grep: /sys/module/wacom*/version: No such file or directory

modinfo wacom | grep version
version:        v2.00-0.46.0
srcversion:     62537F6171CFA60B57FF9D7

$sudo rmmod wacom
rmmod: ERROR: Module wacom is not currently loaded
$ sudo rmmod wacom_w8001
rmmod: ERROR: Module wacom_w8001 is not currently loaded
$ sudo modprobe wacom
modprobe: ERROR: could not insert 'wacom': Exec format error
$ sudo modprobe wacom_w8001
modprobe: ERROR: could not insert 'wacom_w8001': Exec format error
```

And I checked that secure boot is not enabled 

```
$ ls /sys/firmware/efi/
ls: cannot access '/sys/firmware/efi/': No such file or directory
$ mokutil --sb-state
EFI variables are not supported on this system

```

LKreutzer, 2020-11-11:
I have pretty much the same issue, it was working fine until the last Kernel update. 
Does anyone have a solution to this problem?

grep "" /sys/module/wacom*/version
grep: /sys/module/wacom*/version: No such file or directory

modinfo wacom | grep version
version:        v2.00-0.46.0
srcversion:     62537F6171CFA60B57FF9D7

sudo rmmod wacom
rmmod: ERROR: Module wacom is not currently loaded

sudo rmmod wacom_w8001
rmmod: ERROR: Module wacom_w8001 is not currently loaded

sudo modprobe wacom
modprobe: ERROR: could not insert 'wacom': Exec format error

sudo modprobe wacom_w8001
modprobe: ERROR: could not insert 'wacom_w8001': Exec format error

 mokutil --sb-state
SecureBoot disabled

if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking if build path '/home/lars/Desktop/input-wacom-0.46.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/4.15.0-123-generic/build
checking kernel version... 4.15.0-123-generic
checking RHEL6 minor release... 0
checking RHEL7 minor release... 0
checking power supply version... v4.1+
checking abs accessor version... v2.6.36+
checking CONFIG_MODULE_SIG... y
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... y
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... off
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha512
checking for module signing hash algorithm... sha512
checking for kernel autogenerated key at /lib/modules/4.15.0-123-generic/build/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/4.15.0-123-generic/build/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... yes
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... yes
checking for module signing key... /var/lib/shim-signed/mok/MOK.priv
checking for module signing certificate... /var/lib/shim-signed/mok/MOK.der

[... 33 lines cut ...]
make[2]: Leaving directory '/home/lars/Desktop/input-wacom-0.46.0/4.5'
make[2]: Entering directory '/home/lars/Desktop/input-wacom-0.46.0'
make[2]: Leaving directory '/home/lars/Desktop/input-wacom-0.46.0'
make[1]: Leaving directory '/home/lars/Desktop/input-wacom-0.46.0'
[sudo] password for lars: 
Making install in 4.5
make[1]: Entering directory '/home/lars/Desktop/input-wacom-0.46.0/4.5'
make -C /lib/modules/4.15.0-123-generic/build M=/home/lars/Desktop/input-wacom-0.46.0/4.5 modules_install mod_sign_cmd='"/lib/modules/4.15.0-123-generic/build/scripts/sign-file" "sha512" "/var/lib/shim-signed/mok/MOK.priv" "/var/lib/shim-signed/mok/MOK.der"'
make[2]: Entering directory '/usr/src/linux-headers-4.15.0-123-generic'
  INSTALL /home/lars/Desktop/input-wacom-0.46.0/4.5/wacom.ko
  INSTALL /home/lars/Desktop/input-wacom-0.46.0/4.5/wacom_w8001.ko
  DEPMOD  4.15.0-123-generic
make[2]: Leaving directory '/usr/src/linux-headers-4.15.0-123-generic'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin" depmod -a 4.15.0-123-generic
/usr/sbin/update-initramfs -u -k 4.15.0-123-generic
update-initramfs: Generating /boot/initrd.img-4.15.0-123-generic
make[1]: Leaving directory '/home/lars/Desktop/input-wacom-0.46.0/4.5'
make[1]: Entering directory '/home/lars/Desktop/input-wacom-0.46.0'
make[2]: Entering directory '/home/lars/Desktop/input-wacom-0.46.0'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/lars/Desktop/input-wacom-0.46.0'
make[1]: Leaving directory '/home/lars/Desktop/input-wacom-0.46.0'

LKreutzer, 2020-11-11:
I got the tablet working using the 0.45 version of the driver! I still have no idea why .46 stopped working.

jigpu, 2020-12-02:
One of the more often cause of this "Exec format error" after an otherwise-successful compile and install is multiple kernel sources. If your system had updated in the background and was waiting for a reboot its likely that input-wacom would have been compiled against the old/wrong kernel. I would recommend trying to fully update your system (e.g. `apt update && apt upgrade` for Ubuntu-derivatives), rebooting, and then compiling/installing input-wacom. If modprobe still fails with the "Exec format error", please attach the full output from `dmesg` since it should have additional detailed information from modprobe.

adriansarno, 2021-02-22:
I had a similar problem, sometime around Jan my  "Wacom One Pen Display 13" stopped working after an Ubuntu update (I run Ubuntu 18.04.5). My input driver was version .46, and I got it working by installing the new build version 0.47. Thanks!

closed 2021-02-23 by jigpu