linuxwacom/input-wacom #207: Wacom One CTL-672 not detected in Ubuntu 18.04

arunshal, 2020-09-15:
Hi, I had followed the procedure mentioned here (https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) and I was able to get it to work once. But after restart, it failed to detect the tablet again. I tried re-installing the by running those build commands again. But I am no longer able to get it to work. The tablet is detected by ubuntu. But when I run" dmesg", it gives the following output.
[ 1340.971441] usb 1-1: USB disconnect, device number 8
[ 1828.175052] usb 1-1: new full-speed USB device number 9 using xhci_hcd
[ 1828.327416] usb 1-1: New USB device found, idVendor=056a, idProduct=037b
[ 1828.327421] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1828.327424] usb 1-1: Product: CTL-672
[ 1828.327427] usb 1-1: Manufacturer: Wacom Co.,Ltd.
[ 1828.327429] usb 1-1: SerialNumber: 0GE00M1000100
[ 1828.334483] input: Wacom Co.,Ltd. CTL-672 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:037B.0008/input/input19
[ 1828.334974] hid-generic 0003:056A:037B.0008: input,hiddev0,hidraw1: USB HID v1.10 Mouse [Wacom Co.,Ltd. CTL-672] on usb-0000:00:14.0-1/input0
[ 1828.337455] hid-generic 0003:056A:037B.0009: hiddev1,hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. CTL-672] on usb-0000:00:14.0-1/input1
[ 1828.891365] PKCS#7 signature not signed with a trusted key
[ 1828.891389] wacom: version magic '4.15.0-115-generic SMP mod_unload ' should be '4.15.0-117-generic SMP mod_unload '
[ 1828.892405] PKCS#7 signature not signed with a trusted key
[ 1828.892427] wacom: version magic '4.15.0-115-generic SMP mod_unload ' should be '4.15.0-117-generic SMP mod_unload '
[ 1828.892906] PKCS#7 signature not signed with a trusted key
[ 1828.892944] wacom: version magic '4.15.0-115-generic SMP mod_unload ' should be '4.15.0-117-generic SMP mod_unload '
[ 1828.894252] PKCS#7 signature not signed with a trusted key
[ 1828.894286] wacom: version magic '4.15.0-115-generic SMP mod_unload ' should be '4.15.0-117-generic SMP mod_unload '

Please advice a solution

rjbruin, 2020-09-28:
I have the same problem, but in my case the version magic is: `wacom: version magic '4.15.0-117-generic SMP mod_unload ' should be '4.15.0-118-generic SMP mod_unload '

arunshal, 2020-09-28:
This issue is solved. Everytime the linux kernel get updated, we will have to rebuild/reinstall the driver. 
I uninstalled the driver using following command.
sudo make uninstall
sudo make clean

Then re-installed again using the method described in the wiki.

Thank you...

closed 2020-09-28 by arunshal