linuxwacom/input-wacom #213: Yoga Duet 7 13IML05 touchscreen not detected, help requested

BFH-ktt1, 2020-11-12:
Dear all,
I have since yesterday this machine and try desperately to get the touchscreen working (under Ubuntu 20.04LTS).
What I did so far:
1) Identified the wacom tablet, it is a i2c wacom device, and is "kind of" detected as it shows up under:
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WACF2200:00
However the wacom kernel module does not detect the tablet.
2) compiled the latest input-wacom on git -> no change.
3) Added an udev rule -> no change
4) looked everywhere on the internet for "clues".
I would be very happy if somebody could point me in a direction how to get this working (I am even willing to do some driver development, although I'm new in that field).

jigpu, 2020-11-20:
Hi, we've gotten a few reports from users with these WACF2200 devices (see [here](https://github.com/linuxwacom/wacom-hid-descriptors/issues/101), [here](https://github.com/linuxwacom/wacom-hid-descriptors/issues/103), and [here](https://github.com/linuxwacom/wacom-hid-descriptors/issues/107)). My best guess at the moment is an issue with the ACPI tables, which you would need to provide:

Please download my [copy-acpi-tables.sh script](https://gist.githubusercontent.com/jigpu/b16aed03c074524e9ff08f6276b959e3/raw/0edc8e49ed8363fb8dc6422c3866e780a4bebbc8/copy-acpi-tables.sh) and then run `sudo ./copy-acpi-tables.sh` to save a copy of the tables. After you run it, attach the `acpi-tables.tar.gz` tarball that it generates to this issue for review.

BFH-ktt1, 2020-11-21:
[acpi-tables.tar.gz](https://github.com/linuxwacom/input-wacom/files/5577701/acpi-tables.tar.gz)
@jigpu : Thanks a lot for your reaction, please find attached the requested file

Alexfebbraio, 2020-11-25:
hi, i have a lenovo ideapad flex 5 i3 10gen and have the same problem
if it helps this is my tarbal generated by the script
[acpi-tables.tar.gz](https://github.com/linuxwacom/input-wacom/files/5598586/acpi-tables.tar.gz)

kreecm, 2020-11-30:
I'm not sure if it is best for me to just add onto this issue or create a new one. But I just got a Lenovo Yoga 9i, and did a clean install of Ubuntu 20.10. Most everything is working except the touchscreen. The Lenovo model names are super confusing, so I'm not even sure how different this laptop is from the OP's or previous commentors. In any case here is the copy of my ACPI tables: 
[acpi-tables.tar.gz](https://github.com/linuxwacom/input-wacom/files/5613770/acpi-tables.tar.gz)

jigpu, 2020-11-30:
Thanks all, I'll be reviewing the ACPI tables and seeing what I can find.

@kreecm, I would recommend opening a new issue for your particular device since it seems to be different from the Yoga Duet 7 13IML05. In addition to the ACPI table you've already provided, be sure to also attach the output from [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to provide baseline info for your particular device.

jigpu, 2020-11-30:
EDIT: To be clear @kreecm and @Alexfebbraio, please make these new issues with the sysinfo/acpi output in our [wacom-hid-descriptors issue tracker](https://github.com/linuxwacom/wacom-hid-descriptors/issues). This is mostly to ensure I have a proper baseline for your two devices.

FYI, I've opened a new centralized issue specifically for the WACF sensor detection issues at https://github.com/linuxwacom/input-wacom/issues/218. I'm going to close this specific issue for the time being.

closed 2020-11-30 by jigpu

jigpu, 2021-01-25:
Now that #218 has been fixed upstream, I would appreciate if @BFH-ktt1 and @kreecm could update to one of the required kernels and submit updated sysinfo tarballs (@Alexfebbraio -- thanks for your updated submission in the other thread :)). You should be able to see the pen show up in `xsetwacom list`, though I would be careful about actually trying to use the pen until #230 is fixed.

Alexfebbraio, 2021-01-27:
> Now that #218 has been fixed upstream, I would appreciate if @BFH-ktt1 and @kreecm could update to one of the required kernels and submit updated sysinfo tarballs (@Alexfebbraio -- thanks for your [...]

yes

Wacom HID 521A Pen stylus       	id: 15	type: STYLUS    
Wacom HID 521A Finger touch     	id: 16	type: TOUCH     
Wacom HID 521A Pen eraser       	id: 20	type: ERASER

jigpu, 2021-02-10:
Any chance of attaching the output from the [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) so that I can put together a libwacom tablet definition?