linuxwacom/input-wacom #215: Wacom Cintiq 27QHDT with static screen and stylus undetected unless USB is reconnected

nylnook, 2020-11-15:
Hi dear Linux Wacom team,

I use a Wacom Cintiq 27QHDT since several years on Linux, and I updated my OS to Ubuntu 20.04.1 this summer, with Gnome 3.36.3. I use X11, not Wayland. The tablet is working, but it's not ideal: 
 - At boot, if I switch on the screen/tablet after the computer, it doesn't detect anything, so the screen is blank and goes into standby mode, and I have to do an hard reset on the computer.
 - After startup, my stylus is undetected, not reacting. I used to switch on an off the tablet to make it work, but since a recent update, the screen doesn't detect anything and goes into standby mode after that, so I have to do an hard reset on the computer again.
 - So the only way to make the tablet + the stylus work is to disconnect and reconnect the USB cable of the tablet, so the stylus is detected again. I tried all my USB ports (3.1, 3.0, 2.0), and it's always the same.

I tried to reinstall the wacom packages, but no success.

my lsusb:
`:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 007: ID 056a:0337 Wacom Co., Ltd 
Bus 002 Device 006: ID 056a:0337 Wacom Co., Ltd 
Bus 002 Device 005: ID 056a:032d Wacom Co., Ltd Cintiq 27QHDT Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth
Bus 001 Device 018: ID 056a:032c Wacom Co., Ltd DTH-2700 [Cintiq 27QHD touch] touchscreen
Bus 001 Device 017: ID 0403:6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC
Bus 001 Device 016: ID 056a:0331 Wacom Co., Ltd ACK-411050 [ExpressKey Remote]
Bus 001 Device 015: ID 056a:032b Wacom Co., Ltd DTH-2700 [Cintiq 27QHD touch] tablet
Bus 001 Device 014: ID 056a:0338 Wacom Co., Ltd 
Bus 001 Device 013: ID 056a:0338 Wacom Co., Ltd 
Bus 001 Device 012: ID 056a:032e Wacom Co., Ltd Cintiq 27QHDT Hub
Bus 001 Device 007: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 005: ID 045e:00db Microsoft Corp. Natural Ergonomic Keyboard 4000 V1.0
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

`:~$ grep "" /sys/module/wacom*/version
v2.00
`
`:~$ xsetwacom --list devices
Wacom Cintiq 27QHD touch Pen stylus	id: 13	type: STYLUS    
Wacom Cintiq 27QHD touch Pen eraser	id: 14	type: ERASER    
Wacom Express Key Remote Pad pad	id: 15	type: PAD       
Wacom Cintiq 27QHD touch Finger touch	id: 18	type: TOUCH `

I also use a xsetwacom script to use the Expresskey Remote, waiting for [this](https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/415) to land in Ubuntu.

Any idea how to fix that ?
Thanks a lot in advance !

jigpu, 2020-11-20:
This issue with the display that you are seeing is new to me. It sounds like an issue with the graphics driver, however. Do you mind running our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attaching the tarball it generates? It should contain information about what card, driver, and driver version you're using. I'll see if I can connect you with someone who can help based on that output...

As for the tablet detection issue, that sounds like what was reported in #104. We don't have a fix for the issue, but there are a few workarounds mentioned in the thread that seem to reset the connection without having to pull the USB plug. Hopefully that helps in the meantime while you hunt down the graphics issue.

nylnook, 2020-11-21:
HI Jason, and thank you for your ongoing support since years !

Here is the sysinfo:
[sysinfo.XSpQ34xk1S.tar.gz](https://github.com/linuxwacom/input-wacom/files/5577565/sysinfo.XSpQ34xk1S.tar.gz)

And your script in #104 is working, that's a great first step ;)

Have a nice week-end !
Camille

Pinglinux, 2022-07-25:
If it's not a driver issue, we can't do much...

closed 2022-07-25 by Pinglinux