linuxwacom/input-wacom #217: Lenovo X1 Fold - touchscreen not working

fluffypony, 2020-11-28:
I know the X1 Fold is pretty new, so I wasn't that surprised that some stuff wasn't working, including the touchscreen. It's apparently a Wacom touchscreen. In the Windows device manager the HID device for the touch screen is vendor 8086 (Intel) and device 98D0 (which doesn't appear in DeviceHunt or anywhere else I've looked). In the "Intel Precise Touch and Stylus" device, however, there is a "WCOM5102" entry under hardware IDs (below the Intel vendor / device hardware IDs) so I'm fairly certain that the solution might lie in ```input-wacom``` support for this.

I have built and installed 0.46.0 to no avail, and then built and installed head from the git repo. I haven't bothered updating ```xf86-input-wacom``` as the kernel module doesn't appear to be finding the touchscreen yet (unless I'm being dumb and not looking in the right place). I am running ```xserver-xorg-hwe-18.04```, but I figure I'd see something in dmesg if the kernel module picked it up.

Out of curiosity I also live-booted Ubuntu 20.10, but that didn't work either.

jigpu, 2020-11-30:
There's been a rash of issues lately with Lenovo sensors not showing up in Linux. It doesn't look like we have any baseline data on this particular device, would you mind first opening an issue at https://github.com/linuxwacom/wacom-hid-descriptors/issues with the output from [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) attached? This will give us some of that baseline data necessary to debug further.

fluffypony, 2020-11-30:
Thanks @jigpu - [opened an issue and attached the sysinfo output](https://github.com/linuxwacom/wacom-hid-descriptors/issues/114)

jigpu, 2020-12-01:
Perfect, thanks :) For better or worse, your touchscreen doesn't seem to be exhibiting the same issue as all the "WACF2200" device that have recently shown up. There's no hint of a touchscreen in your output, so some additional investigation is needed...

Just to be 100% sure that your device doesn't have a WACF2200, would you mind re-running the sysinfo script from inside an Ubuntu 20.04 live image? I don't expect the touchscreen to work in Ubuntu right now, but I do want to double-check the device list to see if there's anything Ubuntu sees that your current distro doesn't.

fluffypony, 2020-12-01:
Sure thing, will give that a spin shortly. I also realised that when I upgraded the kernel to 5.10-rc6, I forgot to rebuild input-wacom (which is actually a little more painful on rc6 because it doesn't have module.lds). I've re-run sysinfo.sh on 5.10-rc6 running input-wacom v2.00-0.46.0.2.g73efa2c, [uploaded here](https://github.com/linuxwacom/input-wacom/files/5620664/sysinfo.jDLyXX0UNF.tar.gz).

I'll re-run it on Ubuntu 20.10 momentarily!

fluffypony, 2020-12-01:
Alright, ran the script on Ubuntu 20.10, [and uploaded](https://github.com/linuxwacom/input-wacom/files/5620758/sysinfo.yfJYFpQKYC.tar.gz)

jigpu, 2020-12-01:
Thanks again. I've checked through the new sysinfo files and see the same thing. No hint of any touchscreen or pen hardware at all. Most Wacom sensors either "just work" or at the very least are detectable by the kernel, so something else is going on here. Its possible either the sysinfo script doesn't recognize whoever manufactured the sensor, or that something is preventing the kernel from detecting the sensor's presence.

If you still have Windows installed, I would do some digging around inside the Device Manager to learn more about the sensor. If you can find its entry, there's some hope at figuring out what driver "should" handle it in Linux and why it doesn't work.

fluffypony, 2020-12-01:
Did a bit of digging, this is what I can find:

[image]

[image]

[image]

[image]

Lenovo's marketing material says it's a Wacom touch screen, and the pen is apparently a Wacom Active pen. I have no idea why it's using an Intel device driver for it, but there are some hints that it's a Wacom device in the various sub-settings.

fluffypony, 2020-12-01:
Also vendor 8086 is Intel, but nobody seems to have device 98D0, so maybe this is just a matter of extremely new hardware?

jigpu, 2020-12-01:
Thanks for the screenshots. The "Intel Precise Touch and Stylus" (IPTS) is bad news. I don't have much experience with those, but from my (limited) understanding it designates a device which offloads much of the hardware pen/touch processing to the GPU. It requires additional software to set up that GPU processing and create a virtual input device. Until today I thought IPTS was exclusive to Microsoft Surface and their in-house sensors...

Its been a while since I last looked, but there does seem to now be a Linux IPTS kernel driver: https://github.com/linux-surface/intel-precise-touch In addition to the driver you'll apparently need to also install a daemon mentioned in its documentation. There's a reasonable chance that some additional work will need to be done in the driver or daemon to support the Wacom sensor since up till now its only been Microsoft sensors...

StollD, 2020-12-02:
Hi, I'm the main contributor to the linux-surface IPTS stack (the kernel driver and the userspace daemon).

Feel free to open an issue about the X1 Fold on the repository for the kernel driver (linked above), and I will see what can be done to get the touchscreen to work. 

> Until today I thought IPTS was exclusive to Microsoft Surface and their in-house sensors...

There is actually one HP device that uses IPTS (HP Spectre 13 x2), but I dont think anyone has every seriously tried to get linux working on it.

nodefourtytwo, 2021-05-28:
I don't have a Fold yet, but I'm interested, did somebody make it work yet? I can't find the issue on the IPTS repo, did somebody open it?