linuxwacom/input-wacom #218: WACF2200 sensors in new Lenovo tablet PCs are not detected by Wacom driver

jigpu, 2020-11-30:
It seems that many recent Lenovo devices are not having their touchscreen / pen sensor detected by the Wacom driver. The common thread so far seems to be the use of the WACF2200 ACPI ID in these devices. This issue will centralize the multiple bug reports that have been flowing in.

Affected devices:
* Lenovo Yoga C640-13IML (@PunkIslamist, @richibrics): https://github.com/linuxwacom/wacom-hid-descriptors/issues/101
* Lenovo IdeaPad Flex 5 14IIL05 (@cfgnunes, @neonfighter28): https://github.com/linuxwacom/wacom-hid-descriptors/issues/103
* Lenovo YOGA Duet 2020 (@liuqun): https://github.com/linuxwacom/wacom-hid-descriptors/issues/107
* Lenovo Idea Pad Flex 5 14ARE05 (@Weewoolad): https://github.com/linuxwacom/wacom-hid-descriptors/issues/112
* Lenovo Yoga Duet 7 13IML05 (@BFH-ktt1): https://github.com/linuxwacom/input-wacom/issues/213
* Lenovo Flex 5 i3 10gen (@Alexfebbraio): https://github.com/linuxwacom/input-wacom/issues/213
* Lenovo Yoga 9i: (@kreecm): https://github.com/linuxwacom/input-wacom/issues/213

Notes:
* Reported to be "partially working" on the 14ARE05 by @Weewoolad (sysinfo.fGWhYaLpPe; Lubuntu 20.10, Linux 5.8.0-29-generic)

Hypotheses:
* ACPI issue? (no obvious ACPI issues discovered so far)
* 8250 "serial" driver taking control of all "WACF" devices?

jigpu, 2020-11-30:
@Weewoolad, you said:
> Partially working on Lubuntu with kernel 5.8.0-29. Attempted and failed with Void Linux with kernel 5.8 (no detection). Attempted and failed with Lubuntu with kernel 5.10 (no detection).
Can you give me a little more information about the kernel 5.8 versus 5.10 installs of Lubuntu? Were they both Lubuntu 20.04? Were you running Lubuntu-provided kernels both times, or did you get them from some other source (or built yourself)?

jigpu, 2020-11-30:
I haven't been able to find anything suspicious in the ACPI tables so far, unfortunately. I've continued digging and am now growing suspicious of the kernel's 8250 serial port driver. Back in the ancient days there were several Wacom sensors which connected to the system via an (internal) serial port. It looks like the kernel's serial port driver has a wildcard match for any "WACF" device, which could cause problems for these new "WACF2200" devices that are actually connected over I2C. If anyone is comfortable with compiling and testing custom kernels, try using a version of the kernel with your current configuration (usually found at `/proc/config.gz`) but with the following patch applied:
~~~
diff --git a/drivers/tty/serial/8250/8250_pnp.c b/drivers/tty/serial/8250/8250_pnp.c
index de90d681b64c..b1536d564f09 100644
--- a/drivers/tty/serial/8250/8250_pnp.c
+++ b/drivers/tty/serial/8250/8250_pnp.c
@@ -331,7 +331,7 @@ static const struct pnp_device_id pnp_dev_table[] = {
 	/* U.S. Robotics 56K Voice INT PnP*/
 	{	"USR9190",		0	},
 	/* Wacom tablets */
-	{	"WACFXXX",		0	},
+//	{	"WACFXXX",		0	},
 	/* Compaq touchscreen */
 	{       "FPI2002",              0 },
 	/* Fujitsu Stylistic touchscreens */
~~~

BFH-ktt1, 2020-11-30:
@jigpu , Thanks for all your effort! I would be happy to build a custom kernel, but unfortunately the coming three weeks I am still "swamped" with preparing on-line exams, etc. due to the Covid situation. I would have some time in three weeks (during the Xmas holiday period). If nobody can test it before than I'll be happy to dig into it in three weeks.

richibrics, 2020-11-30:
> I haven't been able to find anything suspicious in the ACPI tables so far, unfortunately. I've continued digging and am now growing suspicious of the kernel's 8250 serial port driver. Back in the [...]

Then I have to comment the WACFXXXX line right ? Like this
From 
```
       **Beginning of 8250_pnp.c**
	/* U.S. Robotics 56K Voice INT PnP*/
	{	"USR9190",		0	},
	/* Wacom tablets */
	{	"WACFXXX",		0	},
	/* Compaq touchscreen */
	{       "FPI2002",              0 },
	/* Fujitsu Stylistic touchscreens */
       **End of 8250_pnp.c**
```

To
```
       **Beginning of 8250_pnp.c**
	/* U.S. Robotics 56K Voice INT PnP*/
	{	"USR9190",		0	},
	/* Wacom tablets */
//	{	"WACFXXX",		0	},
	/* Compaq touchscreen */
	{       "FPI2002",              0 },
	/* Fujitsu Stylistic touchscreens */
       **End of 8250_pnp.c**
```

Weewoolad, 2020-11-30:
> @Weewoolad, you said:
>
>
> Can you give me a little more information about the kernel 5.8 versus 5.10 installs of Lubuntu? Were they both Lubuntu 20.04? Were you running Lubuntu-provided kernels both [...]

Yes, the kernel change was done on the same install of lubuntu 20.10. I used kernel 5.10rc5 from kernel.ubuntu-ppa/mainline. The 5.8 kernels (5.8.0-25 and 5.8.0-29) came from kubuntu's installer. The live cd of kubuntu 20.10 works as well.

jigpu, 2020-12-01:
@richibrics -- that's correct

@Weewoolad -- thanks, I'll follow up on those leads!

Weewoolad, 2020-12-01:
@jigpu 

I initially said lubuntu, but it's actually kubuntu. I apologise for the confusion.

BFH-ktt1, 2020-12-01:
@jigpu , I had some spare minutes and tried 2 things on my Yoga Duet:
I compiled a new kernel (5.7.5) with the provided patch: no change.
I tried the Kubuntu 20.10 live cd (kernel 5.8.0-29): same as with Ubuntu 20.04 LTS.
What Information would you require to continue your search?

jigpu, 2020-12-01:
Thanks @BFH-ktt1. This morning I also a look at the Kubuntu 20.10 kernel config (kernel 5.8.0-25) and see that it has the 8250 drivers enabled. Its looking less and less like the 8250 wildcard is at fault. Kubuntu 20.10 working for @Weewoolad but not you is also especially interesting since that implies some kind of hardware or firwmare difference. I'll be making a second comparison of the logs you two have gathered to see if anything stands out...

richibrics, 2020-12-01:
I try Kubuntu and let you know

richibrics, 2020-12-01:
Does not work on Kubuntu with kernel 5.8.0-29-generic both installed or live

neonfighter28, 2020-12-01:
@jigpu I compiled a custom kernel (5.9.11) with your patch, but I had no success in getting anything related to the touchscreen to work. Do you need any outputs/logs? I will happily provide them, just let me know! Also, would you like me to try out the partially working Kubuntu?

jigpu, 2020-12-01:
@neonfighter28 thanks for the report; no logs necessary. A test of Kubuntu would be appreciated though

Alexfebbraio, 2020-12-01:
> I try Kubuntu and let you know

I try Without success with:
Ubuntu 20.10
Kubuntu 20.10
Lubuntu 20.10
Fedora 33
Deepin 20
Manjaro gnome 20.1.2
Manjaro kde 20.1.2
Opensuse15.2
CromeOS branch kernel 4.4
ChromeOS branch next kernel 5.4

neonfighter28, 2020-12-01:
@jigpu No success on Kubuntu 5.8.0-25, `xinput --list` shows the same as on arch

Alexfebbraio, 2020-12-02:
> > I try Kubuntu and let you know
> 
> I try Without success with:
> Ubuntu 20.10
> Kubuntu 20.10
> Lubuntu 20.10
> Fedora 33
> Deepin 20
> Manjaro gnome 20.1.2
> Manjaro kde 20.1.2
> Opensuse15.2
> ChromeOS branch kernel 4.4
> ChromeOS branch next kernel 5.4

Today i installed the 5.10 rc6 kernel on ubuntu 20.10 but the result is always the same

Alexfebbraio, 2020-12-12:
nothing news?

Donald4444, 2020-12-13:
I did some reasearch, because I'm going to get a Lenovo IdeaPad Flex 5 to christmas and found out something interesting. In the Archwiki under discussion of the tablet is somebody who wrote, that the touch is working (At least with a pen).
His output of ``dmesg|grep wacom``
```
[    5.155680] input: Wacom HID 5218 Pen as /devices/platform/AMDI0010:00/i2c-0/i2c-WACF2200:00/0018:056A:5218.0001/input/input25
[    5.155794] input: Wacom HID 5218 Finger as /devices/platform/AMDI0010:00/i2c-0/i2c-WACF2200:00/0018:056A:5218.0001/input/input26
[    5.155884] wacom 0018:056A:5218.0001: hidraw0: I2C HID v1.00 Mouse [WACF2200:00 056A:5218] on i2c-WACF2200:00
```
Link: [https://wiki.archlinux.org/index.php/Talk:Lenovo_IdeaPad_Flex_5_(Ryzen)](url)

Alexfebbraio, 2020-12-13:
> I did some reasearch, because I'm going to get a Lenovo IdeaPad Flex 5 to christmas and found out something interesting. In the Archwiki under discussion of the tablet is somebody who wrote, that [...]

Flex 5 Ryzen it doesn't have the wacom 2200. It is different from flex 5 i3 10gen

Donald4444, 2020-12-13:
Are you sure?
The Path is ``/devices/platform/AMDI0010:00/i2c-0/i2c-WACF2200:00/0018:056A:5218.0001/input/input26``
which includes ``WACF2200`` and as far as I followed this issue this is the sensor. Also on top it says, that the ``Lenovo Idea Pad Flex 5 14ARE05`` has this problem, even though it's a amd.
To the naming:
 ARE = amd and IIL = intel as far as I saw it

Alexfebbraio, 2020-12-13:
> Are you sure?
> The Path is `/devices/platform/AMDI0010:00/i2c-0/i2c-WACF2200:00/0018:056A:5218.0001/input/input26`
> which includes `WACF2200` and as far as I followed this issue this is the sensor. [...]

I'm sorry, you're right.  I got confused with ideapad 5 amd which uses elan's touchscreen

Alexfebbraio, 2020-12-13:
> [https://wiki.archlinux.org/index.php/Talk:Lenovo

> Are you sure?
> The Path is `/devices/platform/AMDI0010:00/i2c-0/i2c-WACF2200:00/0018:056A:5218.0001/input/input26`
> which includes `WACF2200` and as far as I followed this issue this is the sensor. [...]

> I did some reasearch, because I'm going to get a Lenovo IdeaPad Flex 5 to christmas and found out something interesting. In the Archwiki under discussion of the tablet is somebody who wrote, that [...]

the owner uses manjaro 20.0.1 with the 5.6.12-1-MANJARO kernel

This is his hd-probe:
https://linux-hardware.org/index.php?probe=cd3c373e02

ccontrer, 2020-12-14:
I have an IdeaPad Flex 5 Rayzen, the issue with the touchscreen seems to be solved for me with kernel `5.10.0-051000-generic` on Ubuntu 20.04.

Alexfebbraio, 2020-12-14:
> I have an IdeaPad Flex 5 Rayzen, the issue with the touchscreen seems to be solved for me with kernel `5.10.0-051000-generic` on Ubuntu 20.04.

I just installed kernel 5.10.0-051000-generic on Ubuntu as you did but, no touchscreen recognition :-(

ccontrer, 2020-12-14:
Interesting...

> > I have an IdeaPad Flex 5 Rayzen, the issue with the touchscreen seems to be solved for me with kernel `5.10.0-051000-generic` on Ubuntu 20.04.
> 
> I just installed kernel 5.10.0-051000-generic on Ubuntu as you did but, no touchscreen recognition :-(

Interesting, my outputs.
- `dmesg | grep wacom`
```
[    2.337434] wacom 0018:056A:5218.0002: hidraw1: I2C HID v1.00 Mouse [WACF2200:00 056A:5218] on i2c-WACF2200:00
```

- `lsinput`
```
/dev/input/event8
   bustype : BUS_I2C
   vendor  : 0x56a
   product : 0x5218
   version : 256
   name    : "Wacom HID 5218 Pen"
   phys    : "i2c-WACF2200:00"
   uniq    : ""
   bits ev : (null) (null) (null) (null)

/dev/input/event9
   bustype : BUS_I2C
   vendor  : 0x56a
   product : 0x5218
   version : 256
   name    : "Wacom HID 5218 Finger"
   phys    : "i2c-WACF2200:00"
   uniq    : ""
   bits ev : (null) (null) (null)
```

- I get readings with `sudo mtdev-test /dev/input/event9` using my finger. I don't have a pen.

- I don't have the `usbtouchscreen` module installed.

Alexfebbraio, 2020-12-14:
This is my hw-probe with Ubuntu with your kernel 5.10.0-051000-generic:

https://linux-hardware.org/?probe=c046829816

ccontrer, 2020-12-14:
This is my hw-probe

https://linux-hardware.org/?probe=2ed404ea1d

Alexfebbraio, 2020-12-14:
> This is my hw-probe
> 
> https://linux-hardware.org/?probe=2ed404ea1d

it's a mistery.  I don't know what to think / do anymore. 
the touchscreen with windows 10 works so I rule out a hardware problem in my case

Donald4444, 2020-12-14:
I found some differences in your configuration files that could cause the issue. @ccontrer has multiple files in his xorg.conf.d folder, including some things for wacom and touchscreens. You can also see, that in @Alexfebbraio PC the xorg log isn't even something logged related to wacom.
Also has @ccontrer the wacom module loaded. You could try loading it with ``sudo modprobe wacom``. (You can unload it again with ``sudo modprobe -r wacom`` but on my desktop-PC I had to replug the keyboard to get it and the mouse work again) It is also noteworthy, that in general @ccontrer has other modules loaded under hid.

Comparision Link (dmesg):
@ccontrer (working):
https://linux-hardware.org/index.php?probe=2ed404ea1d&log=lsmod
@Alexfebbraio (not working):
https://linux-hardware.org/index.php?probe=c046829816&log=lsmod
And I found that @ccontrer has some modules blacklisted. [(modprobe.d)](https://linux-hardware.org/index.php?probe=2ed404ea1d&log=modprobe.d)
So my guess would be, that some other module takes over the tablet and then fails, but I'm not sure. It could also help to see, if there is something listed under ``/sys/devices/platform/???/i2c-???``, because in @ccontrer's dmesg you can see, that the wacom device is connected via i2c. (Link to the [dmesg](https://linux-hardware.org/index.php?probe=2ed404ea1d&log=dmesg))

ftsiadimos, 2020-12-15:
It seems the latest kernel for  Ideapad Flex 5 14ARE05 can detect the hardware. 

For the below model it can't even detect the hardware correctly.  
IdeaPad Flex 5 14IIL05
https://linux-hardware.org/?probe=634978913d

Alexfebbraio, 2020-12-15:
> It seems the latest kernel for Ideapad Flex 5 14ARE05 can detect the hardware.
> 
> For the below model it can't even detect the hardware correctly.
> IdeaPad Flex 5 14IIL05
> https://linux-hardware.org/?probe=634978913d

exactly.  I wonder why?

Donald4444, 2020-12-16:
I think it is a good to have a live-image with that we can find out, if the Wacom driver is behaving differently on the Amd and Intel tablet PCs, with exactly same software.
I made a repository for that (https://github.com/Donald4444/wacf2200-iso) where you can build the Iso by yourself (just run the generate.sh script in the folder (needs the archiso package on Archlinux)) or you can download the ISO at the releases (https://github.com/Donald4444/wacf2200-iso/releases/tag/kernel-5-9-14). Sadly only the 5.9.14 kernel is currently in the not-testing repository. (I will update the repo, when the 5.10 kernel is in the offical repositorys).
To be sure, that every generation of a ISO has the same packages I configured pacman to use the archlive server, where you can specify a date, from when the packages should be downloaded.
Note that this ISO doesn't need an installation and is a live system.

Usage: When booting up you should see the sddm login manager. You can select on the top the session (including i3, sway, plasma (Xorg) and plasma (Wayland)). Then you select the user ``user`` and press enter (there is no password set).
The important keybindings for i3 and sway are:
``Super+Enter``: Terminal
``Super+E``: Exit
``Super+T``: Dolphin (File Manager)
``Super+D``: Launcher
You can find the full config in ``{kernel-version}/archlive/airootfs/etc/skel/.config/{i3 or sway}/config`` in the repository or in the live-system with ``cat ~/.config/{i3 or sway}/config``.

I think this could help getting clearance, if the Amd and Intel have a bigger difference in some hardware part (e.g. the connection to the Wacom sensor).
I think the most important logs would be ``lsmod``, ``dmesg``, Xorg (``/var/log/Xorg.0.log`` and ``~/.local/share/xorg/Xorg.0.log``), ``xsetwacom list devices``, ``xinput list`` and all found devices (``/proc/bus/input/devices``).
To test the pen I also included xournalpp, gimp and krita.

Because I don't know if this ISO is generally working with the tablet PCs I would ask @ccontrer to test it first and write, if the touchscreen is working in general, because it seems that the kernel can detect the sensor on his tablet PC.

I hope this will help finding out, if the Amd and Intel Systems behave different on the exactly same Kernel/Software.

ccontrer, 2020-12-16:
Hi @Donald4444 , the touchscreen is detected and working very well with the ISO [`kernel-5-9-14`](https://github.com/Donald4444/wacf2200-iso/releases/tag/kernel-5-9-14) (except if I connect an external monitor), including scrolling and two-finger gestures. I don't have a pen.

Here are the logs: 
[lsmod](https://github.com/linuxwacom/input-wacom/files/5704456/lsmod.txt)
[dmesg](https://github.com/linuxwacom/input-wacom/files/5704458/dmesg.txt)
[varXorg](https://github.com/linuxwacom/input-wacom/files/5704459/varXorg.txt)
I couldn't find folder `.local/share/xorg/`
[xsetwacom](https://github.com/linuxwacom/input-wacom/files/5704461/xsetwacom.txt)
[xinput](https://github.com/linuxwacom/input-wacom/files/5704462/xinput.txt)
[devices](https://github.com/linuxwacom/input-wacom/files/5704464/devices.txt)

Let me know.

PD: `Super+E` didn't work.

Donald4444, 2020-12-16:
Ok. So I think if the ISO works out of the box with the tablet PC on Amd, it should also work out of the box with intel. I would ask @Alexfebbraio to try it out. (It would be also nice, if you could include the logs).

If it works with the ISO it seems that there is something weird with the config files in Ubuntu. If it doesn't work I would except something weird in the Hardware or BIOS/UEFI settings. (Or maybe a problem with the connection driver (probably I2C))

BFH-ktt1, 2020-12-16:
@Donald4444 , First of all, thanks for your work! Than: I tried the .iso on my Yoga Duet (intel based) and no detection of the touch-screen, Hope this helps.

Donald4444, 2020-12-16:
@BFH-ktt1 can you attach some logs?
I think it could help to see, if the kernel has some sort of error (e.g. i2c-problems). Thanks in advice

Alexfebbraio, 2020-12-16:
Thanks also from me.
On the fly (I'm cooking vegetables au gratin in the oven), I downloaded the iso and started it on my i3 10gen.
Unfortunately the wacom hardware is not detected.
As soon as I can, I post my logs

Donald4444, 2020-12-16:
Ok. Can you also post the output of ``ls /sys/devices/platform/*/i2c*/``? (It could also return ``zsh: no matches found: /sys/devices/platform/*/i2c*/``)

neonfighter28, 2020-12-16:
@Donald4444 

[devices.txt](https://github.com/linuxwacom/input-wacom/files/5705068/devices.txt)
[dmesg.txt](https://github.com/linuxwacom/input-wacom/files/5705070/dmesg.txt)
[lsmod.txt](https://github.com/linuxwacom/input-wacom/files/5705076/lsmod.txt)
[xinput.txt](https://github.com/linuxwacom/input-wacom/files/5705075/xinput.txt)
[var_xorg.txt](https://github.com/linuxwacom/input-wacom/files/5705074/var_xorg.txt)

here are my logs. 
`ls /sys/devices/platform/*/i2c*/` returns nothing

BFH-ktt1, 2020-12-16:
@Donald4444 , of course, will post it first thing tomorrow morning.

Alexfebbraio, 2020-12-16:
This are my log:
[dmesg.log](https://github.com/linuxwacom/input-wacom/files/5705239/dmesg.log)
[lsmod.log](https://github.com/linuxwacom/input-wacom/files/5705241/lsmod.log)
[xinput.log](https://github.com/linuxwacom/input-wacom/files/5705243/xinput.log)

ls /sys/devices/platform/*/i2c*/ returns nothing

Alexfebbraio, 2020-12-16:
If it helps, this is the msinfo32 log generated by the windows 10 home of my i3 10gen

[msinfo32.log](https://github.com/linuxwacom/input-wacom/files/5705312/msinfo32.log)

Donald4444, 2020-12-16:
Ok. Thank you all. I thought of the possibility that Lenovo swapped out the WACF2200 for some reason but thanks to @Alexfebbraio msinfo32.log I saw that the WACF2200 is definitly used.

I think that there is either a problem with the communication between cpu and the sensor or a problem in the automatic detection.
Can one of you (where the touchscreen isn't working) test, if running ``sudo modprobe wacom usbhid`` lets the touchscreen react or adds some output to ``dmesg`` and ``xinput``?

Also a question to @ccontrer: Did you some sort of firmware upgrade (downloaded something from the lenovo website and installed it (probably under Windows) except drivers)?

neonfighter28, 2020-12-16:
@Donald4444 
[dmesg](https://github.com/linuxwacom/input-wacom/files/5705569/dmesg.txt) this is my `dmesg` after I modprobed the two modules.

`xinput --list` remains the same

thanks for helping us resolve this issue!

ccontrer, 2020-12-16:
> Also a question to @ccontrer: Did you some sort of firmware upgrade (downloaded something from the lenovo website and installed it (probably under Windows) except drivers)?

No, I did not. Just kernel upgrade.

Donald4444, 2020-12-16:
I hope that the things I do help. Sadly I know nearly nothing in C programming and nothing in modifying the linux kernel source code. I just try to find out where the issue is as close as possible, but probably somebody like @jigpu will have to find the issue in the source code itself and make patches. I think currently we would need a kernel which logs, which i2c devices are generally detected, to see if the WACF2200 sensor is generally showing up, because on the current kernel we only get it detected as input or nothing. (See dmesg of @ccontrer (Amd) and dmesg of any other (Intel))

Currently my guess is, that something in the communication on the Intel platform is bad. This would be a problem in the kernel, but I think this is very unlikely. Could you also look in the BIOS/UEFI if there is something related to i2c, wacom or touch?

Sadly I'm currently running out of Ideas, because on multiple seemingly completly identically system (at least software side) it behaves differently and the only two major differences I can find are, that the ``wacom`` (and some other) modules aren't loaded and manually loading them don't work. The other is the platform/hardware (Amd/Intel) and maybe the BIOS/UEFI settings.

BFH-ktt1, 2020-12-17:
[devices.txt](https://github.com/linuxwacom/input-wacom/files/5707790/devices.txt)
[dmesg.txt](https://github.com/linuxwacom/input-wacom/files/5707791/dmesg.txt)
[dmesg_aftermodprobe.txt](https://github.com/linuxwacom/input-wacom/files/5707792/dmesg_aftermodprobe.txt)
[i2c_devices.txt](https://github.com/linuxwacom/input-wacom/files/5707795/i2c_devices.txt)
[lsmod.txt](https://github.com/linuxwacom/input-wacom/files/5707796/lsmod.txt)
[xinput.txt](https://github.com/linuxwacom/input-wacom/files/5707797/xinput.txt)
[Xorg.0.log](https://github.com/linuxwacom/input-wacom/files/5707799/Xorg.0.log)
@Donald4444 , as promised the outputs on my duet, note thet the dmesg_aftermodprobe.txt is the dmesg after perfoeming modprobe wacom. Hope this helps.

Donald4444, 2020-12-17:
Ok. Thank you @BFH-ktt1. I think the ls command I made was wrong. Can some of you try to run ``sudo find /sys/ -iname "*wacf*"`` and if that return's nothing ``sudo find /sys/ -iname "*i2c*"``?
Because from @BFH-ktt1's ``i2c_devices.txt`` output the WACF2200 seems to be detected by the kernel as "it's there, but I don't know what to do with it"

Alexfebbraio, 2020-12-17:
> Ok. Thank you @BFH-ktt1. I think the ls command I made was wrong. Can some of you try to run `sudo find /sys/ -iname "*wacf*"` and if that return's nothing `sudo find /sys/ -iname "*i2c*"`?
> Because [...]

OK
sudo find /sys/ -iname "*wacf*"` 
[wacf.log](https://github.com/linuxwacom/input-wacom/files/5709205/wacf.log)

sudo find /sys/ -iname "*i2c*"`
[i2c.log](https://github.com/linuxwacom/input-wacom/files/5709209/i2c.log)

Donald4444, 2020-12-17:
@ccontrer can you run the command ``sudo find /sys/ -iname "*i2c*"`` too? It seems that the kernel detects the sensor, but the driver doesn't recognize it or something other is causing the driver to fail without giving out an error.
I think with that output we could be rather sure, if everything necessary from the WACF2200 sensor is detected.

ftsiadimos, 2020-12-17:
Flex 5 14IIL05 
5.9.14-200.fc33.x86_64

sudo find /sys/ -iname "*wacf*" > wacf.log
[wacf.log](https://github.com/linuxwacom/input-wacom/files/5710664/wacf.log)

sudo find /sys/ -iname "*i2c*" > i2c.log
[i2c.log](https://github.com/linuxwacom/input-wacom/files/5710666/i2c.log)

BFH-ktt1, 2020-12-17:
@Donald4444 , please find attached my outputs to your commands.
[i2c1.txt](https://github.com/linuxwacom/input-wacom/files/5710691/i2c1.txt)
[i2c2.txt](https://github.com/linuxwacom/input-wacom/files/5710693/i2c2.txt)

Donald4444, 2020-12-17:
Thank you for helping, but currently a log from a working system would be useful to see, if everything from the hardware is detected correctly. If that's the case I would guess, that something is interrupting the driver without causing an exception in the kernel log (dmesg) ...

ccontrer, 2020-12-17:
> @ccontrer can you run the command `sudo find /sys/ -iname "*i2c*"` too? It seems that the kernel detects the sensor, but the driver doesn't recognize it or something other is causing the driver to [...]

Sure. For `5.10.0-051000-generic`

[i2c.log](https://github.com/linuxwacom/input-wacom/files/5710705/i2c.log)

Thank you for doing this @Donald4444

Alexfebbraio, 2020-12-17:
> Thank you for helping, but currently a log from a working system would be useful to see, if everything from the hardware is detected correctly. If that's the case I would guess, that something is interrupting the driver without causing an exception in the kernel log (dmesg) ...

What kind of system logs can I produce?

Donald4444, 2020-12-17:
@ccontrer can you also run the command ``sudo find /sys/ -iname "*wacf*"``?
@Alexfebbraio currently I have not a lot of Ideas, because it get's very near to the kernel and we will probably need someone with more experience...

Donald4444, 2020-12-17:
Can somebody where the touchscreen isn't working run ``sudo find /sys/ -iname "*wacf*"`` and run for every output ``sudo find "{OUTPUT}" -name "driver"`` (e.g. ``sudo find "/sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-WACF2200:00" -name "driver"``) and for every output of that ``readlink {OUTPUT}`` (e.g. ``readlink /sys/devices/pci0000:00/0000:00:11.0/ata1/host0/target0:0:0/0:0:0:0/driver`` (this exaple is from my current pc. This will not be showing up in your output))? @Alexfebbraio if your touchscreen isn't working you can run that too.

With some luck we can find a module that is taking over the wacom device. If this is a bit unclear I can also write a small script.

ccontrer, 2020-12-17:
> @ccontrer can you also run the command `sudo find /sys/ -iname "*wacf*"`?

[wacf.log](https://github.com/linuxwacom/input-wacom/files/5711080/wacf.log)

Donald4444, 2020-12-17:
Thank you. I'm currently a bit confused comparing the outputs:

Working:
```
/sys/kernel/debug/regulator/reg-dummy-regulator-dummy/i2c-WACF2200:00-vddl
/sys/kernel/debug/regulator/reg-dummy-regulator-dummy/i2c-WACF2200:00-vdd
/sys/class/devlink/regulator.0--i2c-WACF2200:00
/sys/devices/platform/reg-dummy/regulator/regulator.0/i2c-WACF2200:00-vddl
/sys/devices/platform/reg-dummy/regulator/regulator.0/consumer:i2c-WACF2200:00
/sys/devices/platform/reg-dummy/regulator/regulator.0/i2c-WACF2200:00-vdd
/sys/devices/platform/AMDI0010:00/i2c-0/i2c-WACF2200:00
/sys/devices/virtual/devlink/regulator.0--i2c-WACF2200:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/AMDI0010:00/WACF2200:00
/sys/bus/acpi/devices/WACF2200:00
/sys/bus/i2c/devices/i2c-WACF2200:00
/sys/bus/i2c/drivers/i2c_hid/i2c-WACF2200:00
```

Not Working (Only differnce between the not working outputs are some dynamic numbers, like ``i2c-14``):
```
/sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-WACF2200:00
/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7e/WACF2200:00
/sys/bus/acpi/devices/WACF2200:00
/sys/bus/i2c/devices/i2c-WACF2200:00
```

I would think because the i2c_hid module is loaded on both systems (see the lsmod files) both should have the ``/sys/bus/i2c/drivers/i2c_hid/i2c-WACF2200:00`` file/directory, but they don't. Like I already wrote I don't know that much about the kernel driver itself and in which order they are loaded.
I think it would be the best to get the loaded drivers for the touchscreens with the commands of the second last comment and see if there is some driver taking over the device and stopping the wacom driver.

Alexfebbraio, 2020-12-17:
> Can somebody where the touchscreen isn't working run `sudo find /sys/ -iname "*wacf*"` and run for every output `sudo find "{OUTPUT}" -name "driver"` (e.g. `sudo find [...]

If I'm not misunderstood, I ran the following commands without getting any output:
sudo find "/sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-2/i2c-WACF2200:00" -name "driver"
sudo find "/sys/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:7e/WACF2200:00" -name "driver"
sudo find "/sys/bus/acpi/devices/WACF2200:00" -name "driver"
sudo find "/sys/bus/i2c/devices/i2c-WACF2200:00" -name "driver"" -name "driver"

[68 more replies]

closed 2021-02-04 by jigpu