linuxwacom/input-wacom #219: old WACOM-IV serial device

jbeale1, 2020-11-30:
Surely not a bug, but simply me as a confused user. I have a 25-year-old WACOM ArtZ II serial tablet speaking "WACOM IV" protocol at 9600 baud. I want to use this with a Raspberry Pi Model 4 computer. I have compiled and am running my own custom kernel for the Pi with CONFIG_TABLET_SERIAL_WACOM4=y
```
# uname -a
Linux rp49 5.4.80-v7l-CUSTOM_WACOM+ #1 SMP Mon Nov 30 10:42:40 PST 2020 armv7l GNU/Linux
```
and I also compiled inputattach from https://github.com/linuxwacom/input-wacom/tree/master/inputattach

I can confirm my tablet (connected with USB-Serial adaptor, on /dev/ttyUSB0) generates the expected Wacom4 format 7-byte binary packets when I move the pen over it, for example:
```
root@rp49:/home/pi/input-wacom/inputattach# xxd /dev/ttyUSB0
00000000: e033 3900 171e 40e0 3331 0017 1b40 e033  .39...@.31...@.3
00000010: 2a00 171b 40e0 3322 0017 1c40 e033 1a00  *...@.3"...@.3..
[...]
```

Now although I can run inputattach 
`./inputattach --daemon --baud 9600 --wacom /dev/ttyUSB0`
(with or without the --daemon mode) and it doesn't generate any errors, it also doesn't create anything new in /dev/input/ 
It's still just event0 and event1 which are from my keyboard and mouse, if I understand /var/log/Xorg.0.log correctly.
```
[    10.951] (II) event1  - Microsoft  Microsoft Basic Optical Mouse v2.0 : device is a pointer
[    11.122] (II) event0  - DELL DELL USB Keyboard: device is a keyboard
```
Should I expect inputattach to cause **event2** or similar, to appear in /dev/input/ as the tablet device?

jbeale1, 2020-11-30:
I should add, the '-dump' command does also work as expected (I think):
```
# ./inputattach --baud 9600 -dump /dev/ttyUSB0
e0 (x) 23 (#) 28 (() 00 (x) 24 ($) 38 (8) 40 (@) 
e0 (x) 23 (#) 7d (}) 00 (x) 25 (%) 07 (x) 40 (@) 
e0 (x) 24 ($) 58 (X) 00 (x) 25 (%) 5b ([) 40 (@) 
e0 (x) 25 (%) 05 (x) 00 (x) 26 (&) 06 (x) 40 (@) 
e0 (x) 26 (&) 0e (x) 00 (x) 26 (&) 7f (x) 40 (@) 
e0 (x) 26 (&) 39 (9) 00 (x) 27 (') 25 (%) 40 (@) 
e0 (x) 27 (') 08 (x) 00 (x) 27 (') 69 (i) 40 (@) 
e0 (x) 27 (') 50 (P) 00 (x) 28 (() 24 ($) 40 (@) 
```

jigpu, 2020-12-01:
Its been ages since I've messed with the serial protocol, but all that sounds correct (except, of course, the lack of a new device in /dev/input). If I had to guess, the kernel driver might not support your device. After running the inputattach command, is there anything interesting at the end of the `dmesg` log? That kernel driver only accepts a limited number of devices and is supposed to print out an ["Unsupported Wacom model" error message](https://github.com/torvalds/linux/blob/b65054597872ce3aefbc6a666385eabdf9e288da/drivers/input/tablet/wacom_serial4.c#L252-L256) if your device isn't in the list.

jbeale1, 2020-12-02:
Thank you very much for your reply. After doing **sudo ./inputattach --daemon --baud 9600 --wacom /dev/ttyUSB0** 
I do see one line added to 'dmesg' : **[87109.273819] serio: Serial port ttyUSB0** but no usable device.  My understanding is this may have worked at one time, but some changes to X11 input setup (as least on Raspberry Pi) over the past 5 years may have closed off the previous method of attaching serial X11 devices.  I also have never tried this, and I may be totally wrong.

Anyway without further X system config, but just by cobbling together various code, I did actually manage to create my very own userspace driver code for this specific device and it is working under X. The tablet appears as an absolute-position 2-button mouse.  It also gets stylus pressure (0-127) and even optionally X and Y axis tilt, although I don't use that. So I'm good for now. What I did is at https://github.com/jbeale1/DataAcq/blob/master/Wacom_Input.c

jigpu, 2020-12-02:
We did remove the serial tablet code from xf86-input-wacom years ago, so its not possible to use the old Xorg configurations that try to connect the serial port directly to the driver. The inputattach daemon (and corresponding kernel drivers) was the replacement for that code, but the peripheral serial driver (as opposed to the driver for built-in serial tablet PC sensors) was pretty rarely used from what I understand. It /should/ be possible to get that driver working, but its pretty far down the priority queue for me...

Glad to hear that you were able to cobble a solution together regardless :)

jbeale1, 2020-12-03:
Thanks again, and it's not fair to ask anymore of your time for what may be the last serial tablet like this on earth. But in case I did want to add stylus pressure sensing, it looks like it becomes a "tablet" type, not a simple (absolute) X,Y mouse which did work ok. When my code includes **ioctl(fd, UI_SET_ABSBIT, ABS_PRESSURE)** set up a tablet pressure value, I do see the /dev/input/event2 still show up apparently OK, but it is reported as ID_INPUT_TABLET and the **/var/log/Xorg.0.log** complains of a missing driver, and the cursor no longer responds to the pen.  Is there any online reference that explains how X11 wants to see a tablet stylus pressure device set up?
```
# udevadm info -q all -n /dev/input/event2
P: /devices/virtual/input/input17/event2
N: input/event2
L: 0
E: DEVPATH=/devices/virtual/input/input17/event2
E: DEVNAME=/dev/input/event2
E: MAJOR=13
E: MINOR=66
E: SUBSYSTEM=input
E: USEC_INITIALIZED=141629041096
E: ID_INPUT=1
E: ID_INPUT_TABLET=1
E: ID_SERIAL=noserial
```

```
# tail /var/log/Xorg.0.log
[141628.961] (II) config/udev: Adding input device Wacom ArtZ II (/dev/input/mouse1)
[141628.961] (II) No input driver specified, ignoring this device.
[141628.961] (II) This device may have been added with another device file.
[141629.151] (II) config/udev: Adding input device Wacom ArtZ II (/dev/input/event2)
[141629.151] (II) No input driver specified, ignoring this device.
[141629.151] (II) This device may have been added with another device file.
```

jbeale1, 2020-12-03:
What I understand the X log saying that makes the mouse pointer work (when my device is seen as an XY mouse and not a pressure-sensitive stylus), is this section in my file **/usr/share/X11/xorg.conf.d/40-libinput.conf**
```
Section "InputClass"
        Identifier "libinput pointer catchall"
        MatchIsPointer "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection
```
Is my problem that as far as X configuration files go, MatchIsPointer is always separate from MatchIsTablet ?

jbeale1, 2020-12-03:
Please ignore my rambling. Turns I hadn't done all the necessary setup. I found additional code examples and now the pressure sensitivity works as I had hoped.  Crude, but working code: https://github.com/jbeale1/DataAcq/blob/master/WacomArtZ.c

jigpu, 2020-12-03:
Just to follow up on a couple of the open questions for anyone else in the future: yes, `MatchIsPointer` and `MatchIsTablet` refer to different sets of devices. The matches depend on the udev tags as you may have found out. In general I think failures of the `/dev/input/mouse*` device are fine / expected, with `/dev/input/event*` being preferred instead (that's definitely the case with xf86-input-wacom, but I'm not sure about libinput).