linuxwacom/input-wacom #220: One by Wacom no longer working after CentOS 7 Kernel update

nigel-woolaway, 2020-12-03:
I am running CentOS 7 with the 3.10.0-1062.12.1 kernel and my One by Wacom tablet was working perfectly using the 0.45 input-wacom modules (I had tried 0.46 but they didn't work). After performing an update the tablet no longer worked (no longer able to move the cursor using the tablet pen) so I tried booting with the previous kernel (3.10.0-1062.12.1) without success. I have tried compiling and installing the 0.45 modules, the 0.46 modules and in desperation even the 0.43 version. The device seems to be recognized in some way but it does not control cursor movement.

lsusb | grep -i wacom returns
Bus 001 Device 005: ID 056a:037b Wacom Co., Ltd CTL-672 [One by Wacom (M)]

grep "" /sys/module/wacom*/version returns
grep: /sys/module/wacom*/version: No such file or directory

but modinfo wacom | grep version returns
rhelversion:    7.7
srcversion:     54DA32341E631C895C1E5D2
vermagic:       3.10.0-1062.12.1.el7.x86_64 SMP mod_unload modversions 

I'm happy to provide any info needed to assist in debugging.

Thanks

jigpu, 2020-12-03:
It sounds like you've already done this, but I just want to be sure that you've recompiled and reinstalled the inpu-wacom driver now that the update is in place. The driver is only compatible with the kernel you compiled it with, so it does stop working whenever your distro ships a kernel update.

Please be sure your tablet is connected, reboot, and then run [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh). After it finishes, attach the generated tarball to this issue for review. This will give us a bit more information about the state of the driver installation and any errors that may be occurring.

nigel-woolaway, 2020-12-03:
Thank you for the fast follow-up. I have recompiled and re-installed but just to be sure I have done it again. I have done this with the 3.10.0-1062.12.1 kernel and version 0.45 of input-wacom since this was the last known working combination but if you need me to test other combinations I am happy to do so. I have attached the tarball from sysinfo.sh

[sysinfo.bcW8nHry6s.tar.gz](https://github.com/linuxwacom/input-wacom/files/5637250/sysinfo.bcW8nHry6s.tar.gz)

jigpu, 2020-12-03:
Thanks for the logs. They're a bit confusing, unfortunately. The device is picked up by the kernel driver, but seems to be processed by our "generic" tablet handler instead of its purpose-built one. This particular tablet doesn't provide our driver with enough information for the generic handler to work, which ends up causing the X driver to refuse to use it.

The mystery is why the generic handler is being used instead of the purpose-built one. CentOS 7.9 was apparently released just a couple weeks ago, and that might be part of the issue. We had to jump through some hoops to get input-wacom to install properly with RHEL / CentOS 7.4, and its possible they've added some more hoops with this latest release. Its possible that the stock 7.9 driver is claiming to support devices that it shouldn't, for example.

jigpu, 2020-12-03:
Actually, after more digging, I notice that you're running the 3.10.0-1062.12.1 kernel, which doesn't match up with your version of CentOS 7.9.2009. Can you switch to the 3.10.0-1160.6.1 kernel and re-run sysinfo.sh for me?

nigel-woolaway, 2020-12-04:
I tried moving back to the 3.10.0-1062.12.1 kernel and version 0.45 because that was the last known working configuration. Here are the tarballs using the 3.10.0-1160.6.1 kernel both with the 0.45 and the 0.46 versions. The symptoms are always the same. If there is any other information you need to help debugging I am happy to help.

[version_0.45.tar.gz](https://github.com/linuxwacom/input-wacom/files/5642113/version_0.45.tar.gz)
[version_0.46.tar.gz](https://github.com/linuxwacom/input-wacom/files/5642114/version_0.46.tar.gz)

nigel-woolaway, 2020-12-10:
I am not sure if this is relevant but doing some research I found an old article which suggested using gpm with Wacom tablets. I installed gpm since this was not present and it didn't make any difference initially but after a reboot the tablet is now working perfectly. It may just be a red herring of course but I thought it could be useful

jigpu, 2020-12-10:
Very strange -- I've never even heard of gpm, let alone it being the key to a tablet fix. Still, glad to hear that its working, whatever the reason may be :)

closed 2020-12-10 by jigpu