linuxwacom/input-wacom #221: [Lenovo Flex 14-API 81SS000BUS] Touchpad and Pen stop working after a while

TechieAndroid, 2020-12-06:
syslogs:
https://github.com/linuxwacom/wacom-hid-descriptors/issues/109

I was running Arch Linux when I opened that issue and since then I've moved to Gentoo Linux. I have yet to extensively test the inputs because I still have a few kernel bugs to iron out myself.

But back then what would happen was basically, I would be using my laptop for a while and I might even have been drawing in Krita with my pen or just leaving my laptop running, and then at some random time the touchpad and touchscreen input would stop responding, but my MX Master 2S doesn't experience this problem so I can still use it when the others drop. Interestingly when the mouse input and touchscreen input drop my keyboard continues to work.

I disassembled my laptop and took pictures of all the ICs to try to better figure out the hw for my drivers. My touchpad uses an [Elan Tech IC](https://imgur.com/XXHC7G9). Unfortunately I don't really know what my touchscreen uses but I think it's a wacom based IC.

TechieAndroid, 2020-12-15:
Is this something that can be fixed? I'd love to get some input on the problem from someone who knows more about it than me, maybe I can try to look into fixing it, I just need to know where to start.

jigpu, 2021-01-06:
Probably the best way to start looking into this would be to grab some logs immediately after the problem presents itself and then attach them to this issue:
* Run the [sysinfo script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) so we can compare it to the working version you previously provided
* Run `dmesg | tail -n 300 > dmesg.log` so we have more information about what had been going on in the kernel recently
* If you are using systemd, run `sudo journalctl -n 300 > systemd.log` so that we can see any other system-wide logs

In addition, it would be useful to check the various driver layers to try to isolate a potential cause:
* Install the "xinput" package and then run `xinput list` followed by `xinput test <id>` with the ID number of your pen or touch device. See if starts printing out any events as you use the device. This tests to see if events are being generated by the Xorg driver.
* Install the "libinput" package and then run `sudo libinput record` and select the pen or touch device from the list.  See if starts printing out any events as you use the device. This tests to see if events are being generated by the kernel driver.
* Run `for F in /sys/bus/hid/devices/*; do NAME=$(awk -F= '/HID_NAME/{print $2}' $F/uevent); DEVICE=$(ls $F/hidraw 2>/dev/null || echo "???"); echo -e "${DEVICE}\t${NAME}"; done` to get a table of hidraw nodes and device names. Run `sudo hexdump /dev/<hidrawN>` with the name of the hidraw node for your sensor. See if it starts printing out anything as you use the device. This tests to see if events are being generated by the hardware.

TechieAndroid, 2021-02-08:
Apologies for the long wait but I had a terrible time finding the missing module in my kernel config to get my touchpad and touchscreen to work in Gentoo.

Now that that's out of the way and it's working I have ran the sysinfo script after WCOM51C7 and WCOM51C7 stopped responding:
[sysinfo.ThELDtpqBS.tar.gz](https://github.com/linuxwacom/input-wacom/files/5941037/sysinfo.ThELDtpqBS.tar.gz)

Hopefully, soon we can figure out the root of my problem and put the final nail in the coffin. Thanks again for your attention and help. :D

TechieAndroid, 2021-02-08:
Just to be as thorough as I can here's a sysinfo log from when my touchpad and touchscreen work.
[sysinfo.Ws9iG8EagR.tar.gz](https://github.com/linuxwacom/input-wacom/files/5941237/sysinfo.Ws9iG8EagR.tar.gz)

jigpu, 2021-02-09:
Thanks. I've checked the two sysinfo results and don't see anything particularly interesting. The devices are still present in the kernel and X11, and I don't see any errors that would explain the problem.

I'll need you to some other tests see if we can isolate what component is malfunctioning. Specifically, perform the actions listed at end of my previous post (involving xinput, libinput, and hidraw) to see which layers are sending events and which layers are not.

TechieAndroid, 2021-02-10:
Okay I ran the `xinput test`, `sudo libinput record`, and `sudo hexdump /dev/<hidrawN>` and all successfully detected my pen (hovering and touching the screen) and the two buttons on it.

I don't know if this has useful information or not but here's my terminal's history of running those tests:
https://pastebin.com/etua9qkn

Should I also run those tests when my pen stops working?

jigpu, 2021-02-10:
Yes; I'll need to know which are not working when the pen stops working.

TechieAndroid, 2021-02-14:
Okay sorry for the late reply. It's been working a lot more it seems like lately.

Today it didn't work and I ran all of those tests and none of them showed anything in the terminal.

I checked not just the wacom device but also the elan device since the two seem to be connected somehow. When I can't draw I also can't use the touchpad, they both stop working at the same time.

TechieAndroid, 2021-02-15:
Only thing that stands out to me is that I get all these xorg errors.
```
[     7.675] (EE) Wacom HID 51C7 Pen stylus: Invalid type 'cursor' for this device.
[     7.675] (EE) Wacom HID 51C7 Pen stylus: Invalid type 'touch' for this device.
[     7.675] (EE) Wacom HID 51C7 Pen stylus: Invalid type 'pad' for this device.
[     7.690] (EE) Wacom HID 51C7 Finger: Invalid type 'stylus' for this device.
[     7.690] (EE) Wacom HID 51C7 Finger: Invalid type 'eraser' for this device.
[     7.690] (EE) Wacom HID 51C7 Finger: Invalid type 'cursor' for this device.
[     7.690] (EE) Wacom HID 51C7 Finger touch: Invalid type 'stylus' for this device.
[     7.690] (EE) Wacom HID 51C7 Finger touch: Invalid type 'eraser' for this device.
[     7.690] (EE) Wacom HID 51C7 Finger touch: Invalid type 'cursor' for this device.
[     7.690] (EE) Wacom HID 51C7 Finger touch: Invalid type 'pad' for this device.
```
[Xorg.0.log](https://github.com/linuxwacom/input-wacom/files/5978809/Xorg.0.log)

jigpu, 2021-02-16:
> Today it didn't work and I ran all of those tests and none of them showed anything in the terminal.
>
> I checked not just the wacom device but also the elan device since the two seem to be connected somehow. When I can't draw I also can't use the touchpad, they both stop working at the same time.

This all seems to indicate that there's something going on at a lower level than our driver. The `sudo hexdump /dev/<hidrawN>` command in particular is reading events from the same layer as our kernel driver. If it doesn't print anything out when the stylus is in proximity, that means that our kernel driver isn't seeing any events either. The fact that the Wacom and Elan sensors cut out at the same time is also potentially an indicator of low-level issues, e.g. with the I2C drivers or hardware.

Can you share the output of `dmesg` the next time you notice this, along with the output from `cat /proc/uptime` and a guess of how long the system was in this broken state (e.g. "1 minute" if it happened while you were using it, or "3 hours" if you left the machine for a while and came back, etc.). Also, have you noticed if this seems to occur more often after the system is woken from sleep, or does it also sometimes happen while the device is either actively in use or idle but not sleeping?

> Only thing that stands out to me is that I get all these xorg errors.

These "errors" are actually completely normal, I'm afraid. They're a byproduct of how we try to automatically determine what type a device is. I might write a patch to silence those "errors" since they just make it harder for the average user to find real issues...

TechieAndroid, 2021-02-20:
All I can find that looks like it might be related to the problem in dmesg is:
```
[    3.198928] i2c_hid i2c-WCOM51C7:00: supply vdd not found, using dummy regulator
[    3.198973] i2c_hid i2c-WCOM51C7:00: supply vddl not found, using dummy regulator
[    3.227020] i2c_hid i2c-ELN4690:00: supply vdd not found, using dummy regulator
[    3.227067] i2c_hid i2c-ELN4690:00: supply vddl not found, using dummy regulator
[    3.227239] i2c_hid i2c-BMA250E:00: supply vdd not found, using dummy regulator
[    3.227293] i2c_hid i2c-BMA250E:00: supply vddl not found, using dummy regulator
```
The elan and wacom device both worked at boot but stopped working after something like 10hrs of uptime if I had to guess, I wasn't able to check as frequently as I wanted. And it's impossible to predict when they will stop working.

I can check but I don't think those messages appear in dmesg when they work.

uptime:
`01:19:52 up 12:07,  6 users,  load average: 0.77, 1.18, 1.04`

dmesg
[dmesg-20-2-21.log](https://github.com/linuxwacom/input-wacom/files/6014347/dmesg-20-2-21.log)

TechieAndroid, 2021-02-21:
On a side note, I was wondering if my laptop would support a WAES v2.0 pen like this one for instance: https://www.lenovo.com/us/en/accessories-and-monitors/pens-and-supplies/pens/TAB-ACC-BO-Lenovo-Pen-Pro-ROW/p/4X80R02889

TechieAndroid, 2021-02-21:
This evening I started up my laptop and only had it up for about 2hrs and then they stopped working, I caught it right when it happened.

`01:41:42 up  2:04,  6 users,  load average: 1.27, 3.58, 4.11`
dmesg log
[dmesg-21-2-21.log](https://github.com/linuxwacom/input-wacom/files/6017343/dmesg-21-2-21.log)

TechieAndroid, 2021-02-21:
Forgive me for spamming but I just had it stop working while I was using it, I was moving the mouse and it suddenly stopped.

uptime
`17:25:57 up 56 min,  6 users,  load average: 0.40, 0.97, 1.06`
[dmesg-21-2-21-[2nd].log](https://github.com/linuxwacom/input-wacom/files/6018831/dmesg-21-2-21-.2nd.log)

jigpu, 2021-02-23:
Thanks. Unfortunately I don't see anything that looks particularly interesting in the logs. Nothing I2C related, and as it can happen while the device is in use it shouldn't be related to power-saving modes or sleep. All three logs have some USB activity around the time of the issue, but I don't think it would be related.

When this issue happens, how do you restore functionality? Does logging out and in again work, or will only a reboot work? Or something else?

TechieAndroid, 2021-02-23:
I haven't tried logging out and back in.

My solution has been to just reboot.

TechieAndroid, 2021-03-03:
So I just realized the other day that I wasn't using the driver you told me to use. I had been on wacom_i2c, and you had told me to use hid_wacom because it was newer, and I guess I got confused and never switched.

Well, a few days ago I noticed that and switched to hid_wacom to see if anything changed. It has been a lot more responsive and less prone to stop working from what I've been able to tell, but it still will stop working just like the other driver did.

I've done my best to keep logs and test out what works and doesn't work.

I re-ran all of those initial input command tests you told me to at the beginning of this chat and none yielded any readings from the pen.

I tried suspending my laptop and resuming as well as relogging back in and neither worked to fix it.

dmesg log from when it works:
[dmesg-working-1-3-21.log](https://github.com/linuxwacom/input-wacom/files/6074594/dmesg-working-1-3-21.log)

dmesg log from when it didn't work:
[dmesg-28-2-21.log](https://github.com/linuxwacom/input-wacom/files/6074596/dmesg-28-2-21.log)

syslog from when it works:
[sysinfo.YYo7CmVsQv.tar.gz](https://github.com/linuxwacom/input-wacom/files/6074621/sysinfo.YYo7CmVsQv.tar.gz)

Syslog from when it doesn't work:
[sysinfo.FcOA2RPYWn.tar.gz](https://github.com/linuxwacom/input-wacom/files/6074623/sysinfo.FcOA2RPYWn.tar.gz)

TechieAndroid, 2021-03-07:
I believe the issue is within the synopsys designware i²c bus driver in the kernel. Since both the touchpad and touchscreen stop working simultaneously and they both work off the i²c bus.

I'm going to try to see if I can get the problem to happen on the stable 5.4 kernel binary I have. If so then I'll keep working my way back until the problem goes away. I'll try to see if I can find out who mainly works on the synopsys designware drivers, if there's anyone in particular, and ask them about the issue.

TechieAndroid, 2021-03-07:
I'm closing this for now. If there does turn out to be a bug in the wacom driver I can always just make a new report but at this point all paths seem to point to the i²c bus.

closed 2021-03-07 by TechieAndroid