linuxwacom/input-wacom #222: MateBook HZ-W19, active stylus detected as finger

daxid, 2020-12-10:
Hi,

This issue follow this other one I opened to add the sysinfo for my tablet: https://github.com/linuxwacom/wacom-hid-descriptors/issues/111

On my MateBook HZ-W19, the active stylus is detected as my finger under Manjaro (confirmed using evtest).

However, it seems the events should be distinct :

```
/dev/input/event7:	Wacom HID 4834 Pen
/dev/input/event8:	Wacom HID 4834 Finger
```
Please find attached a recording of logs when the stylus is used (with a scrip capture.sh) : 
[record_1607586728.tar.gz](https://github.com/linuxwacom/input-wacom/files/5671123/record_1607586728.tar.gz)

And when the finger is used :
[record_1607586894.tar.gz](https://github.com/linuxwacom/input-wacom/files/5671127/record_1607586894.tar.gz)

I can also see an eraser with xsetwacom (but I'm usually under wayland)

```
~ >>> xsetwacom list devices                                              [130]
Wacom HID 4834 Pen stylus       	id: 21	type: STYLUS    
Wacom HID 4834 Finger touch     	id: 22	type: TOUCH     
Wacom HID 4834 Pen eraser       	id: 29	type: ERASER    
~ >>> libinput list-devices | grep Wacom                                       
Device:           Wacom HID 4834 Pen
Device:           Wacom HID 4834 Finger
```

Best regards

jigpu, 2020-12-10:
Thanks for the logs. The raw HID events from the sensor indicate that the hardware is reporting the events as a touch, which is certainly odd. Not all active pens and sensors are compatible with each other, and an incompatible combination could possibly result in this behavior. Did this pen come with the MateBook, or was it purchased as a separate item (even if purchased at the same time -- sometimes the "recommended" pen isn't correct)? In either case, can you share any identifying information about the pen (manufacturer, model number, etc.)?

daxid, 2020-12-10:
Thank you very much for the quick answer.

The tablet-pc is refurbished so it is possible that the pen is not the original...
There is nothing to strictly identify the pen (model or serial number) but it looks exactly as this one :
https://www.aliexpress.com/item/32849894557.html

Which seems to be a Huawei Stylus for Matebooks...

Please let me know if I can run any other test to progress. Unfortunately I don't think that I can find another active pen to test before a while...

jigpu, 2020-12-10:
Thanks, I'm not familiar with that particular pen, but looking at the link and other similar-looking styli on Amazon I'm pretty sure it is an active dumb stylus. This kind of active stylus can be recognized in advertising by claims of compatibility with a wide range of touchscreens, e.g. iPads, Android devices, tablet PCs, etc. This compatibility is achieved by projecting a fairly simple electric field that simulates a finger. They may also mention that they do not support palm rejection, which is because its just a normal finger as far as the sensor can tell. Finally, they are often quite inexpensive due to the simpler electronics.

I'm having a hard time finding out information about the HZ-W19, but the sysinfo you provided earlier points to it using one of Wacom's earlier AES sensors. You might look at Wacom's "[Bamboo Ink Stylus](https://www.wacom.com/en-us/products/stylus/bamboo-ink)" or Huawei's "[MatePen Stylus](https://www.amazon.com/Huawei-MatePen-Stylus/dp/B01GNL937A)" which should both work with older AES sensors. Other AES pens (e.g. from Lenovo) may also work, though I'd definitely try to go hands-on with one first to see.

daxid, 2020-12-10:
Thanks a lot for your kind answer.

I'll close the issue and reopen it if need be after trying with another stylus.

Best regards

closed 2020-12-10 by daxid

daxid, 2020-12-18:
I can confirm that a  "Bamboo Ink Stylus"  is working fine :-)
It is not showing up in gnome settings but that's another story...

jigpu, 2020-12-18:
Glad to hear it :) GNOME's handling of AES pens isn't very good at the moment, but I would expect that particular pen to at least show up in Manjaro since it should have the latest GNOME and libwacom releases...

daxid, 2020-12-18:
Pen and tablet settings now show up after I manually added the hid descriptor from my other issue (linuxwacom/wacom-hid-descriptors#111) to /usr/share/libwacom, and reboot.

The feeling is a bit disappointing though. When I write on some parts of the screen it prints a dotted line. On some other it seems OK... Well maybe the tablet screen is just damaged anyhow.

Thanks again for the great support.

jigpu, 2021-01-06:
Does it seem like there are specific areas of the screen that don't work, or do you randomly get dotted lines anywhere on the screen? If the effect is limited to a specific area then its most likely either some kind of interference, though I suppose damage is also a possible explanation. If the effect occurs everywhere there's a small chance it could be the pen since it wasn't explicitly on the supported device list, but I've had pretty good luck getting arbitrary pen/sensor combinations to work in the past. Could also be a firmware or driver issue; we've had both over the years...

daxid, 2021-01-06:
Thanks for the renewed support @jigpu 

Dotted lines seem to happen on specific parts of the screen.
I made some tests by assigning the pen to touchscreen in xournal++ so it could move my pointer.
I can clearly see that the pointer move is a bit irregular on the problematic parts of the screen.
However, if I move the pen close to the screen without touching it, the cursor follows smoothly.

If the hardware was damaged, I would expect the cursor move to be irregular even in the second case...

Can I gather additional information that could help debug ?

best regards