linuxwacom/input-wacom #223: Xorg no longer accepting Cintiq Companion 2 finger touch device after kernel update from 4.15 to 5.9

maximilianduell, 2020-12-14:
Hi, 

I am running debian on a cintiq companion 2 natively. After updating the kernel, xorg unfortunately no longer accepts the finger touch device, but the stylus and eraser devices still work fine. It is not an xorg issue, because everything still works when booting the same system into the older kernel. 

Also in the newer kernel, the input device is still present and reporting finger touch events (using evtest).

I believe this is an issue of API breaking, probably the relevant line of the xorg log is

Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (EE) Wacom ISDv5 326 Finger touch: xmax value is 0, expected > 0.
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (EE) PreInit returned 8 for "Wacom ISDv5 326 Finger touch"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) UnloadModule: "wacom"

Below is the full part of the log regarding the wacom devices.

Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) config/udev: Adding input device Wacom ISDv5 325 Pen (/dev/input/event12)
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pen: Applying InputClass "evdev tablet catchall"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pen: Applying InputClass "libinput tablet catchall"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pen: Applying InputClass "Wacom USB tablet class"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pen: Applying InputClass "Wacom tablet class"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) LoadModule: "wacom"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) Module wacom: vendor="X.Org Foundation"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         compiled for 1.20.4, module version = 0.34.99
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Module class: X.Org XInput Driver
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         ABI class: X.Org XInput driver, version 24.1
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Graphire4 4x5, Graphire4 6x8, BambooFun 4x5, BambooFun 6x8,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Bamboo1 Medium, Graphire4 6x8 BlueTooth, CTL-460, CTH-461, CTL-660,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         CTL-461/S, Bamboo Touch, CTH-460/K, CTH-461/S, CTH-661/S1, CTH-461/L,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         CTH-661/L, Intuos 4x5, Intuos 6x8, Intuos 9x12, Intuos 12x12,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Intuos 12x18, PTU600, PL400, PL500, PL600, PL600SX, PL550, PL800,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         PL700, PL510, PL710, DTI520, DTF720, DTF720a, DTF521, DTU1931,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         DTU2231, DTU1631, Intuos2 4x5, Intuos2 6x8, Intuos2 9x12,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Intuos2 12x12, Intuos2 12x18, Intuos2 6x8 , Volito, PenStation,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Volito2 4x5, Volito2 2x3, PenPartner2, Bamboo, Bamboo1, Bamboo1 4x6,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Bamboo1 5x8, Intuos3 4x5, Intuos3 6x8, Intuos3 9x12, Intuos3 12x12,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Intuos3 12x19, Intuos3 6x11, Intuos3 4x6, Intuos4 4x6, Intuos4 6x9,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Intuos4 8x13, Intuos4 12x19, Intuos4 WL USB Endpoint,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Intuos4 WL Bluetooth Endpoint, Intuos5 touch S, Intuos5 touch M,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Intuos5 touch L, Intuos5 S, Intuos5 M, Intuos Pro S, Intuos Pro M,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Intuos Pro L, Cintiq 21UX, Cintiq 20WSX, Cintiq 12WX, Cintiq 21UX2,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Cintiq 24HD, Cintiq 22HD, Cintiq 24HD touch (EMR digitizer),
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Cintiq 13HD, DTK2241, DTH2242, Cintiq 22HDT, Mobilestudio Pro 13,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         MobileStudio Pro 16, MobileStudio Pro 13, MobileStudio Pro 16,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Cintiq 13 FHD Pro, Cintiq 16 UHD Pro, Cintiq 13 FHD Pro,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         Cintiq 16 UHD Pro, TabletPC 0x90, TabletPC 0x93, TabletPC 0x97,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         TabletPC 0x9A, CapPlus  0x9F, TabletPC 0xE2, TabletPC 0xE3,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         TabletPC 0xE5, TabletPC 0xE6, TabletPC 0xEC, TabletPC 0xED,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         TabletPC 0xEF, TabletPC 0x100, TabletPC 0x101, TabletPC 0x10D,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         TabletPC 0x116, TabletPC 0x12C, TabletPC 0x4001, TabletPC 0x4004,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         TabletPC 0x5000, TabletPC 0x5002, usb:172f:0024, usb:172f:0025,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         usb:172f:0026, usb:172f:0027, usb:172f:0028, usb:172f:0030,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         usb:172f:0031, usb:172f:0032, usb:172f:0033, usb:172f:0034,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         usb:172f:0035, usb:172f:0036, usb:172f:0037, usb:172f:0038,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         usb:172f:0039, usb:172f:0051, usb:172f:0052, usb:172f:0053,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         usb:172f:0054, usb:172f:0055, usb:172f:0056, usb:172f:0057,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         usb:172f:0058, usb:172f:0500, usb:172f:0501, usb:172f:0502,
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]:         usb:172f:0503, usb:1b96:0001, usb:17ef:6004
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) Using input driver 'wacom' for 'Wacom ISDv5 325 Pen'
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) systemd-logind: got fd for /dev/input/event12 13:76 fd 36 paused 0
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pen: always reports core events
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Option "Device" "/dev/input/event12"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) Wacom ISDv5 325 Pen: type not specified, assuming 'stylus'.
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) Wacom ISDv5 325 Pen: other types will be automatically added.
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (--) Wacom ISDv5 325 Pen stylus: using pressure threshold of 851 for button 1
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (--) Wacom ISDv5 325 Pen stylus: maxX=59152 maxY=33448 maxZ=2047 resX=200000 resY=200000  tilt=enabled
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) Wacom ISDv5 325 Pen stylus: hotplugging dependent devices.
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (EE) Wacom ISDv5 325 Pen stylus: Invalid type 'cursor' for this device.
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (EE) Wacom ISDv5 325 Pen stylus: Invalid type 'touch' for this device.
[... 34 lines cut ...]
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pad pad: (accel) keeping acceleration scheme 1
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pad pad: (accel) acceleration profile 0
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pad pad: (accel) acceleration factor: 2.000
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (**) Wacom ISDv5 325 Pad pad: (accel) acceleration threshold: 4
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) config/udev: Adding input device Wacom ISDv5 325 Pad (/dev/input/js0)
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) No input driver specified, ignoring this device.
[... cut ...]
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) Wacom ISDv5 326 Finger: other types will be automatically added.
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (EE) Wacom ISDv5 326 Finger touch: xmax value is 0, expected > 0.
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (EE) PreInit returned 8 for "Wacom ISDv5 326 Finger touch"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) UnloadModule: "wacom"
Dec 13 19:57:35 jira /usr/libexec/gdm-x-session[1454]: (II) systemd-logind: releasing fd for 13:73

jigpu, 2020-12-15:
I've quickly tested an emulated version of the Cintiq Companion 2 touch sensor on my dev box with a 5.9 kernel and don't see the same behavior. The `xmax value is 0, expected > 0` message does not appear and the device appears to connect without an issue.

Would you mind trying to test a few versions of input-wacom to see if you can pinpoint when this error begins occurring for you? Version 0.40.0 should be very similar to what shipped with Linux 4.17. If it works fine then try later versions until you get to a point that the xmax error re-appears. Build instructions can be found at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

BTW, it looks like our data on the Cintiq Companion 2 is pretty old; would you mind also following the instructions at https://github.com/linuxwacom/wacom-hid-descriptors/#wacom-hid-descriptors to generate and contribute a fresh sysinfo for us before starting the input-wacom testing?

maximilianduell, 2020-12-18:
Many thanks for your very fast reply. I will try to bisect between 0.40.0 and most recent version on the weekend.

edgar79, 2021-02-09:
I can confirm this bug using Debian 10 stable with kernel 4.19.0-14-amd64. With the same log messages as mentioned above.
Before, I tested the device using Ubuntu 20.04 LTS - but the Touchscreen was not working out-of-the-box...

I already tried the latest releases of input-wacom and the X driver, but the `xmax value is 0` message remained...

I would like to help - Could you provide me instructions how to resolve this?

edgar79, 2021-02-10:
I found a working solution an was able to bisect the first commit where the error reoccurred.

- I upgraded to the 5.9 kernel from debian backports..
- installed input-wacom, xf86-input-wacom and libwacom from source (master)
- found input-wacom-0.42.0 as the last working tag
- used git bisect input-wacom-0.43.0 input-wacom-0.42.0

The attached file is the output of `git bisect log`

[Wacom-Cintiq-Companion-2-Touchscreen-bisect-log.txt](https://github.com/linuxwacom/input-wacom/files/5958352/Wacom-Cintiq-Companion-2-Touchscreen-bisect-log.txt)

jigpu, 2021-02-10:
Thanks, I'm reviewing the commit highlighted by the bisect right now. Would you mind following the instructions at https://github.com/linuxwacom/wacom-hid-descriptors/#contributing to contribute a sysinfo log? The last log we received is missing a lot of information and an update would be appreciated.

jigpu, 2021-02-10:
I've made a fix for a suspicious edge case that hopefully addresses this problem. Please checkout my branch by running these commands inside your input-wacom repository, and then rebuild / reinstall to test the fix.

~~~
git fetch https://github.com/jigpu/input-wacom.git fix-223
git checkout FETCH_HEAD
~~~

If it doesn't seem to work, run `cat /sys/module/wacom/version` and verify that it prints out `v2.00-0.47.0.1.g6a40977`

edgar79, 2021-02-10:
The touchscreen is working again.
I tested the fix using kernel 5.9 as well as 4.19
with Xorg (wacom and libinput) and Wayland. 

**Thank you very much!**

jigpu, 2021-02-16:
Patch sent upstream. Waiting for a response.