linuxwacom/input-wacom #224: Touchscreen on Lenovo Yoga 12 fails to initialize after Suspend or Restart

middle-road, 2020-12-19:
I am attempting to setup (2) Yoga 12's. Frustrating, been at this for several days now.
Touchscreen stops functioning after Suspend, Sleep or a Restart. Lid is set to 'Suspend' on close.
It requires a Shutdown to regain function.
This post is similar to #59 except that I do not have a pen present.
Currently am on Mint 20, though the problems is also present on Live boots of other flavors.
I plan on trying a different distro after I swap hard drives, though I'm not sure which one yet.
Touchscreen works under all situations with win7pro.

`uname -a
Linux Yoga-12-4GB 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

modinfo wacom:
was:
`version:        v2.00
srcversion:     D2922491EA8A40046FE52F5`
Has been updated to:
`version:        v2.00-0.46.0
srcversion:     62537F6171CFA60B57FF9D7`

I have gone over the innards and reseated all cables on the mobo(s).
I have a thread going over on the Mint forums with additional information:
[https://forums.linuxmint.com/viewtopic.php?f=49&t=337726&p=1939447#p1939447](url)

middle-road, 2020-12-23:
Tried **Xubuntu 20** on the 2nd laptop. Same behavior. 
Same error as above.
Have _not_ installed .046 driver -yet. 
Did implement @emesik's service found here:  https://gist.github.com/emesik/d01679052ef3aeb83ebe87f46d223b12

`Linux Yoga12-4GB-xbu 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`
`version:        v2.00
srcversion:     D2922491EA8A40046FE52F5`
`xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech M705                           	id=9	[slave  pointer  (2)]
⎜   ↳ Logitech Unifying Device. Wireless PID:4055	id=10	[slave  pointer  (2)]
⎜   ↳ Wacom Pen and multitouch sensor Finger touch	id=11	[slave  pointer  (2)]
⎜   ↳ Wacom Pen and multitouch sensor Pen stylus	id=12	[slave  pointer  (2)]
⎜   ↳ Synaptics TM3075-002                    	id=16	[slave  pointer  (2)]
⎜   ↳ TPPS/2 IBM TrackPoint                   	id=17	[slave  pointer  (2)]
⎜   ↳ Wacom Pen and multitouch sensor Pen eraser	id=18	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C         	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                  	id=15	[slave  keyboard (3)]`

jigpu, 2021-01-06:
It was reported in issue #110 that blacklisting the wacom driver helped with their issue. If you blacklist our driver, reboot, and then confirm the driver was never loaded (e.g. run `lsmod | grep wacom` and get no result) do you still experience the problem with suspend / resume?

middle-road, 2021-01-06:
I blacklisted the driver on both devices. One was running v.046 the other 2.0.
When applied the touchscreen failed completely, even with a hard boot.
I was thinking that maybe it was a hardware problem but Win7 & Win10 function fine.
I'm planning on the taking the second one and starting over from scratch with Mint.
I'm trying to plan out the steps I should take during setup.
I've tried 5 or 6 distros so far without any luck.

Pinglinux, 2022-08-04:
As with #110, this issue is not in the driver/tablet since the device works well on Windows. Discussion at #110 provided a few workarounds. Hope one of those workarounds or newer kernel resolves makes your system work.

closed 2022-08-04 by Pinglinux