linuxwacom/input-wacom #225: Lenovo Yoga C640 Touchscreen not working even after installing wacom X driver

gsambrotta, 2020-12-19:
Hi there, 
I'm new to Linux, I bought a **Lenovo Yoga C460**, and fresh installed **Ubuntu 20.**4 but Touchscreen is not working.
I installed Wacom X Driver, which was successful installed, then rebooted the laptop but touchscreen is still not working out of the box.
I then added a config file here `etc/X11/xorg.conf.d/ 70-wacom.conf`
The content of the file is exactly the one that is here: https://github.com/linuxwacom/xf86-input-wacom/blob/master/conf/70-wacom.conf

Restart the laptop again but touchscreen is still not working.

Any ideas what can I do more?
Should I install the Kernel Driver too?
Should I upgrade my Kernel?

My kernel is: **Linux 5.4.0-58-generic x86_64**

Any help is really appreciated :D

aditsu, 2020-12-22:
You can try this kernel patch: https://patchwork.kernel.org/project/linux-acpi/patch/20201211021814.36193-1-hui.wang@canonical.com/
Someone else with a C640 said that it helped: https://github.com/linuxwacom/input-wacom/issues/218#issuecomment-749156511

gsambrotta, 2020-12-24:
Thank you @aditsu! I read all the tread you pointed me to but is not yet clear to me exactly what to do.. 
Do I need to update to kernel version 5.10?
My kernel version now is: `5.4.0-58-generic`

I never have update a kernel and I'm a bit nervous at the idea to do it. Would this be a good guide to follow to do so?
https://wiki.archlinux.org/index.php/Kernel/Traditional_compilation

After I updated the kernel I have to apply the patch like is suggested here: https://github.com/linuxwacom/input-wacom/issues/218#issuecomment-748884101. Is this correct?

Also, this comment worry me a bit:
```
I hope that the official kernels of ubuntu, fedora, manjaro etc etc will be corrected as soon as possible so as not to have to recompile at every kernel change
```

It means that if I update the kernel and then a new Ubuntu version come out with a new kernel, I will need to re-update the kernel and re-apply the patch?

Thank you in advance for your explanation :)

jigpu, 2021-01-06:
@gsambrotta the patch was accepted as a fix and has been applied to the 5.10.4 kernel already. There are pre-built versions of this kernel available, so testing it out should be pretty straightforward. You can adapt the instructions from [this guide for the 5.9 kernel](https://ubuntuhandbook.org/index.php/2020/10/linux-kernel-5-9-released/) but use files from the [5.10.4 kernel directory](https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.10.4/) instead.

gsambrotta, 2021-01-22:
@jigpu thank you very much! It does work now!! Yeyy :tada:

closed 2021-01-22 by gsambrotta

james454g, 2024-05-26:
@jigpu this is still an issue on the latest Fedora with kernel 6.8.10. Are we still expected to manually recompile the kernel to fix this?