linuxwacom/input-wacom #230: ThinkPad L14 pen input causes crash

mlugg, 2021-01-20:
Hi,
A recent kernel upgrade began causing a system crash when the stylus is touched to the screen on a ThinkPad L14. I've not observed any dmesg output; in fact, I can't necessarily trace it back to input-wacom, but it seems the most likely culprit. The entire system freezes as soon as the pen is touched to the screen, and becomes completely unresponsive, requiring a hard reset. I am uncertain, but looking at when this issue arose, I believe it came from the upgrade to kernel 5.10.8. The for-5.10 branch history suggests to me that this issue may have arisen as a result of commit afa5129 - however, I don't see any way this could be the case looking at that patch.
Thanks!

jigpu, 2021-01-20:
Similar (same?) issue reported at https://github.com/linuxwacom/input-wacom/issues/110#issuecomment-763740113. I've given some advice in that issue on how to use "netconsole" to capture logs since they don't seem to get saved to disk. I'd suggest trying it out if you can.

I'm currently updating the Fedora install on my tablet PC to see if I can replicate this with their 5.10.8-100 kernel (the same distro/kernel from the other issue).

jigpu, 2021-01-20:
I've been able to reproduce a driver crash and was indeed able to pin the cause on that particular commit. Can you please test my "fix-230" branch of input-wacom by cloning my code with the following command and then following the instructions at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source to build / install the driver?

~~~
git clone -b fix-230 https://github.com/jigpu/input-wacom.git
~~~

mairin, 2021-01-21:
@jigpu I just tested fix-230 with kernel-5.8.9-200.fc32.x86_64 on Fedora 32. The crash on pen input is gone. (The suspend issue is half gone, I'll detail in #110.)

NiklasBeierl, 2021-01-21:
Exerperiencing a similar (maybe the same) issue:
Device: Dell XPS 13 7390 2-in-1 
Kernel: 5.10.9-arch1-1 (Arch linux)
Wacom packages:
```
local/libwacom 1.7-1
    Library to identify Wacom tablets and their features
local/xf86-input-wacom 0.39.0-2
    X.Org Wacom tablet driver
```
dmesg once touching the screen with a pen:
```
[Do Jan 21 12:31:26 2021] wacom 0018:056A:48EB.0001: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[Do Jan 21 12:31:26 2021] BUG: kernel NULL pointer dereference, address: 0000000000000000
[Do Jan 21 12:31:26 2021] #PF: supervisor read access in kernel mode
[Do Jan 21 12:31:26 2021] #PF: error_code(0x0000) - not-present page
[Do Jan 21 12:31:26 2021] PGD 0 P4D 0 
[Do Jan 21 12:31:26 2021] Oops: 0000 [#1] PREEMPT SMP NOPTI
[Do Jan 21 12:31:26 2021] CPU: 6 PID: 365 Comm: irq/155-WCOM48E Not tainted 5.10.9-arch1-1 #1
[Do Jan 21 12:31:26 2021] Hardware name: Dell Inc. XPS 13 7390 2-in-1/06CDVY, BIOS 1.6.2 09/17/2020
[Do Jan 21 12:31:26 2021] RIP: 0010:__kfifo_skip_r+0x13/0x40
[Do Jan 21 12:31:26 2021] Code: 83 c0 01 21 c8 0f b6 04 02 c1 e0 08 41 09 c0 44 89 c0 c3 0f 1f 40 00 48 89 f0 8b 77 04 44 8b 47 08 48 8b 4f 10 89 f2 44 21 c2 <0f> b6 14 11 48 83 f8 01 74 11 44 8d 4e 01 45 21 c8 42 0f b6 0c 01
[Do Jan 21 12:31:26 2021] RSP: 0018:ffffa3ebc07a7de0 EFLAGS: 00010246
[Do Jan 21 12:31:26 2021] RAX: 0000000000000002 RBX: ffff8e26d9570018 RCX: 0000000000000000
[Do Jan 21 12:31:26 2021] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff8e26d9570360
[Do Jan 21 12:31:26 2021] RBP: ffff8e26d9570360 R08: 0000000000000000 R09: ffffa3ebc07a7b28
[Do Jan 21 12:31:26 2021] R10: ffffa3ebc07a7b20 R11: ffffffffbd6cb228 R12: 0000000000000001
[Do Jan 21 12:31:26 2021] R13: ffff8e26c0efe000 R14: ffff8e26c4071f00 R15: 00000000000d003e
[Do Jan 21 12:31:26 2021] FS:  0000000000000000(0000) GS:ffff8e2e2f780000(0000) knlGS:0000000000000000
[Do Jan 21 12:31:26 2021] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Do Jan 21 12:31:26 2021] CR2: 0000000000000000 CR3: 00000006d5c10003 CR4: 0000000000770ee0
[Do Jan 21 12:31:26 2021] PKRU: 55555554
[Do Jan 21 12:31:26 2021] Call Trace:
[Do Jan 21 12:31:26 2021]  wacom_raw_event+0x2ef/0x390 [wacom]
[Do Jan 21 12:31:26 2021]  hid_input_report+0x145/0x160
[Do Jan 21 12:31:26 2021]  ? disable_irq_nosync+0x10/0x10
[Do Jan 21 12:31:26 2021]  i2c_hid_irq+0xac/0x100 [i2c_hid]
[Do Jan 21 12:31:26 2021]  irq_thread_fn+0x20/0x60
[Do Jan 21 12:31:26 2021]  irq_thread+0xf5/0x1a0
[Do Jan 21 12:31:26 2021]  ? irq_finalize_oneshot.part.0+0xe0/0xe0
[Do Jan 21 12:31:26 2021]  ? irq_thread_check_affinity+0xd0/0xd0
[Do Jan 21 12:31:26 2021]  kthread+0x133/0x150
[Do Jan 21 12:31:26 2021]  ? __kthread_bind_mask+0x60/0x60
[Do Jan 21 12:31:26 2021]  ret_from_fork+0x1f/0x30
[Do Jan 21 12:31:26 2021] Modules linked in: uinput ccm xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overlay snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic rfcomm cmac algif_hash algif_skcipher af_alg snd_sof_pci ebtable_filter snd_sof_intel_byt snd_sof_intel_ipc ebtables snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_xtensa_dsp snd_sof_intel_hda snd_sof hid_sensor_accel_3d hid_sensor_als hid_sensor_magn_3d hid_sensor_incl_3d hid_sensor_rotation hid_sensor_gyro_3d hid_sensor_trigger snd_hda_ext_core industrialio_triggered_buffer ip6table_filter kfifo_buf hid_sensor_iio_common industrialio ip6_tables snd_soc_acpi_intel_match snd_soc_acpi snd_hda_intel hid_sensor_hub iptable_filter snd_intel_dspcfg cros_ec_ishtp cros_ec intel_ishtp_loader soundwire_intel soundwire_generic_allocation soundwire_cadence bnep intel_ishtp_hid iTCO_wdt snd_hda_codec wacom intel_pmc_bxt
[Do Jan 21 12:31:26 2021]  dell_laptop x86_pkg_temp_thermal dell_wmi intel_powerclamp coretemp iTCO_vendor_support usbhid mei_hdcp hid_multitouch dell_smbios intel_rapl_msr snd_hda_core kvm_intel ledtrig_audio dcdbas snd_hwdep dell_smm_hwmon wmi_bmof dell_wmi_descriptor intel_wmi_thunderbolt kvm soundwire_bus snd_soc_core iwlmvm irqbypass i915 nls_iso8859_1 snd_compress vfat rapl ac97_bus intel_cstate snd_pcm_dmaengine fat mac80211 intel_uncore snd_pcm snd_timer libarc4 snd btusb i2c_i801 joydev iwlwifi i2c_algo_bit pcspkr mousedev soundcore mei_me btrtl i2c_smbus btbcm drm_kms_helper btintel mei tpm_crb cfg80211 bluetooth cec intel_lpss_pci intel_lpss idma64 intel_gtt processor_thermal_device ecdh_generic ucsi_acpi syscopyarea intel_rapl_common sysfillrect rfkill typec_ucsi intel_ish_ipc tpm_tis sysimgblt ecc thunderbolt intel_ishtp fb_sys_fops intel_soc_dts_iosf typec tpm_tis_core mac_hid wmi int3403_thermal i2c_hid soc_button_array int340x_thermal_zone video intel_hid sparse_keymap int3400_thermal
[Do Jan 21 12:31:26 2021]  acpi_thermal_rel acpi_tad acpi_pad drm fuse crypto_user agpgart bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 dm_crypt cbc encrypted_keys dm_mod trusted tpm rng_core rtsx_pci_sdmmc mmc_core crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel crypto_simd cryptd glue_helper serio_raw rtsx_pci xhci_pci xhci_pci_renesas
[Do Jan 21 12:31:26 2021] CR2: 0000000000000000
[Do Jan 21 12:31:26 2021] ---[ end trace cf0d3324c4799b00 ]---
[Do Jan 21 12:31:27 2021] RIP: 0010:__kfifo_skip_r+0x13/0x40
[... cut ...]
[Do Jan 21 12:31:27 2021] R13: ffff8e26c0efe000 R14: ffff8e26c4071f00 R15: 00000000000d003e
[Do Jan 21 12:31:27 2021] FS:  0000000000000000(0000) GS:ffff8e2e2f780000(0000) knlGS:0000000000000000
[Do Jan 21 12:31:27 2021] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[Do Jan 21 12:31:27 2021] CR2: 0000000000000959 CR3: 0000000113e18005 CR4: 0000000000770ee0
[Do Jan 21 12:31:27 2021] PKRU: 55555554
[Do Jan 21 12:31:27 2021] Fixing recursive fault but reboot is needed!
```
Requires reboot as indicated in the dmesg.

NiklasBeierl, 2021-01-21:
Downgrading from `5.10.9-arch1-1` to `5.10.4-arch2-1` fixed it.

stefanradziuk, 2021-01-21:
On ThinkPad Yoga 370:

dmesg output:
```
Jan 20 22:49:20 yeouido kernel: wacom 0003:056A:509C.0005: wacom_wac_queue_insert: kfifo has filled, starting to drop events
Jan 20 22:49:20 yeouido kernel: BUG: kernel NULL pointer dereference, address: 0000000000000000
```

Cannot reproduce after downgrading from `5.10.9.arch1-1` to `5.10.7.arch1-1`.

mlugg, 2021-01-21:
The fix-230 branch looks like it works - I haven't tested on a later kernel yet, but on 5.9 it works. Unfortunately, when I reboot, the old driver (stock kernel version) is used until I rmmod/modprobe - I guess it's being cached in my initramfs? I might just wait until the 5.11 release, which I assume this'll make it into if there are no problems

jigpu, 2021-01-21:
> Unfortunately, when I reboot, the old driver (stock kernel version) is used until I rmmod/modprobe - I guess it's being cached in my initramfs?

@mlugg that's probably the case. The `make install` step tries to update the initramfs automatically, but it doesn't always work. Given the positive results I'll be sending this off to the LKML today. I'll post back when I get news of its acceptance into the stable kernel trees.

For reference, this is a list of kernel versions which are affected by this bug:
* Linux 4.19.168, 4.19.169
* Linux 5.4.90, 5.4.91
* Linux 5.10.8, 5.10.9
* Linux 5.11-rc4

jugendhacker, 2021-01-22:
> On ThinkPad Yoga 370:
>
> dmesg output:
>
> ```
> Jan 20 22:49:20 yeouido kernel: wacom 0003:056A:509C.0005: wacom_wac_queue_insert: kfifo has filled, starting to drop events
> Jan 20 22:49:20 yeouido kernel: [...]

Same here with Lenovo Thinkpad X1 Yoga 2nd Generation.

mefromthepast, 2021-01-23:
I guess I'm experiencing the same issue on my Dell Latitude 7400 2-in-1. When I use my stylus on the touch screen, there is a kernel crash and the touch screen stops working. It's however possible to just use my fingers before it crashes.

```
[ 1263.136327] wacom 0018:056A:48CA.0003: wacom_wac_queue_insert: kfifo has filled, starting to drop events
[ 1263.136344] BUG: kernel NULL pointer dereference, address: 0000000000000000
[ 1263.136353] #PF: supervisor read access in kernel mode
[ 1263.136357] #PF: error_code(0x0000) - not-present page
[ 1263.136361] PGD 0 P4D 0 
[ 1263.136373] Oops: 0000 [#1] PREEMPT SMP NOPTI
[ 1263.136381] CPU: 6 PID: 405 Comm: irq/95-WCOM48CA Tainted: G           OE     5.10.9-arch1-1 #1
[ 1263.136385] Hardware name: Dell Inc. Latitude 7400 2-in-1/0HCNR3, BIOS 1.10.0 10/26/2020
[ 1263.136400] RIP: 0010:__kfifo_skip_r+0x13/0x40
[ 1263.136407] Code: 83 c0 01 21 c8 0f b6 04 02 c1 e0 08 41 09 c0 44 89 c0 c3 0f 1f 40 00 48 89 f0 8b 77 04 44 8b 47 08 48 8b 4f 10 89 f2 44 21 c2 <0f> b6 14 11 48 83 f8 01 74 11 44 8d 4e 01 45 21 c8 42 0f b6 0c 01
[ 1263.136412] RSP: 0018:ffffb69d010afde0 EFLAGS: 00010246
[ 1263.136419] RAX: 0000000000000002 RBX: ffff8c224c7e9018 RCX: 0000000000000000
[ 1263.136423] RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff8c224c7e9360
[ 1263.136427] RBP: ffff8c224c7e9360 R08: 0000000000000000 R09: ffffb69d010afb28
[ 1263.136431] R10: ffffb69d010afb20 R11: ffffffffb2ecb228 R12: 0000000000000001
[ 1263.136435] R13: ffff8c2240ffa000 R14: ffff8c224d6113c0 R15: 00000000000d003e
[ 1263.136441] FS:  0000000000000000(0000) GS:ffff8c25dc580000(0000) knlGS:0000000000000000
[ 1263.136445] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1263.136449] CR2: 0000000000000000 CR3: 0000000008c10006 CR4: 00000000003706e0
[ 1263.136454] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 1263.136458] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 1263.136461] Call Trace:
[ 1263.136482]  wacom_raw_event+0x2ef/0x390 [wacom]
[ 1263.136499]  hid_input_report+0x145/0x160
[ 1263.136514]  ? disable_irq_nosync+0x10/0x10
[ 1263.136523]  i2c_hid_irq+0xac/0x100 [i2c_hid]
[ 1263.136532]  irq_thread_fn+0x20/0x60
[ 1263.136538]  irq_thread+0xf5/0x1a0
[ 1263.136548]  ? irq_finalize_oneshot.part.0+0xe0/0xe0
[ 1263.136556]  ? irq_thread_check_affinity+0xd0/0xd0
[ 1263.136566]  kthread+0x133/0x150
[ 1263.136573]  ? __kthread_bind_mask+0x60/0x60
[ 1263.136584]  ret_from_fork+0x1f/0x30
[ 1263.136592] Modules linked in: ccm rfcomm uinput snd_hda_codec_hdmi cmac algif_hash algif_skcipher af_alg bnep snd_hda_codec_realtek snd_hda_codec_generic nls_iso8859_1 vfat fat cdc_mbim uvcvideo cdc_ncm btusb videobuf2_vmalloc cdc_ether videobuf2_memops btrtl videobuf2_v4l2 qmi_wwan btbcm videobuf2_common btintel cdc_wdm usbnet option bluetooth mii usb_wwan videodev ecdh_generic hid_sensor_incl_3d hid_sensor_gyro_3d hid_sensor_magn_3d hid_sensor_als hid_sensor_prox ecc hid_sensor_accel_3d hid_sensor_rotation uas hid_sensor_trigger industrialio_triggered_buffer kfifo_buf mc hid_sensor_iio_common industrialio hid_sensor_hub cros_ec_ishtp cros_ec intel_ishtp_loader joydev intel_ishtp_hid snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_xtensa_dsp snd_sof_intel_hda snd_sof snd_soc_skl snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi snd_hda_intel snd_intel_dspcfg soundwire_intel soundwire_generic_allocation
[ 1263.136739]  soundwire_cadence snd_hda_codec snd_hda_core x86_pkg_temp_thermal intel_powerclamp snd_hwdep coretemp soundwire_bus iTCO_wdt dell_rbtn intel_pmc_bxt wacom hid_multitouch mei_wdt mei_hdcp iTCO_vendor_support snd_soc_core kvm_intel intel_rapl_msr dell_laptop dell_wmi dell_smbios ledtrig_audio snd_compress kvm dell_wmi_descriptor iwlmvm intel_wmi_thunderbolt wmi_bmof i915 ac97_bus dcdbas snd_pcm_dmaengine irqbypass dell_smm_hwmon snd_pcm mac80211 rapl i2c_algo_bit intel_cstate snd_timer intel_uncore drm_kms_helper libarc4 snd cec i2c_i801 iwlwifi mousedev pcspkr i2c_smbus soundcore mei_me intel_gtt syscopyarea thunderbolt sysfillrect cfg80211 mei intel_lpss_pci intel_ish_ipc processor_thermal_device sysimgblt intel_lpss intel_rapl_common intel_ishtp rfkill idma64 fb_sys_fops intel_soc_dts_iosf intel_pch_thermal ucsi_acpi typec_ucsi typec tpm_crb wmi mac_hid tpm_tis i2c_hid int3403_thermal int340x_thermal_zone video tpm_tis_core int3400_thermal acpi_thermal_rel intel_hid
[ 1263.136902]  soc_button_array sparse_keymap acpi_pad vboxnetflt(OE) vboxnetadp(OE) vboxdrv(OE) drm sg fuse crypto_user agpgart bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 usb_storage usbhid dm_crypt cbc encrypted_keys dm_mod trusted tpm rng_core crct10dif_pclmul crc32_pclmul crc32c_intel rtsx_pci_sdmmc ghash_clmulni_intel mmc_core aesni_intel crypto_simd cryptd glue_helper serio_raw xhci_pci rtsx_pci xhci_pci_renesas
[ 1263.136994] CR2: 0000000000000000
[ 1263.137001] ---[ end trace bba7afa0d3563e9d ]---
[ 1263.993575] RIP: 0010:__kfifo_skip_r+0x13/0x40
[ 1263.993581] Code: 83 c0 01 21 c8 0f b6 04 02 c1 e0 08 41 09 c0 44 89 c0 c3 0f 1f 40 00 48 89 f0 8b 77 04 44 8b 47 08 48 8b 4f 10 89 f2 44 21 c2 <0f> b6 14 11 48 83 f8 01 74 11 44 8d 4e 01 45 21 c8 42 0f b6 0c 01
[ 1263.993582] RSP: 0018:ffffb69d010afde0 EFLAGS: 00010246
[ 1263.993584] RAX: 0000000000000002 RBX: ffff8c224c7e9018 RCX: 0000000000000000
[... cut ...]
[ 1293.781155] FS:  0000000000000000(0000) GS:ffff8c25dc580000(0000) knlGS:0000000000000000
[ 1293.781157] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1293.781158] CR2: 0000000000000959 CR3: 0000000102bc2003 CR4: 00000000003706e0
[ 1293.781159] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 1293.781160] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 1293.781162] Fixing recursive fault but reboot is needed!
[ 1293.781166] sched: RT throttling activated
```

tattsan, 2021-01-25:
I confirmed the issue has been fixed in the kernel 5.10.10.

NiklasBeierl, 2021-01-25:
Affirmative, 5.10.10-arch1-1 works for me as well. Thanks for the quick fix.

jigpu, 2021-01-25:
> I've confirmed the issue has been fixed in the kernel 5.10.10.

> Affirmative, 5.10.10-arch1-1 works for me as well. Thanks for the quick fix.

That is... Unpexpected. While I have sent my fix upstream, it has not yet been integrated. I would expect 5.10.10 to be just as broken as previous versions.

bleaktwig, 2021-01-25:
> I've been able to reproduce a driver crash and was indeed able to pin the cause on that particular commit. Can you please test my "fix-230" branch of input-wacom by cloning my code with the [...]

This fix works with the kernel 5.10.10. Thanks!

tattsan, 2021-01-25:
>That is... Unpexpected. While I have sent my fix upstream, it has not yet been integrated. I would expect 5.10.10 to be just as broken as previous versions.

It was not the Linux vanilla source,  but the one patched by Arch Linux.
I apologize for giving inaccurate information.

jigpu, 2021-01-26:
Patch has been accepted by the subsystem maintainer; now waiting on Linus and then the stable kernel trees after that.

jigpu, 2021-02-03:
Patch has been accepted by Linus and the stable trees. Should be fixed in the following kernel versions:
* Linux 5.11-rc6
* Linux 5.10.12
* Linux 5.4.94
* Linux 4.19.172 

A new input-wacom release (version 0.47.0) was also made which includes the fix, if your distro does not provide one of these kernel updates in a timely manner.

closed 2021-02-03 by jigpu