linuxwacom/input-wacom #231: Intuos BT S problems with Ubuntu 20.04

jigpu, 2021-01-21:
Hello, I own the same tablet [Intuos BT S; Ubuntu 20.04] and installing the database does not make it visible. Bluetooth connects fine but as the system does not recognise it, it disconnects after a few seconds.

_Originally posted by @rav0rmat in: https://github.com/linuxwacom/input-wacom/issues/202#issuecomment-764729121_

jigpu, 2021-01-21:
I suspect that your issue is with the Bluetooth stack rather than our driver since the Bluetooth should not disconnect on its own. Please reboot, attempt to attach the tablet over Bluetooth, and then (after it disconnects) run `sudo systemctl -b0 > attach.log`. Attach the log file to this issue and I'll see if I can find any abnormalities.

rav0rmat, 2021-01-21:
Something is wrong with this option.. 
_systemctl: invalid option -- 'b'_

jigpu, 2021-01-21:
Ah, sorry. Fingers were faster than my brain there :facepalm: 

You'll want to use `sudo journalctl -b0 > attach.log`

rav0rmat, 2021-01-21:
[attach.log](https://github.com/linuxwacom/input-wacom/files/5850976/attach.log)
Ok, done.

rav0rmat, 2021-01-21:
As I notice the error _No cache for C0:28:8D:43:29:5C_ , I found something related https://bbs.archlinux.org/viewtopic.php?id=212785 but setting the mode to either le or bredr makes either my mouse stop working or my keyboard. So that error is probably the root cause but I am yet to find a fix.

rav0rmat, 2021-01-21:
I am making some progress, I found something here https://stackoverflow.com/questions/59936665/bluez-crashes-after-pairing-an-with-my-ble-remote
I updated a line in _/lib/systemd/system/bluetooth.service_
[code]
ExecStart=/usr/lib/bluetooth/bluetoothd --compat --noplugin=sap -E 
[/code]

Now _journalctl -u bluetooth | tail_
gives me 
[code]
bt_uhid_send: Invalid argument (22)
[/code]

rav0rmat, 2021-01-21:
Ok, I caught my bluetooth services crashing and with them, my keyboard and mouse, I got the following: 

_ian 21 23:33:55 matthew-Z390-UD bluetoothd[11329]: bt_uhid_send: Invalid argument (22)
ian 21 23:33:55 matthew-Z390-UD bluetoothd[11329]: bt_uhid_send: Invalid argument (22)
ian 21 23:34:02 matthew-Z390-UD bluetoothd[11329]: Error reading PNP_ID value: Unexpected error code
ian 21 23:34:02 matthew-Z390-UD bluetoothd[11329]: HID Information read failed: Request attribute has encountered an unlikely error
ian 21 23:34:02 matthew-Z390-UD bluetoothd[11329]: Report Map read failed: Request attribute has encountered an unlikely error
ian 21 23:34:02 matthew-Z390-UD bluetoothd[11329]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
ian 21 23:34:02 matthew-Z390-UD bluetoothd[11329]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
ian 21 23:34:02 matthew-Z390-UD bluetoothd[11329]: Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
ian 21 23:34:02 matthew-Z390-UD bluetoothd[11329]: Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error_

Pinglinux, 2022-08-10:
Your issue is in the Bluetooth subsystem, not in Wacom driver. Can you reinstall or upgrade Ubuntu to make sure the system was not corrupted?

closed 2022-08-12 by Pinglinux