linuxwacom/input-wacom #232: AMD-Vi: Event logged IO_PAGE_FAULT issue reported by iommu caused by wacom driver

RemyL, 2021-01-23:
Hi,
I'm coming here after a lot of investigations and final help from i2c-amd-mp2 driver maintainer for linux.
Here are the needed informations:
Laptop: lenovo Yoga 530-14ARR (81H9) with amd ryzen 2500U and Vega 8 integrated with bios up-to-date.

Here is the answer from this maintainer on bugzilla: 
"What happens is that there should be "normal" IO page faults reported by the AMD iommu driver for Lenovo Yoga 530-14ARR users. dmesg from 5.7 kernels and older contain something like:

[] AMD-Vi: Event logged [IO_PAGE_FAULT device=00:00.0 domain=0x0000 address=0xffdefa80 flags=0x0050]
[] AMD-Vi: Event logged [IO_PAGE_FAULT device=00:00.0 domain=0x0000 address=0xffdefaa0 flags=0x0050]
(7 more lines)
[] AMD-Vi: Event logged [IO_PAGE_FAULT device=00:00.0 domain=0x0000 address=0xffde9000 flags=0x0050]

These page faults are expected, they've been there since the beginning for the Yoga 530-14ARR and aren't due to the I2C layer but to the Wacom touchscreen itself (more likely) or the wacom driver handling it (much less likely, IIRC), one HID report during initialization specifically. Since the errors seemed harmless, I didn't push the investigation further back then.

But now since 5.8, amd_iommu_report_page_fault segfaults before reporting the IO page faults. From a quick look it seems that dev_iommu_get() wasn't called yet to initialize dev->iommu for the touchscreen device? Is that possible?

Anyway it's clearly an iommu driver bug.

As a temporary workaround for those who don't want to compile a kernel, if you mostly need the touchpad and can live without the touchscreen, blacklisting the wacom module makes suspend/resume work again."

I was wondering if there could a solution from wacom driver, even if it said that it was more likely coming from the touchscreen itself.

Blacklisting wacom driver solves the issue but obviously the touchscreen doesn't work.

jigpu, 2021-01-26:
There have been a few other reports of suspend/resume issues with various tablet PCs, and blacklisting has helped in some of those cases. Its also been mentioned that the behavior changed at some point in the past, which matches up with maintainer's observation of a segfault beginning in Linux 5.8. I have to wonder if this might be the same problem as those other issues...

I'm far from an IOMMU expert, so I'd have to consult with that subsystem's maintainer to see if there's anything that can be done within the Linux driver. It sounds the IO_PAGE_FAULT errors are an issue either with how the system firmware sets up memory maps or with the device firmware. In either case, I don't think there's anything we can do from the driver level, as wonderful as that would be...

I'll get in touch with the maintainer and update this issue as I learn more.

RemyL, 2021-01-26:
I hope that all my investigations can lead to a solution and help other people.
However, i'm totally lost here and i'm just ready to provide any test or logs that developpers may want from me here or in bugzilla.

One weird thing is that even if i blacklist wacom module, touchscreen is still working.

jigpu, 2021-01-26:
Would you mind following the ["Contributing" instructions](https://github.com/linuxwacom/wacom-hid-descriptors/#contributing) for our wacom-hid-descriptors repository so that we have some additional data on your system model? It'll help us understand what hardware we're dealing with, and will let us compare to other systems.

RemyL, 2021-01-27:
Hello,
here is the file,I hope it will help.

[sysinfo.ouYB4t3irI.tar.gz](https://github.com/linuxwacom/input-wacom/files/5878832/sysinfo.ouYB4t3irI.tar.gz)

Pinglinux, 2022-08-04:
When wacom driver is in the blacklist, hid-generic driver takes care of the touchscreen/tablet. This issue is really outside of the input-wacom's scope. We can't do much about it. The later kernel releases may have fixed the issue.

closed 2022-08-04 by Pinglinux