linuxwacom/input-wacom #233: Wacom tablet no longer detected by Ubuntu 20.04

valeryroy, 2021-01-25:
Hi There.
I use a tablet Wacom One 13" connected to a ThinkPad P50 with Ubuntu 20.04. The configuration was working perfectly well until a few weeks ago, when Ubuntu released a kernel update.
I recompiled   input-wacom driver to no avail: the tablet is not recognized. The compilation does not report any error.

grep "" /sys/module/wacom*/version
yields
v2.00-0.46.0

modinfo wacom | grep version
yields
version:        v2.00-0.46.0
srcversion:     70361D1DF0C8A4D7C20861E

sudo rmmod wacom
yields
rmmod: ERROR: Module wacom is not currently loaded
sudo rmmod wacom_w8001
rmmod: ERROR: Module wacom_w8001 is not currently loaded

I would appreciate any suggestion on how to make this work again.
Thank you.

jigpu, 2021-01-25:
Can you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach its output here? This will provide us with a copy of several important logs that should help with diagnosis.

valeryroy, 2021-01-25:
Here is the output of the script.
[sysinfo.gP5g5urbJH.tar.gz](https://github.com/linuxwacom/input-wacom/files/5870009/sysinfo.gP5g5urbJH.tar.gz)

jigpu, 2021-01-26:
Can you explain a little more what you mean by "the tablet is not recognized" and "tablet no longer detected"? The system appears to detect it just fine: it is listed by both xsetwacom and libwacom-list-local-devices. Is there a specific program that the tablet does not work in? Are you able to use the tablet to navigate around the desktop at all?

valeryroy, 2021-01-26:
When I connect the tablet, nothing shows up on the tablet. I receive the notice "no external display". Could this mean that this is a problem with the video driver I'm using?
It use to plug in just fine.

valeryroy, 2021-01-26:
Fixed!
It was a problem with the nvidia driver which I reinstalled.
Sorry for the trouble. However you pointed me in the right direction.
Thank you.

jigpu, 2021-01-26:
Glad to hear its working again :)

closed 2021-01-26 by jigpu