linuxwacom/input-wacom #234: Tablet crashes after first stylus touch

ambrosebrown, 2021-01-26:
Using my Wacom Intuos M Pistachio tablet on Ubuntu 20.04.1 via USB cable it consistently crashes the first time the stylus touches the pad.   So the pointer moves when I hover over the pad but then as soon as the stylus touches the pointer then freezes and the pad no longer works (mouse does work though).  

I  run sudo rmmod wacom ; sudo modprobe wacom and then it usually logs me out (looks like the desktop restarts) and then comes back in works fine for a while.  Usually when I come back to it after some inactivity I get the same issue.  Sometimes it reboots when I run modprobe the whole machine crashes, reboots and then its fine.  The LED on the tablet is bright white/blue whilst this occurs.

I've seen similar reports elsewhere but no reasonable fix. e.g.

https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/3163

and

https://www.linuxquestions.org/questions/linux-hardware-18/touching-wacom-tablet-crashes-linux-4175571470/

The second link references a kernel patch which is beyond me!

Anyone able to help? I've got the following in apport.log

ERROR: apport (pid 106735) Mon Nov  9 16:59:41 2020: called for pid 1216, signal 6, core limit 0, dump mode 2
ERROR: apport (pid 106735) Mon Nov  9 16:59:41 2020: not creating core for pid with dump mode of 2
ERROR: apport (pid 106735) Mon Nov  9 16:59:41 2020: executable: /usr/lib/xorg/Xorg (command line "/usr/lib/xorg/Xorg vt2 -display
fd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3")
ERROR: apport (pid 106735) Mon Nov  9 16:59:41 2020: is_closing_session(): no DBUS_SESSION_BUS_ADDRESS in environment
ERROR: apport (pid 106735) Mon Nov  9 16:59:45 2020: wrote report /var/crash/_usr_lib_xorg_Xorg.0.crash

And a huge crash log in /var/crash ...

Any help appreciated!!

Thanks

jigpu, 2021-01-26:
The referenced kernel patch has been integrated for many years now (Linux 4.4) so there's nothing you need to worry about there. The log snippet you've provided shows that the Xorg process is crashing, which could be either due to a bug in xf86-input-wacom or some other low-level component.

To get some useful information from the crash dump, please try running `apport-retrace -S system --cache ~/.cache/apport-retrace --stdout /var/crash/_usr_lib_xorg_Xorg.0.crash`. This should download and install all the debug packages required and then print out the stack traces that we can look through. Copy/paste the output from this command and we'll take a look. If its later necessary to upload the full crash dump for deeper analysis, I'll let you know how.

ambrosebrown, 2021-01-26:
Brilliant, many thanks - I'll upload as soon as I get it, much appreciated

ambrosebrown, 2021-01-26:
Ok, I installed apport-retrace, I also had to add -R to the command to prevent this error:
ERROR: report file does not contain one of the required fields: Package
and then I also uncommented all the deb-src lines in /etc/apt/sources.list    
  (i did these 2 things from just looking at instructions on using apport-retrace on google...)  I now get the following:
`
sudo apport-retrace -R -S system --cache ~/.cache/apport-retrace --stdout /var/crash/_usr_lib_xorg_Xorg.0.crash

Installing extra package xserver-xorg-core to get ExecutablePath

E: Cannot find version '2:1.20.9-2ubuntu1.2~20.04.1' of package 'xorg-server'

E: Unable to find a source package for xorg-server

gpgv: Signature made Mon 18 Jan 2021 13:44:51 GMT

gpgv:                using RSA key 50C4A0DDCF31E452CEB19B516569D855A744BE93
gpgv: Can't check signature: No public key
dpkg-source: warning: failed to verify signature on /tmp/tmpulnmm7sg/xorg-server_1.20.9-2ubuntu1.2~20.04.1.dsc
--- stack trace ---
#0  0x00007ff087ee718b in ?? ()
No symbol table info available.
#1  0x000000000a392400 in ?? ()
No symbol table info available.
#2  0x0000000000000000 in ?? ()
No symbol table info available.
--- source code stack trace ---
#0  0x00007ff087ee718b in ?? ()
#1  0x000000000a392400 in ?? ()
#2  0x0000000000000000 in ?? ()
`

Is there some more work I need to do to make the trace more useable?   Cheers...

jigpu, 2021-01-26:
Thanks. Looks like there was some trouble finding the debug packages, so the trace is missing most of its information. Try running `sudo apt update` to make sure your system has the latest package index, and then retry the apport-retrace command, first without the `-R` option, and then with it if the earlier "required fields" error is still there.

**Do not** run `sudo apt upgrade` to install any updates; we just want the system to know that they exist so apport-retrace might have a better chance of locating appropriate debug packages on its own.

Out of curiosity, how large is the /var/crash/_usr_lib_xorg_Xorg.0.crash file?

ambrosebrown, 2021-01-26:
Cool, I ran sudo apt update and then reran.  If I don't do -R then it exits immediately with 

ERROR: report file does not contain one of the required fields: Package

Otherwise I think the output is the same..

Installing extra package xserver-xorg-core to get ExecutablePath
E: Cannot find version '2:1.20.9-2ubuntu1.2~20.04.1' of package 'xorg-server'
E: Unable to find a source package for xorg-server
gpgv: Signature made Mon 18 Jan 2021 13:44:51 GMT
gpgv:                using RSA key 50C4A0DDCF31E452CEB19B516569D855A744BE93
gpgv: Can't check signature: No public key
dpkg-source: warning: failed to verify signature on /tmp/tmpbkeuej9g/xorg-server_1.20.9-2ubuntu1.2~20.04.1.dsc
--- stack trace ---
#0  0x00007ff087ee718b in ?? ()
No symbol table info available.
#1  0x000000000a392400 in ?? ()
No symbol table info available.
#2  0x0000000000000000 in ?? ()
No symbol table info available.
--- source code stack trace ---
#0  0x00007ff087ee718b in ?? ()
#1  0x000000000a392400 in ?? ()
#2  0x0000000000000000 in ?? ()

Looks like the crash file is about 5MB...

jigpu, 2021-01-26:
Hmm... Not really sure why its having so much trouble. Its possible that the crash dump was interrupted by the crash itself (I'm not sure if the apport background service relies on X or not), but that's just a wild guess. Since the crash file isn't too large, go ahead and email it to me and I'll see if I can get anything more out of it: jason.gerecke@wacom.com

ambrosebrown, 2021-01-26:
Brilliant thanks - emailed.

jigpu, 2021-01-26:
Thanks :) I've been able to get the requisite dbgsym and source packages installed and extract the data below. It appears that the `PlayReleasedEvents` function ends up trying to call a function pointer that is invalid. This triggers segfault which crashes the X server.

The question is: "Why is the function pointer invalid?" It isn't some obviously-bad pointer like NULL, so I suspect it is either uninitialized or corrupted. I'm not very familiar with this particular section of the X server but maybe @whot has some ideas of how things could go off the rails... Just to save him some searching; version 1.20.9 of the xserver appears to be in use.

~~~
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ff087ec6859 in __GI_abort () at abort.c:79
#2  0x0000563d7ebba1b0 in OsAbort () at ../../../../os/utils.c:1351
#3  0x0000563d7ebbf5e9 in AbortServer () at ../../../../os/log.c:879
#4  0x0000563d7ebc044a in FatalError (f=f@entry=0x563d7ebf4290 "Caught signal %d (%s). Server aborting\n") at ../../../../os/log.c:1017
#5  0x0000563d7ebb74f9 in OsSigHandler (unused=<optimized out>, sip=0x7ffe52909e30, signo=11) at ../../../../os/osinit.c:156
#6  OsSigHandler (signo=11, sip=0x7ffe52909e30, unused=<optimized out>) at ../../../../os/osinit.c:110
#7  <signal handler called>
#8  0x0000563d7fd86010 in ?? ()
#9  0x0000563d7ea5f527 in PlayReleasedEvents () at ../../../../dix/events.c:1263
#10 0x0000563d7ea6608c in ComputeFreezes () at ../../../../dix/events.c:1343
#11 0x0000563d7ea66268 in CheckGrabForSyncs (thisDev=thisDev@entry=0x563d80658ef0, thisMode=<optimized out>, otherMode=<optimized out>) at ../../../../dix/events.c:1423
#12 0x0000563d7ea6676c in ActivateKeyboardGrab (keybd=keybd@entry=0x563d80658ef0, grab=grab@entry=0x563d808cf030, time=..., passive=passive@entry=1) at ../../../../dix/events.c:1658
#13 0x0000563d7ea61e35 in ActivatePassiveGrab (device=device@entry=0x563d80658ef0, grab=grab@entry=0x563d808cf030, event=event@entry=0x7ffe5290bdd0, real_event=real_event@entry=0x7ffe5290bdd0)
    at ../../../../dix/events.c:3716
#14 0x0000563d7ea622f0 in CheckPassiveGrabsOnWindow (pWin=<optimized out>, device=device@entry=0x563d80658ef0, event=event@entry=0x7ffe5290bdd0, checkCore=checkCore@entry=0 '\000', 
    activate=activate@entry=1 '\001') at ../../../../dix/events.c:3964
#15 0x0000563d7ea6338c in CheckDeviceGrabs (device=device@entry=0x563d80658ef0, event=event@entry=0x7ffe5290bdd0, ancestor=ancestor@entry=0x0) at ../../../../dix/events.c:4046
#16 0x0000563d7eb40b2b in ProcessDeviceEvent (ev=ev@entry=0x7ffe5290bdd0, device=device@entry=0x563d80658ef0) at ../../../../Xi/exevents.c:1765
#17 0x0000563d7eb40d6c in ProcessOtherEvent (ev=0x7ffe5290bdd0, device=0x563d80658ef0) at ../../../../Xi/exevents.c:1868
#18 0x0000563d7eb6b265 in ProcessKeyboardEvent (ev=<optimized out>, keybd=0x563d80658ef0) at ../../../../xkb/xkbPrKeyEv.c:178
#19 0x0000563d7eb96d45 in mieqProcessDeviceEvent (dev=dev@entry=0x563d80658ef0, event=event@entry=0x7ffe5290bdd0, screen=screen@entry=0x563d7fdbf6f0) at ../../../../mi/mieq.c:491
#20 0x0000563d7eb96e59 in mieqProcessInputEvents () at ../../../../mi/mieq.c:551
#21 0x0000563d7ea9369d in ProcessInputEvents () at ../../../../../../hw/xfree86/common/xf86Events.c:151
#22 0x0000563d7ea55c83 in Dispatch () at ../../../../dix/dispatch.c:417
#23 0x0000563d7ea59f84 in dix_main (argc=12, argv=0x7ffe5290cbe8, envp=<optimized out>) at ../../../../dix/main.c:276
#24 0x00007ff087ec80b3 in __libc_start_main (main=0x563d7ea439f0 <main>, argc=12, argv=0x7ffe5290cbe8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffe5290cbd8)
    at ../csu/libc-start.c:308
#25 0x0000563d7ea43a2e in _start () at ../../../../dix/gc.c:814
~~~
~~~
(gdb) frame 8
#8  0x0000563d7fd86010 in ?? ()
(gdb) disassemble
No function contains program counter for selected frame.
(gdb) disassemble 0x0000563d7fd86010
No function contains specified address.
(gdb) disassemble 0x0000563d7fd86010, +16
Dump of assembler code from 0x563d7fd86010 to 0x563d7fd86020:
=> 0x0000563d7fd86010:	(bad)  
   0x0000563d7fd86011:	add    %al,(%rdi)
   0x0000563d7fd86013:	add    %al,(%rdi)
   0x0000563d7fd86015:	add    %al,(%rdi)
   0x0000563d7fd86017:	add    %al,(%rdi)
   0x0000563d7fd86019:	add    %al,(%rdi)
   0x0000563d7fd8601b:	add    %al,(%rdi)
   0x0000563d7fd8601d:	add    %al,(%rdi)
   0x0000563d7fd8601f:	add    %al,(%rdi)
(gdb) x/32b 0x0000563d7fd86000
0x563d7fd86000:	0x00	0x00	0x00	0x00	0x00	0x00	0x00	0x00
0x563d7fd86008:	0x91	0x02	0x00	0x00	0x00	0x00	0x00	0x00
0x563d7fd86010:	0x07	0x00	0x07	0x00	0x07	0x00	0x07	0x00
0x563d7fd86018:	0x07	0x00	0x07	0x00	0x07	0x00	0x07	0x00
~~~
~~~
(gdb) frame 9
#9  0x0000563d7ea5f527 in PlayReleasedEvents () at ../../../../dix/events.c:1263
1263	            (*qe->device->public.processInputProc) (qe->event, qe->device);
(gdb) list
1258	                    break;
1259	                }
1260	
1261	            }
1262	#endif
1263	            (*qe->device->public.processInputProc) (qe->event, qe->device);
1264	            free(qe);
1265	            for (dev = inputInfo.devices; dev && dev->deviceGrab.sync.frozen;
1266	                 dev = dev->next);
1267	            if (!dev)
~~~

whot, 2021-01-29:
Please start `screen`, run `libinput record > wacom.yml` in screen, select the device, and then reproduce the crash with as short an interaction as possible. If the first interaction doesn't trigger it, just cancel and restart `libinput record`. When it crashes, put the stylus away immediately and do not use it until you've re-logged in and cancelled `libinput record`. The goal is to have the *kernel* events that trigger the crash in X in the recording so we can replay it here and see what's going on. Debugging `PlayReleasedEvents()` is terrible even when it happens on a local machine, there's no hope detecting what's going on here remotely.

krillin666, 2021-01-31:
This is happening to me as well in a brand new Thinkpad X1 yoga.

krillin666, 2021-01-31:
Some updates:

I'm using Fedora 33 with the latest kernel available and system up to date. When I use the pen the system crashes (using Xorg) and in the system settings of gnome under the "Power" section the battery power of the stylus pen is 0%.

I then booted from a usb with a Live Fedora 33 and the pen is working fine, even under Wayland, and in the power settings the Pen has 100% battery !

Thank you.

ambrosebrown, 2021-01-31:
Thank you @whot - I managed to reproduce the crash and record it as instructed.  The attached file (renamed from yml to txt) captures the log.  

I hovered the pen for a few seconds but as soon as it touched the tablet it froze.  It's currently frozen (the rest of my computer working fine) - if you want me to check anything whilst its in this state let me know.  It appears to occur only after a significant period of inactivity (e.g ~a day) as I could not reproduce it yesterday when I tried every few hours but this morning it happened first time.   Let me know if I can provide anything else.

[wacom stylus crash.txt](https://github.com/linuxwacom/input-wacom/files/5899582/wacom.stylus.crash.txt)

whot, 2021-02-01:
Thanks for the recording, much appreciated. So, from what I can tell the data itself is fine, and playing this through `libinput debug-events` using `libinput replay` does not show anything either. I've played this on a GNOME Wayland session and a GNOME Xorg session (both Fedora 33) and couldn't rerproduce this.

There's a [kernel update to 5.10.11](https://bugzilla.redhat.com/show_bug.cgi?id=1918486#c14) that sounds quite similar, can you give that one a try please? It doesn't exactly explain the backtrace @jigpu found though.

ambrosebrown, 2021-02-01:
Thanks, I will give that a go.  If this works, is it possible to get it back ported into the Ubuntu LTS 5.4.0 kernel?  Otherwise, presumably, I'll need to keep rebuilding every time there is a kernel update? (If I understand correctly...)
Thanks.

ambrosebrown, 2021-02-01:
Ok, I've rebuilt input-wacom and I'm now running  v2.00-0.42.0.142.g6912b5b

If this works, is it possible to get it back ported into the Ubuntu LTS 5.4.0 kernel?  Otherwise, presumably, I'll need to keep rebuilding every time there is a kernel update? (If I understand correctly...)  many thanks..

whot, 2021-02-01:
For the Ubuntu LTS kernel you'll need to file a bug in launchpad, we don't really have control over that, sorry.

ambrosebrown, 2021-02-01:
Ok, I can do that - presumably I'll need to wait until this fix is released input-wacom first?  I'm not sure how the process goes, if you let me know if/how/when this fix gets approved in input-wacom I'm happy to make a request with the kernel devs at Ubuntu..

jigpu, 2021-02-03:
Did you find that running `v2.00-0.42.0.142.g6912b5b` has resolved your issue?

The fix included in `v2.00-0.42.0.142.g6912b5b` has been accepted upstream and by the stable kernels, so Ubuntu is likely to automatically pull into their own distro soon. We've also just made a new input-wacom release which includes the fix if Ubuntu's version remains broken for a while.

ambrosebrown, 2021-02-03:
Hi, that's great news - thanks.  I've had no issues so far and I've been using the tablet for a few days without reboot (and it's been left switched on and idle for 15+ hours which usually triggers it.)  I've no idea how modules get migrated into Ubuntu LTS releases, I will try and find out!

jigpu, 2021-02-04:
That's great to hear. I didn't expect this issue to be caused by #230 (which is what `v2.00-0.42.0.142.g6912b5b` was for) given the other bug reports pre-dating the bug, but its possible that something else was going on. I'll close this issue for now, but feel free to re-open if you experience it again.

ambrosebrown, 2021-02-04:
ahh!  i just got the error again, was about to post!!

ambrosebrown, 2021-02-04:
sorry, can we reopen please...  i picked up the pen and it crashed again - basically the tablet stops working (rest of the computer fine)
i then restarted wacom (I had to do this twice) and it came back to life

ambrosebrown, 2021-02-04:
I am attaching syslog as there may be some clues in there looking at it.  I the crash is approximately 16:01 to 16:02 and then you can see where I restart it.
[syslog.txt](https://github.com/linuxwacom/input-wacom/files/5927322/syslog.txt)

ambrosebrown, 2021-02-04:
Apologies was literally writing the post as you closed it! @jigpu !

ambrosebrown, 2021-02-04:
I checked and I'm still running the version that I compiled from github

ambrosebrown, 2021-02-04:
I notice in syslog that at 04:57 there is this:

Feb  4 04:57:14 dimanche /usr/lib/gdm3/gdm-x-session[1144]: (EE) Wacom Intuos BT M Pen cursor: Error reading wacom device : No such device
Feb  4 04:57:14 dimanche /usr/lib/gdm3/gdm-x-session[1144]: (II) config/udev: removing device Wacom Intuos BT M Pen stylus

(I am in bed at this time,the computer is idle, unless my cats are using it somehow ...)  Could it be the module is being unloaded?

jigpu, 2021-02-04:
No worries :smile: Can you let me know if the pen still works after the X server is restarted post-crash? #230 will cause the pen to stop working until you reboot, so if that's not the case for you then its _definitely_ something else.

ambrosebrown, 2021-02-04:
Everything is working fine now, no reboot - I just had to do sudo rmmod wacom ; sudo modprobe wacom  
I  had to do this twice (maybe three times) before the tablet started working again. ..  sometimes when i do this the machine completely reboots.

jigpu, 2021-02-04:
Thanks. Technically an rmmod/modprobe cycle could also fix #230 (though in my experience the kernel refuses to let you rmmod the module after the crash). Run `cat /sys/module/wacom/version` and let me know what it reports. I want to be sure the patched driver is actually in use and not "just" installed.

ambrosebrown, 2021-02-04:
v2.00-0.42.0.142.g6912b5b

jigpu, 2021-02-04:
Perfect, thanks. So definitely not #230 but instead something else. The `Error reading wacom device : No such device` messages are seen whenever disconnecting a device or going to sleep (since the device has to be disconnected first). I suspect, however, that the lines from 04:57 were triggered by fwupd (which is mentioned just a few lines above) since I believe it does a device reset as part of its probe process.

I still suspect something going on with the grabs rather than the driver itself, I think... But debugging that will be tricky, especially if @whot is having difficulty replicating...

whot, 2021-02-05:
I'm now vaguely wondering if this is a race where the device is frozen in response to a grab, gets unplugged and then unfrozen as the grab is released. That *could* be a bug in the server, getting a reliable test case for this is going to be... fun.

ambrosebrown, 2021-02-09:
It's happening fairly frequently - I tutor kids online using bitpaper; I tend not to use the pad (and sometimes the computer) all day until about 4pm when I pick up the pad just before class and test it - the last two days the pointer freezes and I have to restart the module - often after the first time the desktop exits to the login screen, I login and restart the module and then everything is fine.  I'm attaching the syslog file from today when this happened in case there are any clues...

[syslog_crash_snippet.txt](https://github.com/linuxwacom/input-wacom/files/5953586/syslog_crash_snippet.txt)

punyidea, 2021-04-09:
Possibly related to #246?

fili, 2021-04-12:
I originally posted https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/3163 and am still experiencing frequently this issue. 

I like to add (which I did not in the other thread) that I recently completely changed every bit of my hardware (from Lenovo workstation to a completely custom built PC), with the exception of the keyboard, screens and tablet, installed 20.04 LTS on the new hard drive and the issue is very much still alive. Restarting with `sudo modprobe wacom` does the trick often, but after a few times it will still crash nonetheless upon touching the tablet with the pen.

I am unable to manually trigger it, and it happens randomly. Meaning sometimes I go a few days without any crashes or frozen cursors, then I have it randomly (could be after not using the computer for an hour or longer, could be while using the computer- although the latter case happens less frequently).

ambrosebrown, 2021-04-12:
I still have the exact same issue and it's not clear to me if/when/how this will get fixed...  I've seen the following kernel patch which may be a solution but it's not clear if/when it'll reach the LTS kernel of Focal (looks like it's going in the HWE one for now from what I can understand)

https://bugs.launchpad.net/ubuntu/+source/linux-signed-hwe-5.8/+bug/1920632

punyidea, 2021-04-16:
In the link you sent, some users mention the recent kernel 5.8.0-49 fixed their issue. Have you been able to try it out? I myself have experienced a system crash after using my touchscreen with two fingers in this kernel but not an immediate crash.  If I have time, I could give more information in a separate bug report.

fili, 2021-05-17:
I am using `5.8.0-53-generic` in Ubuntu and this still happened today to me again.

fili, 2021-09-01:
FYI this is still happening on 

`Linux computer 5.11.0-27-generic #29~20.04.1-Ubuntu SMP Wed Aug 11 15:58:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

krummja, 2022-03-16:
Happening to me on `Linux 5.4.0-104-generic`

Pinglinux, 2022-07-26:
Does 5.18+ make a difference?

fili, 2022-07-26:
I just installed 22.04 and can't say yet. 

`Linux computer 5.15.0-41-generic #44-Ubuntu SMP Wed Jun 22 14:20:53 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

I did still have the issue on the previous Ubuntu versions, up to a few days ago.

fili, 2022-07-27:
OK, just happened again, but on kernel 5.15.0-41-generic (see previous post).

Lambo-555, 2022-07-31:
Hello! Sorry for awaiting.
Drivers now work well, but in random time type of device switch from
desktop type to tablet/android type.
I just use device keys combo - left and right buttons pushed in one time.
It helps.

Pinglinux, 2022-08-03:
The desktop to tablet/android mode change is another issue, which may be related to fwupd version is older than 1.62. Does it make a difference if  you try:

sudo systemctl stop fwupd
sudo systemctl disable fwupd

whot, 2022-08-03:
> sudo systemctl stop fwupd
> sudo systemctl disable fwupd

btw, shortcut as  single command only: `sudo systemctl disable --now fwupd` (works for enable too)

closed 2022-08-12 by Pinglinux