linuxwacom/input-wacom #238: compiling on fedora for android x86

StAletheus, 2021-02-16:
Following up from sourceforge link. i'm trying to compile and place the binary in its proper android x86 location... or does this require adding to android sourcecode and compiling my own kernel? or are there compilation tools FOR android?

jigpu, 2021-02-16:
You have two options for how to proceed. You can either 1) compile the entire kernel yourself or 2) compile _just_ this out-of-tree module and copy the binary over. The former is more complicated but less likely to have problems if done correctly. The latter is easier, but has a high chance of not working properly.

Before going that far though, assuming you are using the Android-x86 project, have you checked to see if the device works "out of the box"? I've checked its kernel sources and the Wacom drivers appear to be enabled by default. Also, the kernel should be new enough to automatically configure most Wacom devices that might be connected. If you haven't already given it a shot, I'd try booting the environment before going too far down this rabbit hole.

What is the Wacom device you're trying to use with this Android install? This may have some impact on exactly what needs to be done...
* The sensor integrated into the "Samsung Notebook 9 pro 15. (2017)" you mentioned in the other bug?
* The sensor integrated into the "Motion M1400" mentioned in the link the bug referred to?
* Some other integrated sensor?
* A USB peripheral tablet?

StAletheus, 2021-02-16:
i have read that it is functional out of the box, but i cant find the citation now.  The main problem is when i flip the keyboard over, (viriual keyboard engages fine), the pen input seems to stay in landscape mode, so when drawing it's offset by INCHES, and the x-y axis seems to still be offset by 90 degrees.  

where would i go about looking on the android partition for where this would go? looking into compiling this rn.

jigpu, 2021-02-23:
Unfortunately, installing an updated kernel driver won't fix this particular issue. The kernel driver is just responsible for raw hardware communication. It sounds like the driver is working fine and producing the raw hardware events. The problem is that Android is not taking that raw data and rotating it to match whatever orientation it has flipped the screen into. You'll need to see if Android requires any configuration to have pen rotation work properly. Unfortunately, I'm not really sure where to start looking for information like that...

Pinglinux, 2021-09-15:
We probably don't want to mess up with an Android device ;)

closed 2021-09-15 by Pinglinux