linuxwacom/input-wacom #240: Build Error:   ‘KEY_ONSCREEN_KEYBOARD’ undeclared (first use in this function)

git-oliver, 2021-03-02:
Hello,
i use ubuntu 16.04 and try to build your software:
If I execute
`if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"`

I get the following error:

```
make[2]: Verzeichnis „/home/oliver/Downloads/xf86-input-wacom-0.40.0/src“ wird betreten
  CC       wcmUSB.lo
../src/wcmUSB.c: In function ‘usbParseBTNEvent’:
../src/wcmUSB.c:1644:8: error: ‘KEY_ONSCREEN_KEYBOARD’ undeclared (first use in this function)
   case KEY_ONSCREEN_KEYBOARD:
        ^
../src/wcmUSB.c:1644:8: note: each undeclared identifier is reported only once for each function it appears in
Makefile:553: die Regel für Ziel „wcmUSB.lo“ scheiterte
make[2]: *** [wcmUSB.lo] Fehler 1
make[2]: Verzeichnis „/home/oliver/Downloads/xf86-input-wacom-0.40.0/src“ wird verlassen
Makefile:519: die Regel für Ziel „all-recursive“ scheiterte
make[1]: *** [all-recursive] Fehler 1
make[1]: Verzeichnis „/home/oliver/Downloads/xf86-input-wacom-0.40.0“ wird verlassen
Makefile:428: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
Build Failed
```

What did I wrong?

I'm using
```
OS:      Ubuntu 16.04 xenial
Kernel: x86_64 Linux 4.15.0-136-generic
 ```

jigpu, 2021-03-03:
I think we're probably missing a definition of `KEY_ONSCREEN_KEYBOARD` in this driver for users with older kernel versions. Let me see if I can write a patch for you...

git-oliver, 2021-03-10:
I did a mistake...
The error comes if I build from source the `xf86-input-wacom` driver [1]
So the issue ticket should be moved to `xf86-input-wacom`

I cloud successfully install `input-wacom` [2]   but not `xf86-input-wacom`

I will close the ticket because I believe my kernel is just too old. Maybe I have to upgrade to the newest ubuntu version.
Thanks for all

[1] https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver
[2] https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

closed 2021-03-10 by git-oliver