linuxwacom/input-wacom #241: Pen not detected issue (cintiq pro 13 on ubuntu 18.04.5 LTS)

sinamone, 2021-03-06:
Hi everyone,
I have a problem with my cintiq 13 pro, which is just detected as a screen and not as a cintiq (and the pen stop being detected to).
I am using a wacom link plus adaptator. It used to worked perfectly well until few weeks ago. I may have update something wrong ?
I don't understand what's the problem or what should i do and i am afraid of destroying everything if i just try by myself.
Could someone help me please ?

I have look on the different issues topics and some of them seems similar to mine but i am an ubuntu baby user so i didn't understand a single word i have read.
Sorry if I am not writing at the good place.

dtskinner, 2021-03-09:
I have a Cinetiq 16 and have just had the same problem! It has worked fine for months and today has suddenly stopped picking up the pen.

When I contacted WACOM support they suggested looking at all the technical driver stuff (which I don't understand either!!) but the thing that worked was simply plugging into a different USB port! 

It's certainly worth a try and will hopefully save you some pain!!

sinamone, 2021-03-09:
Hahaha best solution ever. It works. Thanks a lot <3

dtskinner, 2021-03-09:
> Hahaha best solution ever. It works. Thanks a lot <3

Glad it worked!

sinamone, 2021-03-14:
Hello again !
Well... Changing the usb port have worked for.. 5 days... And now... I am back to the same prolem 😭
I did not change anything during this 5 days, except unpluging the usb port once to plug my printer.

I do suspect a driver problem (but what i am suppose to do with that... I don't know... Sound so mysterious...).

Could anyone help me please ?

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Pinglinux, 2021-09-17:
It sounds like a system/USB port issue. Did reboot the system make a difference? I suspect somehow something got corrupted. If that is the case, disconnect the tablet along cycling its power OR/AND reboot the system should help.