linuxwacom/input-wacom #242: Wacom pen tablet CTL-472 sometimes stop working

MasterLucas, 2021-03-10:
Dear linuxwacom community,

I am having some issues with my pen tablet. Sometimes, when I plug it into my laptop, it does not work. To be more precise, when I plug it, the light showing that the device is on twinkles twice, and then it turns off completely. But sometimes, like now, I plug the tablet, and it works perfectly. 

Do you have some idea about what may be happening? Is some driver that I do not have installed? or is it a hardware problem?

Pinglinux, 2021-09-15:
Well, the first step for us to troubleshoot your issue is to make sure your systen is running with the latest Wacom kernel driver. If you are not running the latest version of your Linux distribution, please update the kernel driver by the steps at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

ltsdw, 2021-11-17:
Hi there @Pinglinux, I think I'm having the same problem above.

Yes I have the latest driver.

It works sometimes, it's totally random, I have to plug and unplug it until it connects (it can take 20, 30, 40 tries before it gets to work), sometimes, when it connects it maps things wrong. Like, if I move the pen up it goes down, if I move the pen down it goes up, etc.

```
uname -a
Linux shadow 5.15.2-arch1-1 #1 SMP PREEMPT Fri, 12 Nov 2021 19:22:10 +0000 x86_64 GNU/Linux
```

```
Name            : input-wacom-dkms-git
Version         : 0.48.0.10.g59402f2-1
```

```
lsusb | grep -i wacom
Bus 002 Device 008: ID 056a:037a Wacom Co., Ltd CTL-472 [One by Wacom (S)]
```

```
xsetwacom list devices
Wacom One by Wacom S Pen stylus 	id: 15	type: STYLUS    
Wacom One by Wacom S Pen eraser 	id: 16	type: ERASER
```

```
[   49.751804] usb 2-2: Manufacturer: Wacom Co.,Ltd.
[   49.757681] input: Wacom Co.,Ltd. CTL-472 Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:056A:037A.0003/input/input21
[   49.758714] hid-generic 0003:056A:037A.0003: input,hiddev96,hidraw2: USB HID v1.10 Mouse [Wacom Co.,Ltd. CTL-472] on usb-0000:00:14.0-2/input0
[   49.760630] hid-generic 0003:056A:037A.0004: hiddev97,hidraw3: USB HID v1.10 Device [Wacom Co.,Ltd. CTL-472] on usb-0000:00:14.0-2/input1
[   49.847972] wacom: loading out-of-tree module taints kernel.
[   49.848101] wacom: module verification failed: signature and/or required key missing - tainting kernel
[   49.849510] input: Wacom One by Wacom S Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:056A:037A.0003/input/input23
[   49.849635] wacom 0003:056A:037A.0003: hidraw2: USB HID v1.10 Mouse [Wacom Co.,Ltd. CTL-472] on usb-0000:00:14.0-2/input0
[   75.691846] usb 2-2: Manufacturer: Wacom Co.,Ltd.
[   75.697947] input: Wacom One by Wacom S Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:056A:037A.0005/input/input29
[   75.698077] wacom 0003:056A:037A.0005: hidraw2: USB HID v1.10 Mouse [Wacom Co.,Ltd. CTL-472] on usb-0000:00:14.0-2/input0
[   81.877793] wacom 0003:056A:037A.0005: wacom_set_report: ran out of retries (last error = -110)
[  511.742498] usb 2-2: Manufacturer: Wacom Co.,Ltd.
[  511.748320] input: Wacom One by Wacom S Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:056A:037A.0007/input/input35
[  511.748430] wacom 0003:056A:037A.0007: hidraw2: USB HID v1.10 Mouse [Wacom Co.,Ltd. CTL-472] on usb-0000:00:14.0-2/input0
[  517.927754] wacom 0003:056A:037A.0007: wacom_set_report: ran out of retries (last error = -110)
[  558.466451] input: Wacom One by Wacom S Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/0003:056A:037A.0009/input/input41
[  558.467005] wacom 0003:056A:037A.0009: hidraw2: USB HID v1.10 Mouse [CTL-472] on usb-0000:00:14.0-2/input0
```

ltsdw, 2021-11-18:
I wanted to test out an older version, like 0.43.0, is there a way to compile using clang (I'm using a custom kernel compiled with clang)?

I tried but it always result in:

```
$ make CC=clang LDFLAGS="-fuse-ld=lld"

...

ld: unrecognised emulation mode: llvm
Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu elf_l1om elf_k1om i386pep i386pe
make[4]: *** [scripts/Makefile.build:298: /home/little/input-wacom/4.5/wacom.lto.o] Error 1
make[3]: *** [Makefile:1868: /home/little/input-wacom/4.5] Error 2
make[3]: Leaving directory '/usr/lib/modules/5.15.2-1-ltsdw-lto/build'
make[2]: *** [Makefile:40: all] Error 2
make[2]: Leaving directory '/home/little/input-wacom/4.5'
make[1]: *** [Makefile:379: all-recursive] Error 1
make[1]: Leaving directory '/home/little/input-wacom'
make: *** [Makefile:310: all] Error 2
```

I'll try with a kernel compiled with gcc later.

ltsdw, 2021-11-19:
tried on fedora, and it worked flawless... maybe is something on this arch installation, I'll try downgrading some stuff and see what I get. Also tried kernel 5.10.79-1-lts, but the result was the same, at least I know that is not the kernel, and not the tablet.

ltsdw, 2021-12-07:
@Pinglinux sorry to ping you again

Recently tested on EndeavourOS live usb, and it worked normally there... I noticed that in Endeavour the xorg.conf use libinput as the driver to tablets so I tried:

```
Section "InputClass"
    Identifier "Wacom Tablet"
    MatchIsTablet "on"
    Driver "libinput"
    MatchDevicePath "/dev/input/event*"
EndSection
```

but the result was the same.

for now I have (but it isn't necessary, even without it it will work).
```
Section "InputClass"
    Identifier "Wacom Tablet"
    MatchIsTablet "on"
    Driver "wacom"
    MatchDevicePath "/dev/input/event*"
EndSection
```

I simply don't know what could be going wrong, all logs, dmesg, journalctl, xsetwacom, all report my device, it simply doesn't respond. It will only respond after trying removing and loading the modules about 2 or three times (it is more consistent this way than plugging and unplugging the device, because unplugging and unplugging may take more retries than simply removing and loading the module again), then it will work.

Is there a better way to debug?

ltsdw, 2021-12-07:
Here is also the logs from sysinfo script
[tablet-WORKING-sysinfo.oQIZzWVgdt.tar.gz](https://github.com/linuxwacom/input-wacom/files/7671262/sysinfo.oQIZzWVgdt.tar.gz)

[tablet-NOT-working-sysinfo.Rx3PlKY5uy.tar.gz](https://github.com/linuxwacom/input-wacom/files/7671275/sysinfo.Rx3PlKY5uy.tar.gz)

jigpu, 2021-12-09:
Please install the "hid-recorder" tool (available in the "hid-replay-git" AUR package) and perform the following steps:

1. Run `sudo hid-recorder`
2. Select the One by Wacom device from the list and press Enter.
    * You should see a bunch of information be printed out, and then even more information whenever the pen is being used.
3. Try to use the pen and see if new data is printed out as the pen moves around

When the tablet is working you should see events printed out at a very rapid pace whenever the pen moves. When the tablet is not working, it would be interesting to know if any events are printed out at all. If they aren't, then something is causing the tablet to stop sending data; if they are, then something is causing the driver to stop relaying the data.

ltsdw, 2021-12-09:
Hi, thank you for your answer

when the tablet is working as you said, there's a lot of info being printed out very fast, which also happens when the tablet isn't working but there's something different in the info being printed (in the format I say). Like:

here's some info when the tablet is working:

```
...
E: 10.100114 10 c0 00 00 00 00 00 00 00 00 01
E: 11.693974 10 02 80 00 00 00 00 00 00 00 00
E: 11.701970 10 02 80 00 00 00 00 00 00 00 00
E: 11.708051 10 02 80 00 00 00 00 00 00 00 00
...
```

here it's not working:

```
...
E: 0.000001 10 c0 00 00 00 00 00 00 00 00 00
E: 5.049872 10 c0 00 00 00 00 00 00 00 00 00
E: 6.407767 4 01 00 00 00
E: 6.587817 4 01 80 03 06
E: 6.593760 4 01 80 00 0c
...
```

ltsdw, 2021-12-10:
I tried, just for the matter of testing, commenting out all my kernel parameters (nothing changed).

Also I don't know if this changes anything, but, if I plug it, don't interact with it, run the hid-recorder and wait just a bit. It'll always work.

this is the info (while idle) that is reported

```
E: 0.000000 10 c0 00 00 00 00 00 00 00 00 01
```

besides that it ends with 01, is a confirmation that I can interact with the tablet and it will work.

but if I interact with the tablet before running hid-recorder.

```
E: 0.000001 10 c0 00 00 00 00 00 00 00 00 00
```

it will end with 00 and will not work. But maybe it's just a coincidence.

but I tried 10 times running the hid-recorder before interacting with the tablet, and it worked all the 10, and 10 interacting with the tablet before running hid-recorder and it didn't worked.

Also the time that I wait for interacting with the tablet doesn't seem to matter, as long as I run hid-record before interacting with it.

jigpu, 2021-12-10:
The final bit of that message does seem to match up with the documentation: it says what mode the tablet it is in. Unfortunately, the docs also say that it is always "1", and I have no idea how it would get to be "0" when I can't even set that manually. Given that you say it will eventually switch to mode 1 if you wait long enough without touching it, it sounds like some kind of initialization bug but that still doesn't give me enough to dig in without being able to replicate locally.

ltsdw, 2021-12-10:
That's ok, thank you for your help.

I guess I'll try nuking the OS partition and starting a new one and see if it changes something.

ltsdw, 2021-12-11:
Fresh arch install... sadly it didn't worked neither... same behaviour... I'm out of ideas now.

pacstrap base, xorg, linux, linux-headers, grub, dwm, netctl, dnscrypt, lvm2, mesa, lib32-mesa, make, wpa_suplicant (and some files which sets my environment variables for my zsh setup)... I think these are all things that I installed/changed, the rest is default. No ideia what could be.

jigpu, 2021-12-13:
Can you check if fwupd is installed on your system? That should be the only userspace component that ever probes the hardware (even though fwupd should never probe this particular device...)

ltsdw, 2021-12-13:
nope, no fwupd installed 😭

jigpu, 2021-12-17:
@ltsdw We just received another report of the same odd behavior with a possible workaround. If you have Windows on your machine, you might want to read #294.

ltsdw, 2021-12-17:
thank you for letting me know this. As I don't have/use Windows, I can't test it, but I think it was just luck, like I reported above, it does work normally on Fedora or EndeavourOS, and sometimes if I boot up (on Arch) while the tablet is connected it will work normally until I disconnect and connect it again (or maybe not, it's somewhat random). So I think maybe is the same behavior while using Windows and then coming back to the distro affected.

jigpu, 2022-01-07:
From #294 (which has sysinfo logs from "unresponsive" and "working" instances of the CTL-472):
> There seem to be some curious differences in how the device responds when it is first discovered on the USB bus. Specifically, it seems that in the "unresponsive" case the tablet is able to report back its "product" string (i.e. "CTL-472") but not in the "working" case. The missing product string occurs both in the dmesg output as well as the verbose lsusb output.
>
> The next difference is that in the "unresponsive" case, once our our driver loaded we're unable to command the tablet into its normal operating mode: `wacom 0003:056A:037A.0008: wacom_set_report: ran out of retries (last error = -110)` Being in the wrong mode would explain why we're getting weird data, but then the question becomes why the command fails. The error number "-110" is a timeout, usually indicating the device doesn't like the request. It could be some kind of firmware issue given the difference in behavior when first discovering the device...

SweepingBishops, 2022-02-22:
I'm having the same issue.  
But mine isn't showing on xsetwacom either.  
Everything else is the same.

liamtruter, 2022-02-26:
I wrote a small script that runs on startup to make sure it's in the right mode. It's almost definitely not the best way to do it, but it works for me. Basically just using Itsdw's solution using rmmod and modprobe until the tablet is reporting the correct sequence of bytes at idle, but sometimes the port needs to be reset for xsetwacom to detect the device. Feedback is welcome.

```
#!/bin/bash

if [ "$EUID" -ne 0 ]
  then echo "Please run as root"
  exit
fi

update_idle_bytes (){
	idle_bytes=$(usbhid-dump -s "$device" -e stream -t 10000 | grep -m 1 "C0 00 00 00 00 00 00 00 00 0")
}

device=$(lsusb | grep CTL-472 | sed -e 's/:.*//g' -e 's/[A-Z][a-z]* 0*//g' -e 's/ /:/')

bus=$(echo $device | sed -e 's/:.*//g')
devnum=$(echo $device | sed -e 's/.*://g')
echo "$bus"
echo "$devnum"

port=$(lsusb -t | sed -n "/Bus 0$bus/{:start /Dev $devnum/!{N;b start};s/.*Port \([0-9]*\): Dev $devnum.*/\1/p}")

echo $port

update_idle_bytes

while :
do		
	tablet_id=$(xsetwacom --list devices | grep STYLUS | tr -cd "[:digit:]")
	echo "$idle_bytes"
	echo $tablet_id
	if [[ "$idle_bytes" == *01 && $tablet_id != "" ]]; then
		echo "success"
		break
	elif [[ "$idle_bytes" == *01 && $tablet_id == "" ]]; then
		echo "reset port..."
		echo "$bus-$port" > /sys/bus/usb/drivers/usb/unbind
		echo "$bus-$port" > /sys/bus/usb/drivers/usb/bind
		sleep 4
	else
		echo "reset module..."
		rmmod wacom && modprobe wacom
		update_idle_bytes
		sleep 4
	fi
done

echo "configuring tablet settings"
xsetwacom --set $tablet_id MapToOutput DP-3
xsetwacom --set $tablet_id Area 0 0 18500 8550```

SweepingBishops, 2022-02-26:
After the latest libinput update everything seems to be working fine. For now at least...  
I'll update if it stops working again.

ltsdw, 2022-04-20:
sorry for bumping you guys again.

I'm using the [OpenTabletDriver](https://github.com/OpenTabletDriver/OpenTabletDriver) and it's working quite well (besides the systemd service always failing, I can run the daemon manually). So if anyone still having problems try that out.

Pinglinux, 2022-07-15:
Looks like the issue was resolved by either a newer libinput or OpenTabletDriver.

closed 2022-07-15 by Pinglinux