linuxwacom/input-wacom #243: 2760p

RickySupriyadi, 2021-03-12:
using HP Elitebook 2760p installed

jigpu, 2021-03-12:
Please run our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/82fb9362998dd8ca582b4efb558ec3abfb11668d/capture.sh) and select the touch device from the list. When it says the recording has started, reproduce the issue and then hit CTRL+C to stop recording. Attach the generated tarball to this issue for review.

RickySupriyadi, 2021-03-13:
I'm sorry I'm new to linux and still learning but i only able to get this far...

> ID number of the device to record: 15
Found evdev device node: /dev/input/event17
ls: cannot access '/sys//devices/pnp0/00:04/tty/ttyS4/serio5/input/input23/event17/device/device/hidraw/': No such file or directory

i tried to open the folder and can't find it, it only reach
>/sys//devices/pnp0/00:04/tty/ttyS4/serio5/input/input23/event17/device/device/

/hidraw folder is missing

well i only get this far sorry if this taking more of your time.

jigpu, 2021-03-15:
My apologies; the script did not support the old UART-style devices that your machine appears to use. Please try [this updated capture.sh](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/a29ec841671cc6a785cd308326e8814ad97504d4/capture.sh) which should hopefully work. There's a chance it could fail again (since I don't have a UART-device to test the new code with), so please copy/paste any new error messages. If it begins recording, follow the same instructions I previously provided.

RickySupriyadi, 2021-03-16:
Thank you very much sir,
sudo bash capture.sh

> ⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech Wireless Receiver Mouse          id=9    [slave  pointer  (2)]
⎜   ↳ PS/2 Generic Mouse                        id=12   [slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad                id=13   [slave  pointer  (2)]
⎜   ↳ Wacom Serial Penabled 2FG Touchscreen Pen stylus  id=14   [slave  pointer  (2)]
⎜   ↳ Wacom Serial Penabled 2FG Touchscreen Pen eraser  id=17   [slave  pointer  (2)]
⎜   ↳ Wacom Serial Penabled 2FG Touchscreen Finger touch        id=15   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
↳ Power Button                              id=6    [slave  keyboard (3)]
↳ Video Bus                                 id=7    [slave  keyboard (3)]
↳ Sleep Button                              id=8    [slave  keyboard (3)]
↳ HP HD Webcam [Fixed]: HP HD Web           id=10   [slave  keyboard (3)]
↳ AT Translated Set 2 keyboard              id=11   [slave  keyboard (3)]
↳ HP WMI hotkeys                            id=16   [slave  keyboard (3)]

15

> ID number of the device to record: 15
Found evdev device node: /dev/input/event17
Found evdev sysfs at /sys/devices/pnp0/00:04/tty/ttyS4/serio5/input/input23/event17
Unable to find a hardware device from /sys/devices/pnp0/00:04/tty/ttyS4/serio5/input/input23/event17

RickySupriyadi, 2021-04-01:
well i have temporary solution so that the mouse cursor not erratic move around the screen and randomly clicking things... i just disable the touchscreen (https://invent.kde.org/kde/wacomtablet.git) and it temporary solve this.
however i still continue to figure out how to solve it.

closed 2021-09-17 by Pinglinux