linuxwacom/input-wacom #244: Q11K V2 Bluetooth Issue

jdavidsjaramillo, 2021-03-12:
Hi everyone, 

I've got a Huion Q11K V2 tablet, when plugged using the cable everything seems to work just fine, but when trying to use it via the bluetooth dongle it came with I get an issue when I try to use the Stylus, the cursor goes crazy and moves to the upper left side of the screen, and then almost immediately it goes to where it should really go. It doesn't seem like something that should bother, but it happens every time I let the Stylus get far enough to not get detected by the tablet and put it close again so it happens like each 5 seconds when using applications like blender, krita, etc. and it also happens almost every time I use the Stylus buttons, so if I try to rotate or pan in the same applications it will happen and move the view somewhere else making the usage of this software nearly impossible or at least very frustrating.

Right now my distro is Manjaro Linux, but I got the same problem with Pop_OS!, Ubuntu, Linux Mint, and also with OpenSuse Tumbleweed.

The version of the kernel I'm using is 5.11.0-1, but I also tried with 5.9.16-1.

I'm using an Asus Tuf FX504 laptop, and I also tried with a PC and had the same result.

Before moving again to Linux I was using Windows and I had not this problem, so I think there's no problem with the tablet itself.

Thanks in advance for your help.

jigpu, 2021-03-12:
Please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the output here. Its unlikely that your Huion tablet is using any of our drivers, but the output should let us know which drivers are in use so we can point you to the appropriate place.

jdavidsjaramillo, 2021-03-12:
Hello @jigpu , thank you so much for your help. The output is as follows:

[sysinfo.aoKA9nugrb.tar.gz](https://github.com/linuxwacom/input-wacom/files/6133428/sysinfo.aoKA9nugrb.tar.gz)

jigpu, 2021-06-18:
Apologies for the delay. It looks like your tablet is using the "uclogic" driver. I believe you'll want to follow up with the DIGImend project at https://digimend.github.io/. Their developer is more familiar with that driver and may be able to help with your issue.

closed 2021-06-18 by jigpu