linuxwacom/input-wacom #246: input-wacom 0.46.0 and 0.47.0 don't install correctly when multiple kernels present or less-functional module exists

rfs34, 2021-03-18:
Version 0.46.0 of input-wacom appears to have the following bugs, which do not seem to have been removed in 0.47.0:

1. /etc/depmod.d/input-wacom.conf gets to contain

   override wacom * extra

   which has several problems:

   a) It fails to override hid-wacom which may already exist - but not 
      work properly - in the existing list of kernel modules that come
      with the kernel; and
   
   b) If there is more than one kernel installed, and two are
      sufficiently close that weak modules will work with both, the
      module installation may correctly place the new hid-wacom.ko
      modules not in /lib/modules/<kernel-version>/extra but in a
      location pointed to by a symlink in
      /lib/modules/<kernel-version>/weak-updates . In that case we
      need - for the later of the two (or more) kernels but not for
      the first of such a set of kernels - not
   
      override wacom * extra
      override hid-wacom * extra
   
      but
   
      override wacom * weak-updates
      override hid-wacom * weak-updates
   
      In order to put this together so that it works with all of such a
      set of kernel versions, we need
   
      override wacom <kver-first> extra
      override hid-wacom <kver-first> extra
      override wacom <kver-next> weak-updates
      override hid-wacom <kver-next> weak-updates
      ... and so on for each further kernel version in the weak-compatible set.
   
      (also wacom and hid-wacom may not be the only two modules
      affected, e.g. is there also a wacom_w8001 module ?). 
   
2. The 3.17 Makefile recipe for making the target signature appears to
   have forgotten that hid-wacom.ko exists.

Please do tell me if you think I am wrong.

punyidea, 2021-03-30:
I'm not sure if I'm affected by this issue but I can confirm that not all functionality is present on my HP Zbook x360 after installing input-wacom 0.47.0. 
I have kernel 5.8.0-44-generic with stock driver. After upgrading to 5.8-0-45-generic I experienced issues, and tried installing input-wacom. This was unsuccessful.
Then I tried kernel 5.8.0-48-generic. I installed 0.47.0 and fixed some issues with my stylus. However, my touchscreen is not working.  I tried looking in syslog, and noticed that the name of my touchscreen had changed. However, I have not been able to successsfully troubleshoot the issues and instead keep on running kernel 5.8.0-44.

punyidea, 2021-04-09:
Thank you so much for your detailed comment! For the moment, I can confirm that on my system the new modules are not included in the kernel. I can see both the new and old modules in both `/lib/modules/<newkernelversion>/kernel`and `/usr/lib/modules/<newkernelversion>/kernel`, under the `/extra` and `/kernel` folders, respectively. However, running the lsinitrd command, I only see the old modules included. Therefore I think that this is the issue I am experiencing with this new kernel.

So far I have not been able to solve the issue. I am writing the following not necessarily to ask for help but to provide information to the developers.

I have modified the `/etc/depmod.d/input-wacom.conf` file to the following

    override wacom 5.8.0-45-generic extra
    override wacom 5.8.0-48-generic extra
    override hid-wacom 5.8.0-45-generic extra
    override hid-wacom 5.8.0-48-generic extra 
    override wacom_w8001 5.8.0-45-generic extra
    override wacom_w8001 5.8.0-48-generic extra

the output of `depmod --dry-run | grep 'wacom'`is

    kernel/drivers/input/tablet/wacom_serial4.ko:
    kernel/drivers/input/touchscreen/wacom_i2c.ko:
    extra/wacom_w8001.ko:
    extra/wacom.ko: kernel/drivers/hid/usbhid/usbhid.ko kernel/drivers/hid/hid.ko

however, after running `sudo dracut`and then  `sudo lsinitrd  /boot/initramfs-5.8.0-48-generic.img  | grep 'wacom'`
I can see that only the old file seems to be included, and searching for 'extra' gives me empty results. So it seems that the new drivers are still not being included in the kernel, with this new fix.

Note: For the moment, I cannot see the 'weak-updates' folder you mentioned.

As you point out, it would be helpful to have a comment from the developers to confirm your observations.

rfs34, 2021-04-09:
Please excuse me for asking what may be a silly question: Did you run 

depmod 5.8.0-45-generic
depmod 5.8.0-48-generic

(without the --dry-run flag) after modifying /etc/depmod.d/input-wacom.conf but before running dracut ? If not, you need to, because the /lib/modules/\<newkernelversion>/modules.dep and .../modules.dep.bin files need updating before dracut is run.

After running depmod, you should be able to see the new modules (but not the old ones) listed in the modules.dep file produced by depmod.

Please forgive me if I'm teaching my grandmother to suck eggs.

punyidea, 2021-04-09:
Haha, I wish that this solved my issue! Those tests were done to troubleshoot after the real run had no results ;)

rfs34, 2021-04-09:
There's clearly something wrong with the running of dracut. Did you remember to pass the desired kernel version to dracut as well as the full path to the initramfs-\<newkernelversion>.img that you want it to build ? E.g.

dracut /boot/initramfs-\<newkernelversion>.img \<newkernelversion>

? (Because otherwise it'll build an initramfs for the current kernel version named as if it's for the new kernel version.)

Another potential issue (that we haven't yet got to) is if you actually need a module called hid-wacom (which I do) - in which case why isn't it showing up in your /lib/modules/\<newkernelversion>/extra directory ? But first work out why dracut isn't including the new modules.

rfs34, 2021-04-13:
punyidea: 

Here's another idea as to what might be wrong when you run dracut, potentially also wrong of course:

Since about kernel version 4, dracut also needs to be told about modules that need adding via adding a file

/etc/dracut.conf.d/input-wacom.conf

containing a line something like

add_drivers+=" wacom wacom_w8001 "

Does that help you get dracut to include your new modules ? (You might still need one called hid-wacom, I don't know.)

punyidea, 2021-04-13:
We seem to have made some progress. I confirmed your statements about dracut, as they were in the manual for dracut.conf. (`man dracut.conf`).

After adding the line that you suggested to `/etc/dracut.conf.d/input-wacom.conf`, I was able to get the following modules to show up.

    usr/lib/modules/5.8.0-48-generic/extra/wacom.ko
    usr/lib/modules/5.8.0-48-generic/extra/wacom_w8001.ko
    usr/lib/modules/5.8.0-48-generic/kernel/drivers/hid/wacom.ko

However, it didn't seem that this was fixing my initial issue, and the drivers would not load properly. So my curiosity got me trying out the original install. I went to the directory where i had downloaded the drivers and built and installed as before, and this time it seemed to work! My guess is that this has to do with the dracut line. The next post will be a summary for my workaround.

punyidea, 2021-04-13:
A potential workaround for this issue, found 90% due to Roger's help! (@rfs34)

Run all of the steps below. Since I myself know not too much about what I'm doing, I recommend the use of caution if you run this

## 1. Dracut
Modify (or create) the file `etc/dracut.conf.d/input-wacom.conf` to include the line:
`add_drivers+="  wacom wacom_w8001 "`

## 2. Depmod

Modify the  `/etc/depmod.d/input-wacom.conf` file to include
    override wacom * extra
    override hid-wacom * extra
    override wacom_w8001 * extra
Note: to be safer, you can replace *  with the kernel version you'd like to update.

run `sudo depmod`.

## 3. Make and install the drivers

Now, follow the instructions on the page as usual to make and install the drivers on your machine. The current install scripts will overwrite  `/etc/depmod.d/input-wacom.conf` so do take note.

## Warnings and Caution

This is only based on my experience with the machine. It's possible that the depmod steps are unnecessary. I included them because on my machine I noticed that the install script skipped depmod on my machine! I also wanted to bring another possible conflict to people's attention, the [secure boot conflict I found in the wiki](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot). It's possible that in my case, this could have caused the issue, and that my workaround is a red herring.

Happy patching!

jigpu, 2021-06-23:
I've created a pull request as #257 which should address most of the issues brought up in this thread. The depmod configuration has been updated, and a new dracut configuration file created (along with a fix to actually run dracut on `make install`).

Note that I've deliberately left "weak-updates" out of the updated depmod configuration. It is inconvenient to have the tablet stop working after a kernel upgrade, but I feel that inconvenience of periodically recompiling / reinstalling input-wacom outweighs possible security issues not getting fixed because a user doesn't realize their system is using an out-of-date version of input-wacom.

Pinglinux, 2021-09-15:
Thank you @jigpu for your effort. Since we didn't hear any complaints after your pull request, I assume your fix made them all happy.

closed 2021-09-15 by Pinglinux