linuxwacom/input-wacom #247: Can't build from source

dpp23, 2021-03-22:
I am trying to install a newer version of `input-wacom` on Ubuntu as the latest available for Ubuntu currently is 0.39 and I am seeing issues similar to #230 

I am sure that the problem is in my setup, but can't figure out how to fix it. Please see the output of the build command:
```
~/Downloads/input-wacom-0.47.0 $ if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking if build path '/home/dimitar/Downloads/input-wacom-0.47.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/5.8.0-45-generic/build
checking kernel version... 5.8.0-45-generic
checking RHEL6 minor release... 0
checking RHEL7 minor release... 0
checking power supply version... pre-v4.1
checking abs accessor version... pre-v2.6.36
checking CONFIG_MODULE_SIG... y
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... 
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... off
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha512
checking for module signing hash algorithm... sha512
checking for kernel autogenerated key at /lib/modules/5.8.0-45-generic/build/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/5.8.0-45-generic/build/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... yes
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... yes
checking for module signing key... /var/lib/shim-signed/mok/MOK.priv
checking for module signing certificate... /var/lib/shim-signed/mok/MOK.der

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/5.8.0-45-generic/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.
[... 670 lines cut ...]
                        ^
In file included from ./include/linux/bug.h:5:0,
                 from ./include/linux/mmdebug.h:5,
                 from ./include/linux/gfp.h:5,
                 from ./include/linux/slab.h:15,
                 from ./include/linux/hid.h:19,
                 from /home/dimitar/Downloads/input-wacom-0.47.0/4.5/wacom_wac.h:9,
                 from /home/dimitar/Downloads/input-wacom-0.47.0/4.5/wacom_wac.c:11:
./include/linux/usb.h:1976:3: error: expected identifier or '(' before string constant
   WARN_ON(usb_pipeout(pipe));
   ^
In file included from <command-line>:0:0:
././include/linux/compiler_types.h:239:24: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 #define asm_inline asm __inline
                        ^
make[4]: *** [scripts/Makefile.build:286: /home/dimitar/Downloads/input-wacom-0.47.0/4.5/wacom_wac.o] Error 1
make[3]: *** [Makefile:1783: /home/dimitar/Downloads/input-wacom-0.47.0/4.5] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-5.8.0-45-generic'
make[2]: *** [Makefile:42: all] Error 2
make[2]: Leaving directory '/home/dimitar/Downloads/input-wacom-0.47.0/4.5'
make[1]: *** [Makefile:377: all-recursive] Error 1
make[1]: Leaving directory '/home/dimitar/Downloads/input-wacom-0.47.0'
make: *** [Makefile:308: all] Error 2
Build Failed
```

dpp23, 2021-03-22:
I've since realized that this probably will require building the kernel or parts of it as well. My issue was fixed by a kernel downgrade though, so that is now solved (the offending kernel was 5.8.0-45-generic). 

Apologies for the spam.

closed 2021-03-22 by dpp23

jigpu, 2021-03-23:
I'm surprised to see so many warning and error messages. I'd like to test the build on Ubuntu locally to see if it is a general problem or not. Can you let me know what version of Ubuntu you were using? The install process only builds this driver (not the kernel as a whole or other parts of it) so its usually pretty reliable...

dpp23, 2021-03-23:
@jigpu that was on Ubuntu 20.10