linuxwacom/input-wacom #250: Stuck in Android Mode when connected via Bluetooth - Intous BT M CTL6100WLE0

jsj1027, 2021-03-25:
When I connect the tablet with bluetooth it stays in Android mode. I can hold the first and last express keys but it never changes to PC mode.

If I connect with my wire, it starts in Android mode but I can shift it to PC mode by holding the first and last express keys for 3-4 seconds. Is there anyway I can fix this?

jsj1027, 2021-03-25:
I have found one additional weird thing the Android mode usb id is 2d1f:0378 and it has no listed bluetooth id. 
But im pretty sure even in bt mode my devices is being set as a 2d1f:0378 instead of the listed bluetooth id of 056a:0379.

As found in the libwacom data areas [p3-wl](https://github.com/linuxwacom/libwacom/blob/master/data/intuos-m-p3-wl.tablet) and [p3-wl-android](https://github.com/linuxwacom/libwacom/blob/master/data/intuos-m-p3-wl-android.tablet)

jigpu, 2021-03-25:
Could you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) while the tablet is connected over Bluetooth and attach the output here? Also, please let us know the symptoms that make you think your tablet is in Android mode. Android mode should only be possible when connected via USB, but its possible that similar symptoms may still be possible over Bluetooth.

jsj1027, 2021-03-25:
[sysinfo.JkhgHUVPCu.tar.gz](https://github.com/linuxwacom/input-wacom/files/6207684/sysinfo.JkhgHUVPCu.tar.gz)

Here is the output.
The symptoms that make me believe the tablet is in Android Mode is that the express keys no longer appear for mapping in Gnome settings for the Wacom Tablet. 
xsetwacom also lists a similiar small list of devices like it does in Android Mode.

## Devices in Bluetooth:
 
 Wacom Intuos BT M Pad pad       	id: 10	type: PAD       
 Wacom Intuos BT M Pen stylus    	id: 15	type: STYLUS
 
## Devices in Android Mode usb:

Wacom Co.,Ltd. Intuos BT M stylus	id: 10	type: STYLUS    
Wacom Co.,Ltd. Intuos BT M eraser	id: 15	type: ERASER

## Devices in PC Mode usb:

Wacom Intuos BT M Pad pad       	id: 10	type: PAD       
Wacom Intuos BT M Pen stylus    	id: 15	type: STYLUS    
Wacom Intuos BT M Pen eraser    	id: 16	type: ERASER    
Wacom Intuos BT M Pen cursor    	id: 17	type: CURSOR

## Main Issue

The main problem is that my buttons not longer are configurable.

jsj1027, 2021-03-25:
Okay getting asked this question led me to a question which led me to an answer.

"What if the buttons work? But you just cant configure them with the gui?"

So I tried it, I connected it with Bluetooth and used these commands to set up the buttons

xsetwacom set "Wacom Intuos BT M Pad pad" Button 1 "key +Control_L"
xsetwacom set "Wacom Intuos BT M Pad pad" Button 2 "key +Shift_L"
xsetwacom set "Wacom Intuos BT M Pad pad" Button 3 "key +Alt_L"
xsetwacom set "Wacom Intuos BT M Pad pad" Button 8 "key +Control_L +z -z -Control_L"

And it works!

But the question remains why buttons were unset after switching over to bluetooth. Im also going to do some testing to see if the buttons lose their settings if I disconnect and reconnect.

jsj1027, 2021-03-25:
Additional information gathered, the Express key settings reset when you disconnect and reconnect them via Bluetooth :(.

The usb settings dont do that. It's weird for sure.

Pinglinux, 2021-09-15:
Here comes my answer to your questions (you probably have already figured that out by now ;):

1. "Wacom Intuos BT M" doesn't support mode switch between Android and PC;
2. If you configured the buttons from GUI, it should stay with the GUI. If not, that would be a GUI bug;
3. If you configured the buttons with xsetwacom, you'll need to put those xsetwacom commands into a system start script so it can issue the xsetwacom commands for you when you login. Please take a look at below link for examples:
https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-2%3A-Example-scripts

closed 2021-09-15 by Pinglinux