linuxwacom/input-wacom #253: Ubuntu 20.04 crashes when ThinkPad Pen Pro touches screen of Thinkpad L13 Yoga Gen 2

therunomask, 2021-03-31:
While I am still able to click buttons on the screen using my fingers, the system crashes consistently every time the pen touches the screen. In this case, the screen freezes and the system does not react to anything I do, also ctr+alt+F3 does not change anything.

I used the pen until a few weeks ago, it was working fine then. When I tried again yesterday the bug appeared for the first time.

Any help would be appreciated.

lsitongia, 2021-04-07:
I just want to leave a "me, too". On this.

lsitongia, 2021-04-07:
I just want to leave a "me, too". On this. I don't see anything logged in the system or X11. It's really odd. I touch the screen with the active stylus and the system just locks up. Display doesn't change, still shows whatever was there. Keyboard is inactive. Mouse inactive. Internal or external, doesn't matter. Have to power-cycle the laptop.

punyidea, 2021-04-09:
I experienced this issue too, and reverted to kernel version 5.8.0-44-generic which did not have this issue.

punyidea, 2021-04-09:
See issue #234 as well?

lsitongia, 2021-04-09:
> See issue #234 as well?

In my case it is a Wacom digitizer in the screen. The laptop locks up. It is not possible to perform any commands. Must hard reboot.

therunomask, 2021-04-11:
> I experienced this issue too, and reverted to kernel version 5.8.0-44-generic which did not have this issue.
> My issue is in part related to #246 as I've already installed the new driver version 0.47.0.

While this does not fully resolve the issue, it's a good workaround for the time being. Thanks

punyidea, 2021-04-13:
#246 has a potential workaround if the drivers have been installed from source. (see secure boot link as well)

lsitongia, 2021-04-14:
I just got 5.8.0-49-generic as a regular update in the Ubuntu Software app, and that has fixed the problem.

therunomask, 2021-04-14:
> I just got 5.8.0-49-generic as a regular update in the Ubuntu Software app, and that has fixed the problem.

It worked for me too.

closed 2021-04-14 by therunomask