linuxwacom/input-wacom #254: Wacom one not recognized on Ubuntu 18.04.5 LTS

davidem88, 2021-06-03:
I installed the kernel driver, but my Wacom One is not detected even after rebooting the system. The tablet does not show in the list of devices if I use `lsusb` or `xsetwacom --list devices`. I attach the output of the driver(input-wacom-0.42.0) installation

```checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/5.4.0-73-generic/build
checking kernel version... 5.4.0-73-generic
checking RHEL7 minor release... 0
checking power supply version... v4.1+
checking abs accessor version... v2.6.36+

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/5.4.0-73-generic/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/davide/input-wacom-0.42.0'
Making all in 4.5
make[2]: Entering directory '/home/davide/input-wacom-0.42.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/5.4.0-73-generic/build M=/home/davide/input-wacom-0.42.0/4.5
make[3]: Entering directory '/usr/src/linux-headers-5.4.0-73-generic'
  Building modules, stage 2.
  MODPOST 2 modules
make[3]: Leaving directory '/usr/src/linux-headers-5.4.0-73-generic'
make[2]: Leaving directory '/home/davide/input-wacom-0.42.0/4.5'
make[2]: Entering directory '/home/davide/input-wacom-0.42.0'
make[2]: Leaving directory '/home/davide/input-wacom-0.42.0'
make[1]: Leaving directory '/home/davide/input-wacom-0.42.0'
Making install in 4.5
make[1]: Entering directory '/home/davide/input-wacom-0.42.0/4.5'
make -C /lib/modules/5.4.0-73-generic/build M=/home/davide/input-wacom-0.42.0/4.5 modules_install
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-73-generic'
  INSTALL /home/davide/input-wacom-0.42.0/4.5/wacom.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/davide/input-wacom-0.42.0/4.5/wacom_w8001.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-73-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-73-generic'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin" depmod -a 5.4.0-73-generic
/usr/sbin/update-initramfs -u -k 5.4.0-73-generic
update-initramfs: Generating /boot/initrd.img-5.4.0-73-generic
I: The initramfs will attempt to resume from /dev/nvme0n1p1
I: (UUID=a8d48686-4911-43ca-aa58-2866daab21df)
I: Set the RESUME variable to override this.
make[1]: Leaving directory '/home/davide/input-wacom-0.42.0/4.5'
make[1]: Entering directory '/home/davide/input-wacom-0.42.0'
make[2]: Entering directory '/home/davide/input-wacom-0.42.0'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/davide/input-wacom-0.42.0'
make[1]: Leaving directory '/home/davide/input-wacom-0.42.0' ```

jigpu, 2021-06-18:
Please follow these steps so we can get a clean log of what occurs when the device tries to connect:
1. Disconnect the Wacom One HDMI and USB plugs from the system
2. Reboot the PC
3. Connect the HDMI and USB plugs of Wacom One to the system
4. Run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh)
5. Attach the tarball that it generates to this issue by dragging & dropping it onto the reply box.

davidem88, 2021-06-19:
[sysinfo.5mGfJ2Jlk6.tar.gz](https://github.com/linuxwacom/input-wacom/files/6680407/sysinfo.5mGfJ2Jlk6.tar.gz)
Tarball attached, thanks.

jigpu, 2021-07-15:
Very strange... There doesn't seem to be any indication at all that the USB connection is being made in any of the logs. No hint of the device or even failure messages. I would recommend trying other USB ports, rebooting into a different OS (e.g. Windows or some other live USB Linux distro), and trying to use the device with other PCs. Also, it may be worth mentioning that the custom USB/HDMI cable that Wacom provided with the tablet needs to be used; a standard USB-C cable will not work from what I've been told (I don't know exactly what doesn't work, but I suppose it might cause similar symptoms).

Pinglinux, 2021-09-15:
@davidem88 Did Jason's suggestion fix your issue? If I don't hear from you in the next few days, I will assume your issue has been fixed.

closed 2021-09-17 by Pinglinux

davidem88, 2021-09-17:
Yes, I solved it thanks. It was indeed a problem with the UBS port.