linuxwacom/input-wacom #256: Lenovo 14ARE05 No Touchscreen/Pen Input

raymondwzeng, 2021-06-13:
Stock Ubuntu 21.04 with kernel version 5.12.6 has issues picking up the touchscreen and pen input.

I tested changing the kernel to 5.11-rc2 as per [this comment from issue #218](https://github.com/linuxwacom/input-wacom/issues/218#issuecomment-756300996), then 5.12.6 as per the last comment on [here](https://bugzilla.kernel.org/show_bug.cgi?id=207759) which fixes an issue with the pen without the workaround. 

It is a WACF2200 so I assumed that upgrading the kernel to one with the patches would have solved the issue, but perhaps there is more at play here? For the record, there is a non-zero chance that this may be unrelated to the kernel as I recently had some issues following a run-in with Windows.

[sysinfo.b3s2UfeWvZ.tar.gz](https://github.com/linuxwacom/input-wacom/files/6643499/sysinfo.b3s2UfeWvZ.tar.gz)

jigpu, 2021-06-18:
Looking at the source code for kernel version 5.12.6, I see that the fix for #218 is present. I can also confirm that the WACF2200 device is not being picked up by our kernel driver, though it does show up in sysfs under the AMDI0010:00 I2C bus. The symptoms are strikingly similar to #218 so I'm not sure why the fix isn't working in your case.

You mentioned that a fix described by https://bugzilla.kernel.org/show_bug.cgi?id=207759 and present in 5.12.6 "fixes an issue with the pen without the workaround." Could you go into some more detail about what that issue was and what workaround is no longer necessary? I don't know if its related to this issue or not, but it might give me something else to start digging into.

raymondwzeng, 2021-06-19:
Ah, I should have linked it originally.

The issue in the Bugzilla is referenced in [HID Descriptors #99](https://github.com/linuxwacom/wacom-hid-descriptors/issues/99), which seemed to be a related workaround for a similar issue with the touch screen. I decided to patch the kernel to 5.12.6 due to the last message and assumed that the other previous fixes from prior versions are included. I did not modify the kernel beyond that.

Pinglinux, 2022-08-04:
Do you mean pen works, but touch still doesn't? Have touch ever worked? If touch worked before you bring the stylus to the tablet, it could be the issue that Jason fixed recently: [Force pen out of prox](https://github.com/linuxwacom/input-wacom/commit/3fe3a4540bf8afd529c4fe3b859494c628bc763b).

Pinglinux, 2022-08-12:
I don't see the Wacom device from your dmesg. Upgrading your system is probably the easiest approach to get it working.

Pinglinux, 2022-08-15:
This is a "Lenovo 14ARE05" specific issue, not a Wacom driver issue.

closed 2022-08-15 by Pinglinux