linuxwacom/input-wacom #258: XP Pen Star 03 used to work out of the box in 18.04

tinodesigns, 2021-07-06:
Hi, 

in Ubuntu 18.04 my XP Pen Star 03 used to work fine with the wacom driver right out of the box. All later versions of Ubuntu (and some other distros I don't have a list of) it is not working anymore. It is recognized in the settings, but says pen not recognized and tablet not recognized. 

It is also not working. 

I was able to get it back to work by adding the following section to my 70-wacom.conf in my new Ubuntu 20.04 install but only the pen is recognized in the settings. Not the tablet.

Section "InputClass"
        Identifier "XP Pen Star 03"
        MatchUSBID "5543:0081"
        MatchProduct "TABLET 1060N"
        MatchDevicePath "/dev/input/event*"
        MatchIsTablet "true"
        Driver "wacom"
EndSection

See screenshots for the missing tablet. It used to recognize the XP Pen Star 03 just fine, right out of the box. 

Is there a fix to this and can you guys make sure newer version of Ubuntu will recognize it right away?

[image]
[image]

I opened a bug at https://bugs.launchpad.net/ubuntu/+source/xf86-input-wacom/+bug/1926860 before. It took me  a while to figure out it was the wacom driver.

Thanks a bunch.

Tino

Pinglinux, 2021-09-15:
XP pen devices don't run on Wacom kernel driver. input-wacom can not be the root cause of the issue.

closed 2021-09-15 by Pinglinux

tinodesigns, 2021-09-16:
What do they run on then?
I got it back working when I added "wacom" as driver just the settings for the tablet are missing.

It was confirmed as a xf86-input-wacom package bug on https://bugs.launchpad.net/ubuntu/+source/xf86-input-wacom/+bug/1926860
Is that different from the Wacom kernel driver / input-wacom?

If you close this it would be nice to have at least something that points into the right direction.  

Thanks

Pinglinux, 2021-09-17:
input-wacom is not xf86-input-wacom, which is the X server Wacom driver.

tinodesigns, 2021-09-20:
got it thanks for the clarification