linuxwacom/input-wacom #259: changing active area mapping for Android CTL-672

tuleo1, 2021-07-06:
so this is just as basic as it can get, I was able to change and update the driver of my Wacom in my kernel and turn on hid in the kernel everything works fine the tablet is connected buttons work everything but my whole phone is mapped into the tablet and the active area is just too big for my phone...if I wanna go from right to left or left to right it just takes so much time and it's not efficient u can't draw or do anything with it and since my tablet is medium sized it's just...not good

is there a way to change the active area mapping on Android ? maybe configuring the driver files a bit ? or installing something in the kernel or something...I can do all of that so....

jigpu, 2021-07-14:
Unfortunately there's no way to change the mapping on Android that I'm aware of. That said, if you've gone to the trouble of compiling a kernel for your device though and don't mind a hacky solution, you may have luck with modifying the kernel driver slightly.

If you edit the `drivers/hid/wacom_wac.c` file and search for the string `wacom_features_0x37B` you'll find the following few lines:

~~~
static const struct wacom_features wacom_features_0x37B =
        { "Wacom One by Wacom M", 21600, 13500, 2047, 63,
          BAMBOO_PEN, WACOM_INTUOS_RES, WACOM_INTUOS_RES };
~~~

These define various attributes of the tablet including its active area: 21600 x 13500. You can experiment with reducing those two values by a common factor (e.g. 75%, 50%, etc.) to get a good size. Just as a note, on this device 100 units corresponds to 1mm. The active area will be pinned to the top-left of the tablet, but there's no way to change that without much more invasive changes. I'm also not 100% sure how Android will respond to events outside of the active area, but hopefully it just ignores them :)

closed 2021-07-14 by jigpu

tuleo1, 2021-08-09:
OMG URE AMAZING thank u soooo much!!!! I changed the mapping to 8000x11000 which i think it's good enough! :D it helped me a lot uwu I can finally draw again and this tablet won't be useless anymore! hehe thank u a lot