linuxwacom/input-wacom #261: Wacom tablet active area not mapping correctly when in landscape orientation

lewdlime, 2021-07-09:
See attached video for the representation of the issue with the current driver on Android tablets. I am currently using a CTL4100WL, but this issue also happens on using a Wacom Intuos Pro, and on older models, such as the CTL-480 (Intuos Pen S (2013)). I've asked both Samsung Support who wasn't much help about the problem ("do a factory reset first and then bring it into a service center") and also asked Wacom support, who mentioned that the issue is with how the driver interacts with Android, bringing me here.

I'm sure there might be comments of "just use the S Pen", but the S Pen hurts to hold for extended periods of time when drawing, and doesn't exactly have the best feeling when using it to draw with tilt. And while Dex mode is a good workaround for the orientation/tablet mapping issue, I'm sure a lot of people would like to be able to use all the screen real estate available while in landscape mode, especially now that Clip Studio Paint works on Android.

https://user-images.githubusercontent.com/1916661/125022944-65c87a00-e043-11eb-9475-b2ffe732944e.mov

jigpu, 2021-07-14:
This is a known long-standing issue with Android's tablet mapping code. Our Linux kernel driver sends correct data to the Android OS but the Android driver has problems with proper mapping of external tablet sensors like the Intuos. Current versions of the CTL-4100 firmware do have an "Android Mode", but that it simply distorts the tablet's active area to improve the mapping in portrait mode. I'm not aware of any workarounds for getting reasonable mapping in landscape mode or in other circumstances. I briefly looked at developing a fix for the Android driver in the past but have not had the time to dive into how their driver works.

I'm afraid I can't be much help other than to suggest either filing a bug with Google (which is probably bound to go nowhere) or perhaps reach out to developers on the [XDA-Developers forums](https://www.xda-developers.com/) who are more familiar with the Android codebase and build process. If they are able to develop a fix, Google is much more likely to take notice... I'm happy to provide technical support from the hardware/kernel side if you can get Google or an XDA developer to bite.

lewdlime, 2021-07-14:
I’ll give that a shot, it’s worth a try!!

closed 2021-09-15 by Pinglinux