linuxwacom/input-wacom #262: Cursor jumps erratically, `kfifo has filled, starting to drop events`

Ma124, 2021-07-12:
Hi, thanks for your work on this amazing project.

My pen used to work fine, but doesn't now. Below, you can see a screenshot of the text. I kept continuously writing and you can see that sometimes the pen produced odd shapes and most of the time didn't work at all and skipped whole words or sentences.

[image]

I tried another pen and on multiple kernels with similar results.

`dmesg` is spammed by (at least 815 times):
`wacom 0003:056A:5158.0005: wacom_wac_queue_insert: kfifo has filled, starting to drop events`

This sounded similar to #230 and I wasn't sure if my kernel already had the fix so I installed the latest `input-wacom` version [from the `master` branch](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=input-wacom-dkms-git) and the issue still appeared.

I then tried to patch `input-wacom` myself by just blindly enlarging the FIFO by a factor of 10 with `kfifo_alloc` to see if that fixed it.

```diff
diff --git a/4.5/wacom_sys.c b/4.5/wacom_sys.c
index 24dd73f..206856a 100644
--- a/4.5/wacom_sys.c
+++ b/4.5/wacom_sys.c
@@ -85,7 +85,7 @@ static void wacom_wac_queue_insert(struct hid_device *hdev,
 
 	while (kfifo_avail(fifo) < size) {
 		if (!warned)
-			hid_warn(hdev, "%s: kfifo has filled, starting to drop events\n", __func__);
+			hid_warn(hdev, "%s: kfifo has filled, starting to drop events (4.5)\n", __func__);
 		warned = true;
 
 		kfifo_skip(fifo);
@@ -1330,7 +1330,7 @@ static int wacom_devm_kfifo_alloc(struct wacom *wacom)
 	if (!pen_fifo)
 		return -ENOMEM;
 
-	error = kfifo_alloc(pen_fifo, WACOM_PKGLEN_MAX, GFP_KERNEL);
+	error = kfifo_alloc(pen_fifo, 10*WACOM_PKGLEN_MAX, GFP_KERNEL);
 	if (error) {
 		devres_free(pen_fifo);
 		return error;
```

The issue still occurred. Sometimes without logging anything to `dmesg` sometimes logging the new message, so I can be sure I actually ran the patched version.

I hope something can be done about this as it is quite annoying. I'll gladly help you testing patches or further debugging the issue.

Software Versions:
- Linux 5.12.9-1-MANJARO
- Linux 5.12.9-1-MANJARO with [`input-wacom` at `9af9f`](https://github.com/linuxwacom/input-wacom/commit/9af9f57cae4a3129d5af389e013a1842211100e3)
- Linux 5.12.9-1-MANJARO with [`input-wacom` at `9af9f`](https://github.com/linuxwacom/input-wacom/commit/9af9f57cae4a3129d5af389e013a1842211100e3) with patch
- Linux 5.10.42-1-MANJARO

Laptop: Lenovo L390 Yoga

Pens:
- Wacom Bamboo Ink
- Wacom Bamboo Ink Plus

jigpu, 2021-07-14:
I think it would be useful to capture the raw HID data being sent from the sensor to see if this is a hardware/firmware problem or a driver problem. Please see https://github.com/linuxwacom/input-wacom/wiki/Capturing-a-HID-Recording for details on how to create such a recording. For step 5 of its instructions, please simply write a few sentences -- enough to trigger the kfifo warnings a few times.

Do you have an idea of what might have changed since the pen last worked correctly? Did you install any updates on Linux, change distributions, install new updates or firmware from Windows, etc.?

Ma124, 2021-07-14:
Hi, thanks for the quick response.

I have no idea when the pen stopped working, since I didn't use it for a couple of months. My laptop always stayed on Manjaro, but I do update my kernel regularly. I haven't really used it on Windows except to debug issues which I could not reproduce (but sometimes I even have a hard time triggering the behaviour on Linux).

[The HID log](https://github.com/linuxwacom/input-wacom/files/6817796/hid.log.gz)

jigpu, 2021-07-15:
Thanks. I do see a few suspect areas of the log where a kfifo error could potentially be triggered, but the pen seemed to report its serial number relatively quickly in most instances. I wasn't able to actually trigger a kfifo error when replaying the log on my system, though it looks like strokes were still being dropped since I could not see anything resembling handwriting being written. I reverted the kernel driver to a version before the introduction of the kfifo but it had similar problems.

I notice that the pen is occasionally reporting a very low battery level, so that would be the first thing I would look at. If you haven't used the pens for a few months it is possible that their batteries may have discharged to the point that they are not working reliably. The Bamboo Ink pen appears to use a AAAA battery which can be hard to find, but the Bamboo Ink Plus appears to be rechargable -- I would suggest leaving it to charge for a few hours to see if it helps.

Ma124, 2021-07-24:
Sorry for the long silence, but I had a busy week.

> I could not see anything resembling handwriting being written

I think I configured the colour and thickness of the pen before actually writing for a bit, so maybe that's why that looks odd. I recorded `hid_basic_02.log.gz` where I start writing sooner, maybe that helps.

> reporting a very low battery level

I used the battery sent in the packaging with the Bamboo Ink which I bought a couple of days before opening the issue.

I also borrowed another, older Bamboo Ink from a friend, which worked completely fine (`hid_basic_old_01.log.gz`). And put it's battery into my own one (`hid_basic_03.log.gz`) which also didn't work. Maybe I can try to get some brand-new ones for a future test.

> Bamboo Ink Plus appears to be rechargable

It is, but I suspected it to be broken because it has a charging indicator and I remember that it turned blue once fully charged, but it doesn't do that any more. Which is also why I bought the Bamboo Ink. The recording with the Plus (`hid_plus_01.log.gz`) also went abysmal, as next to no text was actually written. 

## Recordings
`hid_basic_01.log.gz`
- The first recording I sent you
- Bamboo Ink
- Maybe started writing a bit late

[image]

---

[`hid_basic_02.log.gz`]
(https://github.com/linuxwacom/input-wacom/files/6872041/hid_basic_02.log.gz)
- Bamboo Ink
- I almost immediately start writing

[image]
(the parts in over- and underlined in red were wrong)

---

[`hid_basic_03.log.gz`](https://github.com/linuxwacom/input-wacom/files/6872042/hid_basic_03_newish_battery.log.gz)
- Bamboo Ink
- Other (but not completely new) battery that works fine with the older/borrowed Bamboo Ink

[image]

---

[`hid_plus_01.log.gz`](https://github.com/linuxwacom/input-wacom/files/6872046/hid_plus_01_empy.log.gz)
- Bamboo Ink Plus
- Charged via USB
- Maybe broken

[image]
(Basically everything is missing)

---

[`hid_basic_old_01.log.gz`](https://github.com/linuxwacom/input-wacom/files/6872058/hid_basic_old_01.log.gz)
- Older Version of Bamboo Ink
- Works completely fine, no issues at all
- Same battery as in `hid_basic_03.log.gz`
- Not really writing anything in this recording, only wanted to capture serial number.

jigpu, 2021-08-09:
These new recordings match up much better with the screenshots that you've posted. If I extract and plot the data from the logs I see very similar output to what you show. I'm still not sure why my plot of `hid_basic_01.log.gz` in particular does not match up with the screenshot (using the toolbar to change color/thickness isn't the problem), but whatever the case it seems to be unrelated.

The data from `hid_basic_02.log.gz` seems to indicate that your original battery is fairly full. When you switched batteries and recorded `hid_basic_03.log.gz` the status indicates that it is nearing empty (the conversion is not straightforward, but I think somewhere around 25%). Both clearly have trouble with dropping strokes on your system, however, despite the borrowed pen battery working fine with the borrowed pen.

Especially given that the borrowed pen works fine, I now suspect the that the problem is one of compatibility, interference, or an issue with the pen itself.

* **Compatibility:** I've generally had good luck with using pens on a wide range of devices, but there is always a chance that a particular combination will have issues. I don't think this is a great explanation, however, based on what you've said.
    * If you've been using Manjaro the whole time, the kfifo code should have been causing similar problems this whole time. The kfifo code was introduced in 2018 and hasn't really been touched since.
    * Pen compatibility could potentially change if the sensor's firmware was updated. I don't know if any firmware updates were issued (I don't think Wacom makes very many), but since you do use Windows on occasion I suppose there's a chance that this could explain things.

* **Interference:** The serial number data is too big to be sent by the pen in a single transmission window, so it has to be sent in several chunks. RF interference could corrupt the data packets and could significantly increase how long it takes for the serial number to be received successfully.
    * Did you test your own pen and the borrowed pen at the same time?
    * There is a chance that the borrowed pen is a bit more immune to interference since it is a slightly different model. I have no way to know for sure though.
    * You could try using the pen and laptop outside and away from electrical devices to see if it makes any difference.

* **Pen issue:** Its possible that something is wrong with the pen, but since you had left it alone for a few months, I find this explanation also unlikely.

***

I don't know if the root cause of this issue will be found, but working around it is certainly possible. Since increasing the kfifo size doesn't seem to be sufficient, you can try disabling use of the kfifo entirely by commenting out the call to the `wacom_wac_pen_serial_enforce` function. I'm pretty sure this would do the trick:

~~~diff
diff --git a/4.5/wacom_sys.c b/4.5/wacom_sys.c
index 77dcdce..221dde5 100644
--- a/4.5/wacom_sys.c
+++ b/4.5/wacom_sys.c
@@ -187,8 +187,8 @@ static int wacom_raw_event(struct hid_device *hdev, struct hid_report *report,
 	if (size > WACOM_PKGLEN_MAX)
 		return 1;
 
-	if (wacom_wac_pen_serial_enforce(hdev, report, raw_data, size))
-		return -1;
+	//if (wacom_wac_pen_serial_enforce(hdev, report, raw_data, size))
+	//	return -1;
 
 	memcpy(wacom->wacom_wac.data, raw_data, size);
 
~~~

Ma124, 2021-08-18:
Thanks for the comprehensive answer.

I tried the patch inside and in my garden a bit further away from other devices, and it exhibited the usual buggy behaviour both times.
I can also create new recordings if you need them.

> Did you test your own pen and the borrowed pen at the same time?

When I tested the borrowed pen, the my own ones were always in the same room.
When I initially tested my own ones, the borrowed pen wasn't near them.

---

For me personally, this issue dropped in priority because my friend gifted me the pen that works because he doesn't use it that much any more. But I'll still gladly help you to test any other ideas. I'll leave this issue open in case others encounter the same problem.

Pinglinux, 2021-09-15:
It feels like a battery or an incompatible pen issue. Basically, it doesn't look like a driver issue.

closed 2021-09-15 by Pinglinux

sergeG, 2023-06-02:
Disabling use of the kfifo entirely, as suggested above, fixes for me very similar issue that I observe with two laptops  (both Lenovo X1 yoga gen 2). In both cases similar artifacts were observed together with "kfifo has filled..." in logs.

andersk, 2023-11-01:
Probably related: https://bugzilla.kernel.org/show_bug.cgi?id=205809