linuxwacom/input-wacom #263: Kernel not detetcing driver

two10, 2021-07-23:
I performed the steps provided on the wiki with latest 0.47.0 but the kernel is still not detecting the wacom 

Centos Version  - 7.9
Kernel Version - 
`[root@localhost ~]# uname -r
3.10.0-1160.36.2.el7.x86_64
`

 

`Jul 23 14:44:41 localhost kernel: usb 1-2: Product: CTL-472
Jul 23 14:44:41 localhost kernel: usb 1-2: Manufacturer: Wacom Co.,Ltd.
Jul 23 14:44:41 localhost kernel: usb 1-2: SerialNumber: 0KE00L1063577
Jul 23 14:44:41 localhost kernel: input: Wacom CTL-472 Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:056A:037A.0004/input/input19
Jul 23 14:44:41 localhost kernel: wacom 0003:056A:037A.0004: hidraw1: USB HID v1.10 Mouse [Wacom Co.,Ltd. CTL-472] on usb-0000:00:14.0-2/input0
Jul 23 14:44:41 localhost kernel: wacom 0003:056A:037A.0005: Unknown device_type for 'Wacom Co.,Ltd. CTL-472'. Ignoring.`

jigpu, 2021-08-02:
We recently fixed some issues with installing on RHEL 7, but these fixes have not made their way into a release yet. These issues cause the RHEL-supplied driver to be used if the tablet is connected to the system while is is booting. Disconnecting and reconnecting the tablet afterwards does not fix the issue. The only reliable way to have our driver load with the current input-wacom release is to unplug the tablet, reboot the system, and *then* connect the tablet.

I would recommend you use the current development version of the driver so that the input-wacom driver is properly installed and correctly replaces the RHEL-supplied driver. This should fix the issue mentioned above and allow the tablet to work. Follow the steps below to install the development version:

~~~
$ sudo yum install git
$ git clone https://github.com/linuxwacom/input-wacom.git
$ cd input-wacom
$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
~~~

Pinglinux, 2021-09-02:
@two10 Did Jason's suggestion resolve your issue? If we don't hear from you in the next few days, I will close this case.

closed 2021-09-15 by Pinglinux