linuxwacom/input-wacom #264: [ISSUE]  - Unable to install from source in manjaro linux

addy3494, 2021-07-30:
I have installed all the required arch dependencies per suggestion, installation fails as per below ....
-------------------------------------------------------------------------
checking for linux-based kernel... yes
checking for kernel source/headers... not found
configure: WARNING: Unable to find build config in any of:
	  [/lib/modules/5.10.53-1-MANJARO/source],
	  [/lib/modules/5.10.53-1-MANJARO/build],
	  [/usr/src/linux],
	  [/usr/src/linux-5.10.53-1-MANJARO]
	

configure: WARNING: Kernel directory does not appear to have needed config files
checking kernel version...
configure: WARNING: Spcified '5.10.53-1-MANJARO' kernel, but found '' instead
configure: error: We could not find the development environment to build modules for the '5.10.53-1-MANJARO' kernel within the '' directory. Please install the kernel source or the kernel development package and try again.
Build Failed
-------------------------------------------------------------------------

Please advice on how to fix this !!!

jigpu, 2021-08-02:
This error can be caused if a kernel update was recently installed but you have not yet rebooted. If you installed updates prior to attempting to build the driver from source I would suggest rebooting and then trying again. If that does not fix the issue, please run the [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) from our wacom-hid-descriptors repository and attach the generated tarball here for review.

addy3494, 2021-08-04:
Have genrated the tar file
[sysinfo.cRh6j5jbUH.tar.gz](https://github.com/linuxwacom/input-wacom/files/6934630/sysinfo.cRh6j5jbUH.tar.gz)

jigpu, 2021-08-10:
I don't see anything too strange in the sysinfo, but after having set up a Manjaro system for testing I notice that it offers multiple versions of the linux-headers package. If you accidentally chose the wrong version that could explain the problem. Please try running `pacman -S linux510-headers` and then re-attempt the compile.

If that doesn't work, please provide the output of `pacman -Q linux linux-headers` to double-check that the kernel and headers are using matching versions. If they are different you'll want to update your system and then try again.

Pinglinux, 2021-09-02:
@addy3494 Did Jason's suggestion work for you? If we don't hear from you in the next few days, I'll close the issue.

addy3494, 2021-09-03:
Hi @jigpu @Pinglinux 

Pardon for the late response

The suggestion didnt work,

```

~/Downloads/input-wacom-0.47.0 
❯ sudo !!
sudo pacman -S linux510-headers
[sudo] password for gh0s1: 
resolving dependencies...
looking for conflicting packages...

Packages (2) pahole-1.21-1  linux510-headers-5.10.60-1

Total Download Size:    28.75 MiB
Total Installed Size:  124.43 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 pahole-1.21-1-x86_64                                              256.7 KiB   195 KiB/s 00:01 [########################################################] 100%
 linux510-headers-5.10.60-1-x86_64                                  28.5 MiB  4.73 MiB/s 00:06 [########################################################] 100%
 Total (2/2)                                                        28.7 MiB  3.74 MiB/s 00:08 [########################################################] 100%
(2/2) checking keys in keyring                                                                 [########################################################] 100%
(2/2) checking package integrity                                                               [########################################################] 100%
(2/2) loading package files                                                                    [########################################################] 100%
(2/2) checking for file conflicts                                                              [########################################################] 100%
(2/2) checking available disk space                                                            [########################################################] 100%
:: Processing package changes...
(1/2) installing pahole                                                                        [########################################################] 100%
(2/2) installing linux510-headers                                                              [########################################################] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Updating module dependencies...

~/Downloads/input-wacom-0.47.0 took 18s 
➜ 

~/Downloads/input-wacom-0.47.0 took 18s 
➜ ./configure 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... no
checking whether make supports nested variables... no
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... no
checking dependency style of gcc... none
checking if build path '/home/gh0s1/Downloads/input-wacom-0.47.0' has spaces... no
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... not found
configure: WARNING: Unable to find build config in any of:
	  [/lib/modules/5.13.12-1-MANJARO/source],
	  [/lib/modules/5.13.12-1-MANJARO/build],
	  [/usr/src/linux],
	  [/usr/src/linux-5.13.12-1-MANJARO],
	  [/usr/src/linux-2.6]
	

configure: WARNING: Kernel directory does not appear to have needed config files
checking kernel version... 
configure: WARNING: Spcified '5.13.12-1-MANJARO' kernel, but found '' instead
configure: error: We could not find the development environment to build modules for the '5.13.12-1-MANJARO' kernel within the '' directory. Please install the kernel source or the kernel development package and try again.

~/Downloads/input-wacom-0.47.0 
❯ pacman -Q linux linux-headers
linux513 5.13.12-1
linux510-headers 5.10.60-1

```

jigpu, 2021-09-03:
According the output from `./configure` and the `pacman -Q linux linux-headers` it looks like your kernel was recently upgraded from 5.10.x to 5.13.x. This means that the `linux510-headers` package I previously suggested won't work anymore.

You'll need to run `pacman -S linux513-headers` to coincide with the `linux513` kernel that is installed.

addy3494, 2021-09-04:
After installing the right headers - it was successful.

closed 2021-09-05 by addy3494