linuxwacom/input-wacom #265: Cintiq Pro 16 spontaneously enabling touch mode when disabled

loopier, 2021-08-10:
When touch mode is disabled, it randomly enables itself again.  I haven't been able to track a specific set of actions that trigger this, it just happens while drawing on the tablet, but not always.  I know this doesn't help much in tracking down the problem, but I honestly don't see a pattern that provokes this behavior.  It looks like there's some kind of scheduled action happening in the background that keeps enabling it after it has been disabled for a while.

```System:    Kernel: 5.10.53-1-MANJARO x86_64 bits: 64 Desktop: KDE Plasma 5.22.4 Distro: Manjaro Linux 
Machine:   Type: Laptop System: LENOVO product: 20QNCTO1WW v: ThinkPad P53 serial: <filter> 
           Mobo: LENOVO model: 20QNCTO1WW v: SDK0J40697 WIN serial: <filter> UEFI: LENOVO v: N2NET46W (1.31 ) date: 06/22/2021 
Battery:   ID-1: BAT0 charge: 79.2 Wh (100.0%) condition: 79.2/90.0 Wh (88.0%) 
CPU:       Info: 6-Core model: Intel Core i7-9850H bits: 64 type: MT MCP cache: L2: 12 MiB 
           Speed: 800 MHz min/max: 800/4600 MHz Core speeds (MHz): 1: 800 2: 800 3: 800 4: 800 5: 800 6: 800 7: 800 8: 800 
           9: 800 10: 800 11: 800 12: 800 
Graphics:  Device-1: Intel CoffeeLake-H GT2 [UHD Graphics 630] driver: i915 v: kernel 
           Device-2: NVIDIA TU106GLM [Quadro RTX 3000 Mobile / Max-Q] driver: nvidia v: 470.57.02 
           Device-3: Logitech HD Webcam C525 type: USB driver: snd-usb-audio,uvcvideo 
           Device-4: IMC Networks Integrated Camera type: USB driver: uvcvideo 
           Display: x11 server: X.Org 1.20.11 driver: loaded: modesetting,nvidia resolution: 1: 2560x1440~60Hz 
           2: 1920x1200~60Hz 3: 1920x1080 
           OpenGL: renderer: Quadro RTX 3000/PCIe/SSE2 v: 4.6.0 NVIDIA 470.57.02 
// ...
Sensors:   System Temperatures: cpu: 52.0 C mobo: 49.0 C gpu: nvidia temp: 49 C 
           Fan Speeds (RPM): cpu: 2586 fan-2: 2241 
Info:      Processes: 313 Uptime: 2h 52m Memory: 30.98 GiB used: 7.84 GiB (25.3%) Shell: Bash inxi: 3.3.06 ```

loopier, 2021-08-13:
[sysinfo.HZ2EkHiWjq.tar.gz](https://github.com/linuxwacom/input-wacom/files/6983841/sysinfo.HZ2EkHiWjq.tar.gz)

> Has this issue always occurred with this device, or is it a recent change? 

Always, as far as I'm concerned.  I've been with this setup for about a year now.

> Have you noticed if touch suddenly gets enabled when you're using the device, or does it only seem to become enabled while you're away (and you only notice after returning)? 

It happens while using the device.  If it has something to do with the touch toggle softkey, I dare say I'm usually quite away from it, because I'm right handed and my hand rests on the lower part of the screen while drawing.  Could it be static?

> About how long does touch stay disabled before it is re-enabled?
I can't tell.  I'll try to pay attention the next time and see if I can give an update.  But my guess is a few minutes.

Joshua-Dickens, 2021-09-01:
You said this has been an issue since you got the tablet, have you tried it on multiple different computers or just one? And if so what OS's did you try it on?

Pinglinux, 2021-09-02:
@loopier Since the touch switch is a softkey, we/someone/something may accidentally touch it without noticing it. It is possible that the key has been touched somehow. Did you try it on another system, especially running a different OS like Windows, to see the issue, as suggested by @Joshua-Dickens?

If you want to disable touch, you could use: xsetwacom set touch_id touch off
to turn it off. You could also add that xsetwacom to a system start script so it is turned off for you automatically.

jigpu, 2021-09-03:
It might be interesting to see if the `xsetwacom set <touch_id> touch off` command suggested by @Pinglinux works any better. If touch becomes randomly re-enabled after running that command, it could be that the device is disconnecting/reconnecting from the system for some reason (e.g. bad USB port / cable or flaky USB controller / driver). The `journalctl -b0` logs could also be examined for spurious disconnect/reconnect events.

closed 2021-09-15 by Pinglinux