linuxwacom/input-wacom #266: Detection region becomes half sized when unplugging, plugging tablet.

Davidy22, 2021-08-13:


Joshua-Dickens, 2021-08-18:
Hi Davidy22, this seems to be a previously solved issue #174 
Please try the stated fix for that issue and see if it works.

Davidy22, 2021-08-18:
I have noticed other issues where a workaround is prescribed, and the workaround does work, but the issues are all closed for some reason. I'd usually closed my own issue for being a duplicate but the other issues are all closed while the bug is still present and there's only a workaround.

Joshua-Dickens, 2021-08-24:
Alright so I'm trying to find a more permanent solution for you, however I'm trying to replicate your issue but I'm struggling to reproduce the issue. Could you go more into detail on the steps you take to cause the issue, and also run this [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) with the tablet connected and post the resulting tar here? Thanks

Davidy22, 2021-08-25:
To note, the affected devices aren't the full range of wacom products, although they do seem to be [well known](https://github.com/linuxwacom/libwacom/wiki/Troubleshooting#special-cases). Issue occurs when plugging in device, but works if device was plugged in at startup.

[wacom_android.tar.gz](https://github.com/linuxwacom/input-wacom/files/7043207/wacom_android.tar.gz)
[wacom_desktop.tar.gz](https://github.com/linuxwacom/input-wacom/files/7043209/wacom_desktop.tar.gz)

Davidy22, 2021-08-29:
Hum, this alternate workaround seems to work even without rebooting actually, can unplug/plug the tablet and it starts in the correct mode. Need line breaks before each sudo, but I assume github just mangled your formatting. Still a workaround, but potentially implies that the fix should be implemented there. Will check further, may close this if a fix happens/happened there.

Pinglinux, 2021-08-30:
@Davidy22 Thank you for your prompt replies. Since it is an issue related to fwupd, can we close this request?

Davidy22, 2021-08-31:
The earlier test does seem to indicate it's probably fwupd. I headed over there and the maintainer says it's fixed in the latest release, although getting it installed is fairly involved and I haven't been able to check if it is fixed. [This resource](https://github.com/linuxwacom/libwacom/wiki/Troubleshooting#special-cases) could possibly be updated, and repos are going to need to have fwupd updated to 1.6.3 or later to fix this.

closed 2021-09-15 by Pinglinux

Davidy22, 2021-09-26:
Alright, fwupd 1.6.4 released with a snap that I could install and I can probably confirm that it does indeed fix the issue, which means that the resource in the documentation will be obsolete and removable once all distros update fwupd.