linuxwacom/input-wacom #269: Is Wacom one 13.3" not yet supported?

rehabmass, 2021-08-27:
Hi there,

I'm on fedora 25, and the Wacom One "Creative Pen Display" works after installing the driver only as monitor, i.e. touch and pen do not work. 
I followed the instructions at:
https://github.com/linuxwacom/xf86-input-wacom/wiki/wdaemon

Until I tried :
#wdaemon -a

I get:
Unable to find the wacom udev directory (/dev/input/wacom-tablets) or no tablets found. Trying all event devices
Tablet not supported yet: 0x56a:0x3a6

Could that be because I'm using an old Kernel? Or the device is not yet supported?

Thanks in advance!

Rehab

rehabmass, 2021-08-29:
According to 
056a:03a6 	DTC133 	Wacom One Creative Pen Display 		4.10 	2.6.32 	1.3
https://github.com/linuxwacom/input-wacom/wiki/Device-IDs
as my Kernel is 4.13 it seems that drivers should work, right? 
I should install libwacom 1.3 (rather than 0.47.0 which I have intalled), is there currently a development branch where version 1.3 is being developed?

rehabmass, 2021-08-29:
Ok, I guess 1.3 is the minimum libwacom version, after installing it, and running 
$ libwacom-list-local-devices
# Device node: /dev/input/event7
[Device]
Name=Wacom One Pen Display 13
ModelName=DTC133
DeviceMatch=usb:056a:03a6;
Class=PenDisplay
Width=11
Height=6
IntegratedIn=Display;
Styli=0xffffe;0xfffff;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=0

---------------------------------------------------------------
Now that it is correctly detected, it is still the case that when I try: 
sudo wdaemon -a
I get that the device is not supported. 
Is that because my laptop itself doesn't have touch screen?

rehabmass, 2021-08-29:
After installing libwacom 1.3 the pen seems to be detedted and working properly (pen cursor appears and it can move/close windows). However it doesn't work yet inside any application. For example, Gimp input devices settings see the Wacom tablet, but it says permission denied; and the pen doesn't work inside the app.

Pinglinux, 2021-09-02:
Why did you need to run: "sudo wdaemon -a"? Do you use xorg.conf to config the device or something on your system need to access the device port statically? You only need to update your kernel driver (from input-wacom) and libwacom to make the device work. If your system is running on a kernel older than 3.7, you need to update xf86-input-wacom as well. Otherwise, input-wacom and libwacom should make the device work for you.

closed 2021-09-15 by Pinglinux