linuxwacom/input-wacom #270: Touch Event on Ubuntu 20.04 fires too many clicks (Wacom DTH-2452)

oxydon, 2021-09-21:
I run Ubuntu 20.04LTS with a Wacom-Display DTH-2452. The pencil works fine but I have an issue with the fingertouch. The touch event is registered correctly however it is not recognized as a correct click event but as a lot of clicks. Even a short contact is interpreted as 5 clicks or more and if I rest my finger on a button the wacom keeps on fireing click-events instead of mouse-down-events. I tried to fiddle around with the parameters for a couple of days but nothing seems to change anything. Even in a completly new installed Ubuntu 20.04 I have the same problem so I don't think it's my configuration...

I attach the config file of the wacom Touch Device (Wacom DTH-2452 TOUCH Finger touch):
```
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Button: Actions are not supported by xorg.conf. Try shell format (-s) instead.
Property 'Wacom Hover Click' does not exist on device.
Property 'Wacom Hardware Touch Switch' does not exist on device.
Property 'Wacom Proximity Threshold' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Wheel Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Strip Buttons' does not exist on device.
Property 'Wacom Pressure Recalibration' does not exist on device.
Option "Area" "0 0 17568 9882"
Option "DebugLevel" "5"
Option "CommonDBG" "0"
Option "Suppress" "100"
Option "RawSample" "4"
Option "PressCurve" "90 0 95 50"
Option "Mode" "Absolute"
Option "Touch" "on"
Option "Gesture" "on"
Option "ZoomDistance" "0"
Option "ScrollDistance" "0"
Option "TapTime" "250"
Option "Rotate" "none"
Option "Threshold" "2000"
Option "Serial" "0"
Option "PanScrollThreshold" "1300"
```

And the list of the registered devices:
```
Wacom DTH-2452 Pen stylus       	id: 12	type: STYLUS    
Wacom DTH-2452 Pad pad          	id: 13	type: PAD       
Wacom DTH-2452 TOUCH Finger touch	id: 14	type: TOUCH     
Wacom DTH-2452 Pen eraser       	id: 16	type: ERASER
```

I'd very much appreciate your help :)
Thanks a lot in advance!

Joshua-Dickens, 2021-09-29:
Hi Oxydon, have you tried to use this tablet on another computer and if so does the same issue happen? Another question, is this an issue that came up after normal use or has the tablet done this since you got it?

oxydon, 2021-09-29:
Hi Joshua-Dickens,
under Windows it worked just fine with the same Hardware-Computer. We have several of those Screens (not Tablets) and the issue exists on all 32 of them. We have the issue since we changed the OS from Windows to Ubuntu and then right from the start.

Joshua-Dickens, 2021-09-29:
Thanks! I'll look into this asap.

jigpu, 2021-11-10:
Fix for the issue has been sent and accepted by the HID maintainer upstream. It should be appearing in stable kernels "soon" and we will be working on our own backport to input-wacom shortly.

Joshua-Dickens, 2021-12-02:
Hi Oxydon, sorry for the bit of delay but the patch has been backported to the input-wacom repository and the fix available to you now. If you build the now up to date repository your issue should no longer be a problem.

oxydon, 2021-12-05:
I am sorry to inform you that the issue seems to persist. I installed both the `xf86-input-wacom` and `input-wacom` from source according to your instructions given in the readme files. The installation worked just fine. "Build failed" was never prompted. However one Touch with a finger still results in the same amount of clicks as before.

Joshua-Dickens, 2021-12-07:
Are you sure you built the latest repository from source? I just tested it locally and the fix seems to hold on my end.

oxydon, 2021-12-08:
I used the latest two releases. Are the changes included in those already or do i need to download the current master?

Joshua-Dickens, 2021-12-08:
We haven't made a new release with the fix, you'll want to build it from the current master yes.

oxydon, 2021-12-10:
Thanks a lot! When I build from master, everything works as expected!! Great Work!!

closed 2021-12-10 by oxydon